ArduCAM/Arduino #77: Control of exposure

aldobranti, 2016-08-18:
@ArduCAM how can I control the "shutter" time and the lighting level of a shot ?

-- I have OV5642 on RevC+ shield. I notice that images are blurred if the camera is moving; they are also over exposed in broad daylight

[image]

aldobranti, 2016-08-28:
@ArduCAM is there no way to control the "shutter" time on this rig ?  It's not operating in a low light environment

aldobranti, 2016-08-29:
It is not as if this functionality isn't exposed 
http://www.dragonwake.com/download/camera/ov5642/OV5642-DS.pdf

ArduCAM, 2016-08-30:
@aldobranti, the given examples use the auto-exposure, if you want to control the exposure time, you have to set to the manual exposure yourself and write the exposure value to the regsiter.

4.6.2  manual exposure control
To manually change exposure value, you must first set both 0x3503[0] and 0x3503[2], where 0x3503[0] enables manual 
exposure control and 0x3503[2] enables manual frame length - the number of lines in each frame or maximum exposure 
time, which is defined by registers 0x350C and 0x350D. In auto exposure mode, the maximum exposure values in 
registers 0x350C/0x350D automatically change. In manual exposure mode, these registers will not automatically 
change. The manually set exposure in registers 0x3500~0x03502 must be less than the maximum exposure value in 
0x350C/0x350D. The exposure value in registers 0x3500~0x3502 is in units of line*16 - the low 4 bits (0x3502[3:0]) is 
the fraction of line, the maximum value in 0x350C/0x350D is in unit of line. If the manually set exposure value is less than 
one pre-defined frame period (e.g., 1/15 second in 15fps), there is no need to change 0x350C/0x350D. If the exposure 
value needs to be set beyond the pre-defined frame period; in another words, if the frame period needs to be extended 
to extend exposure time, then the maximum frame value in 0x350C/0x350D needs to be set first, then the exposure can 
be set in registers 0x3500~0x3502 accordingly.

aldobranti, 2016-08-31:
Nice ! beautiful cut and paste action, wish I had your job if that it is all it takes to support and develop this product

goetzmoritz, 2017-04-15:
Mr. Lee, could you provide a little code snippet to set the regs accordingly with arduino?
I tried:
myCAM.wrSensorReg16_8(0x3503 ,0x01);
myCAM.wrSensorReg16_8(0x3500 ,0x10); 
myCAM.wrSensorReg16_8(0x3502 ,0x40);

But it did not do anything.

dklemer, 2017-05-02:
**aldobranti** is correct -- it's all in the document in the link above. The register settings to control exposure are given on page 43 of the manual.

From the quoted text from **Arducam**: "you must first set both 0x3503[0] and 0x3503[2]," meaning that bits 0 and 2 of register 0x3503 must be set ([0x3503] = 0x05). Then you can set the 20-bit exposure value in registers 0x3500-0x3502, keeping in mind that it is big-endian. So the code to set an exposure of 240 lines would be:
```
myCAM.wrSensorReg16_8(0x3503 ,0x05);
myCAM.wrSensorReg16_8(0x3500 ,0x00);
myCAM.wrSensorReg16_8(0x3501 ,0x00);
myCAM.wrSensorReg16_8(0x3502 ,0xF0);
```
This worked for me. HTH.

alippe27, 2017-06-05:
I am having trouble setting the exposure time for the ArduCAM mini 2MP (OV2640). How do I calculate the values to write to the register in order to get a 2ms exposure? I looked at this PDF http://www.arducam.com/downloads/modules/OV2640/OV2640%20Camera%20Module%20Software%20Application%20Notes1.04%20OVT.pdf and I'm trying to figure out a formula for converting between exposure in seconds and exposure in line units.

alippe27, 2017-06-15:
I have been working on writing code for exposure control of the OV2640 and ran into some trouble. I read the values of registers 4,10,13 and 45 (0x04, 0x10, 0x13, 0x45 respectively) using this code:
	 myCAM.rdSensorReg8_8(0x13,&R13);
	  myCAM.rdSensorReg8_8(0x04,&R4);
 	 myCAM.rdSensorReg8_8(0x10,&R10);
 	 myCAM.rdSensorReg8_8(0x45,&R45);
and then used the equation (exposure time=tline*AEC value) and tline=1922 according to the data sheet.
I executed this code to change the register values accordingly:

 uint16_t Exp=5000000; //Exposure time in us
  uint16_t a=round(Exp/53.39316); //DEC number corresponding to exposure time 
  uint16_t r10rs=8;
  uint16_t r45ls=10;
  uint16_t r4s=14;
  uint16_t r10ls=6;
  uint16_t b = a<<r4s;
  uint8_t r4 = b >> r4s; //Extracts bits 0 and 1 
  uint8_t r10= (a << r10ls) >> r10rs; // Extracts bits 2-9
  uint8_t r45= a >> r45ls; // Extracts bits 10-15

  for (int i=0; i<2; i++){
  bitWrite(R4,i,bitRead(r4,i));
}

for (int i=0; i<8; i++){
  bitWrite(R10,i,bitRead(r10,i));
}

for (int j=0; j<5; j++){
  bitWrite(R45,j,bitRead(r45,j));
}

  bitClear(R13,0);
  bitClear(R13,1);

 myCAM.wrSensorReg8_8(0x13,(int) &R13);
  myCAM.wrSensorReg8_8(0x04,(int) &R4);
  myCAM.wrSensorReg8_8(0x10,(int) &R10);
  myCAM.wrSensorReg8_8(0x45,(int) &R45);

But when I tested different Exp values (exposure times) nothing changed.

alippe27, 2017-06-20:
I figured it out. Thank you Lee for the helpful link: http://www.arducam.com/manual-exposure-ov2640/

void loop(){
  delay(1000);
  uint8_t R4; //Register 4: bits 0 and 1 correspond to bits 0 and 1 of exposure multiplier
  uint8_t R10; //Register 10: the 8 bits correspond to bits 2-9 of exposure multiplier
  uint8_t R13; // Register 13: bit 0 switches auto exposure control on(1)/off(0); bit 2 switches auto gain control on(1)/off(0)
  uint8_t R45; // Register 45: bits 0-5 correspond to bits 10-15 of exposure multiplier
  myCAM.wrSensorReg8_8(0xFF, 0x01);
  myCAM.rdSensorReg8_8(0x13,&R13);
  myCAM.rdSensorReg8_8(0x04,&R4);
  myCAM.rdSensorReg8_8(0x10,&R10);
  myCAM.rdSensorReg8_8(0x45,&R45);
  Serial.print("Register 4: ");
  Serial.println(R4,BIN);
  Serial.print("Register 10: ");
  Serial.println(R10,BIN);
  Serial.print("Register 13: ");
  Serial.println(R13,BIN);
  Serial.print("Register 45: ");
  Serial.println(R45,BIN);

  
  uint16_t Exp[]={50,500,5000,50000}; //Exposure time in us
for (int j = 0; j<4; j++){
  
  uint16_t a=round(Exp[j]/53.39316); //DEC number corresponding to exposure time 

  Serial.print("AEC byte: ");
  Serial.print(a,DEC);
  Serial.print(" | ");
  Serial.println(a,BIN);
 // a = 1 << 15;

  uint16_t r4shift=14;
  uint16_t r10leftshift=6;
  uint16_t r10rightshift=8;
  uint16_t r45leftshift=10;
  uint16_t b = a<<r4shift;
  uint8_t r4 = b >> r4shift; //Extracts bits 0 and 1 
  uint8_t r10= (a << r10leftshift) >> r10rightshift; // Extracts bits 2-9
  uint8_t r45= a >> r45leftshift; // Extracts bits 10-15

  Serial.print("r4: ");
  Serial.println(r4,BIN);
  Serial.print("r10: ");
  Serial.println(r10,BIN);
  Serial.print("r45: ");
  Serial.println(r45,BIN);

  for (int i=0; i<2; i++){
  bitWrite(R4,i,bitRead(r4,i));
}

for (int i=0; i<8; i++){
  bitWrite(R10,i,bitRead(r10,i));
}

for (int i=0; i<6; i++){
  bitWrite(R45,i,bitRead(r45,i));
}

  bitClear(R13,0);
//  bitClear(R13,2);

  Serial.print("R4: ");
  Serial.println(R4,BIN);
  Serial.print("R10: ");
  Serial.println(R10,BIN);
  Serial.print("R45: ");
  Serial.println(R45,BIN);
  Serial.print("R13: ");
  Serial.println(R13,BIN);

  
  
  myCAM.wrSensorReg8_8(0x13,(int) R13);
  myCAM.wrSensorReg8_8(0x04,(int) R4);
  myCAM.wrSensorReg8_8(0x10,(int) R10);
  myCAM.wrSensorReg8_8(0x45,(int) R45);

  uint8_t new13, new4, new10, new45;

  myCAM.rdSensorReg8_8(0x13,&new13);
  myCAM.rdSensorReg8_8(0x04,&new4);
  myCAM.rdSensorReg8_8(0x10,&new10);
  myCAM.rdSensorReg8_8(0x45,&new45);
  Serial.print("Register 4: ");
  Serial.println(new4,BIN);
  Serial.print("Register 10: ");
  Serial.println(new10,BIN);
  Serial.print("Register 45: ");
  Serial.println(new45,BIN);
  Serial.print("Register 13: ");
  Serial.println(new13,BIN);

  myCAMSaveToSDFile((int) Exp[j]);
  
}
Serial.println("Done everything! Sleepy time");

}

javramos, 2019-04-05:
Hi there! I am dealing with the same thing. Since my application will need longer exposure times I am wondering how to add the dummy lines time and dummy pixel time to the equation and how to relate them with the time in us. Thanks in advance for your help!

ArduCAM, 2019-04-07:
@javramos You have to manually set the AEC and AGC to manual mode, then change the REG2A (0x2A) and FRARL (0x2B), ADDVSL (0x2D), ADDVSH (0x2E) for total line length and total frame length.
You have to read the datasheet carefully to find all these details.

kaansarica, 2019-09-03:
Hello everyone, 

I have question too. I am using OV5642 too. I did set exposure value regarding to your guidances. But i am still having trouble determining actual exposure time and couldn't find any information about how to get actual exposure time. 
And anyone knows what is the system frequncy of Arducam 5MP module? It looks somewhat related to exposure time. Thanks in advance for your help.