ArduCAM/Arduino #79: In SD Card saved corrupt Image from ArduCAM_ESP8266_V2_OV5642_Capture2SD

TU-KL-Crazyflie, 2016-08-19:
Hi,

We used the Adafruit ESP8266 board. The Camera works well with the Example "ArduCAM_ESP8266_OV5642_Capture". It takes picture and sends it to Computer. 

So as the next step, we are trying to capture a picture and then save it in SD card. We used the exsample code "ArduCAM_ESP8266_V2_OV5642_Capture2SD". Now the problem is, that the camera do save a JPG file in SD card, but the file cannot open. The file is here attached.

Have someone else also the problem?

Thanks very much!

[image]

Nick223, 2016-08-19:
At first: change card reader wires a few times. I had the same problem with mine. Text file was corrupted or created but not overwritten.

ArduCAM, 2016-08-20:
@Nick223, make sure if the camera work with host application with USB-Serial connection before moving to SD card.
would you please attach the card reader photo?  Sometimes the sd card reader is not well designed and occupy the SPI bus without tri-state capacity.

ArduCAM, 2016-08-20:
@TU-KL-Crazyfile, From your JPG file, there is no JPEG header, I'm afraid that the camera is not well initialized to JPEG mode.
Would you please show me your setup?

Nick223, 2016-08-20:
I'm not using SD Card reader. It's not my thread. I have bigger problems: your camera is working with Arduino but not with ESP8266. At all.

ArduCAM, 2016-08-20:
@Nick223, The ESP8266 library should be modified to work with ArduCAM, you should follow the user guide from the link: http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/

Nick223, 2016-08-20:
You sent me nothing.

`Arduino: 1.6.6 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

WARNING: Category '' in library ArduinoJson is not valid. Setting to 'Uncategorized'
Build options changed, rebuilding all
In file included from C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:56:0,

                 from C:\Users\MIKOAJ~1\AppData\Local\Temp\arduino_003e657806bfcbefb02f8c324913aa4d\ArduCAM_ESP8266_OV5642_Capture.ino:18:

C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/detail/RequestHandler.h:7:18: warning: unused parameter 'method' [-Wunused-parameter]

     virtual bool canHandle(HTTPMethod method, String uri) { return false; }

                  ^

C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/detail/RequestHandler.h:7:18: warning: unused parameter 'uri' [-Wunused-parameter]

C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/detail/RequestHandler.h:8:18: warning: unused parameter 'uri' [-Wunused-parameter]

     virtual bool canUpload(String uri) { return false; }

                  ^

C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/detail/RequestHandler.h:9:18: warning: unused parameter 'server' [-Wunused-parameter]

     virtual bool handle(ESP8266WebServer& server, HTTPMethod requestMethod, String requestUri) { return false; }

                  ^

C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/detail/RequestHandler.h:9:18: warning: unused parameter 'requestMethod' [-Wunused-parameter]

C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/detail/RequestHandler.h:9:18: warning: unused parameter 'requestUri' [-Wunused-parameter]

C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/detail/RequestHandler.h:10:18: warning: unused parameter 'server' [-Wunused-parameter]

     virtual void upload(ESP8266WebServer& server, String requestUri, HTTPUpload& upload) {}

                  ^

C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/detail/RequestHandler.h:10:18: warning: unused parameter 'requestUri' [-Wunused-parameter]

C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src/detail/RequestHandler.h:10:18: warning: unused parameter 'upload' [-Wunused-parameter]

In file included from C:\Users\MIKOAJ~1\AppData\Local\Temp\arduino_003e657806bfcbefb02f8c324913aa4d\ArduCAM_ESP8266_OV5642_Capture.ino:19:0:

C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS/ESP8266mDNS.h:66:8: warning: unused parameter 'ip' [-Wunused-parameter]

   bool begin(const char* hostName, IPAddress ip, uint32_t ttl=120){

        ^

C:\Users\Mikołaj\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266mDNS/ESP8266mDNS.h:66:8: warning: unused parameter 'ttl' [-Wunused-parameter]

C:\Users\MIKOAJ~1\AppData\Local\Temp\arduino_003e657806bfcbefb02f8c324913aa4d\ArduCAM_ESP8266_OV5642_Capture.ino: In function 'void setup()':

ArduCAM_ESP8266_OV5642_Capture:206: error: 'OV5642_CHIPID_HIGH' was not declared in this scope

   myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);

                         ^

ArduCAM_ESP8266_OV5642_Capture:207: error: 'OV5642_CHIPID_LOW' was not declared in this scope

   myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);

                         ^

exit status 1
'OV5642_CHIPID_HIGH' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
`

TU-KL-Crazyflie, 2016-08-20:
@ArduCAM 
thanks for your help!
I haven't changed the code in exsample. And the exsample "ArduCAM_ESP8266_OV5642_Capture" works well, so I think the Camera works with my ESP8266. So I really have no idea about this problem. 

#include <ArduCAM.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
// use #define OV5642_CAM_BIT_ROTATION_FIXED in memorysaver.h for the HUZZAH !!!!

// set GPIO16 as the slave select :
const int CS = 16;
// Version 2,set GPIO15 as the slave select :
const int SD_CS = 15;
ArduCAM myCAM(OV5642, CS);

void setup(){
  uint8_t vid, pid;
  uint8_t temp;

  Wire.begin();
  Serial.begin(115200);
  Serial.println("ArduCAM Start!");

  //set the CS as an output:
  pinMode(CS,OUTPUT);
  digitalWrite(CS, HIGH);
  
  pinMode(SD_CS, OUTPUT);
  digitalWrite(SD_CS, HIGH);
  delay(1);
  
  // initialize SPI:
  SPI.begin();
  SPI.setFrequency(4000000); //4MHz
  delay(1000);

      //Initialize SD Card
  if(!SD.begin(SD_CS)){
    Serial.println("SD Card Error");
  }
  else
  Serial.println("SD Card detected!");

  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  
  Serial.println(temp, HEX); //######################
     
  if (temp != 0x55){
    Serial.println("SPI1 interface Error!");
    while(1);
  }
 
   
  myCAM.clear_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK); //disable low power
  delay(100);
//Check if the camera module type is OV5642
//myCAM.wrSensorReg16_8(0xff, 0x01);               //### sorgt seltsamerweise dafür, dass die Kamera gebrickt wird. -> powercyclen !!!
  myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);

  Serial.println(vid, HEX); //######################
  Serial.println(pid, HEX); //######################
  
   if((vid != 0x56) || (pid != 0x42))
   Serial.println("Can't find OV5642 module!");
   else
   Serial.println("OV5642 detected.");
   myCAM.set_format(JPEG);
   myCAM.InitCAM();
   myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
   myCAM.OV5642_set_JPEG_size(OV5642_640x480);        //################################ Auflösung anpassen!!!
}

TU-KL-Crazyflie, 2016-08-20:
@Nick223 At first the Camera also works not with my ESP8266. But if I build the SD card reader together with Camera in ESP8266. The Camera works somehow. I dont know why, but it just works. :)

Nick223, 2016-08-20:
Very interesting. I must try out adding SD card reader.
But I've seen people that got this to work without it.
I mean, video streaming example (ArduCAM_ESP8266_OV5642_Capture) doesn't even utilize SD cards.

TU-KL-Crazyflie, 2016-08-20:
@Nick223 do you konw Logic Analyzer?

https://www.saleae.com/?gclid=Cj0KEQjw_9-9BRCqpZeZhLeOg68BEiQAOviWAk4BeMhGAz03DdZFSiuFi4jjsao3RMDjBiXKlSdR_KsaAgxD8P8HAQ

We want to use the software to find out, why the Camera works not. And the magic things is, that if we connected the Camera with the black saleae box, it works...... There is some things to do with the SCK pin and the Ground Pin

TU-KL-Crazyflie, 2016-08-20:
@Nick223 Maybe I understand your Question not right, my Code can pass the complie without SD Card. But SPI only works with SD Card.

Nick223, 2016-08-20:
And that's what I think is odd but interesting.

ArduCAM, 2016-08-22:
@Nick223, Have you read the user guide http://www.arducam.com/downloads/ESP8266_UNO/ArduCAM_ESP8266_UNO_DS.pdf from the post: http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/ 
You have to manually modify the memorysaver.h in the library to enable the OV5642 camera and disable the OV2640 camera. Only one camera can be enabled at a time.

ArduCAM, 2016-08-22:
@TU-KL-Crazyfile, It seems that you use the wrong SD_CS pins, please check the ESP8266 UNO pinout from http://www.arducam.com/pinout-esp8266-20160612update/

Nick223, 2016-08-22:
I didn't read it. The first thing I did was changing definitios.

Nick223, 2016-08-22:
I finally got it to work.
Resets!
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000318
~ld

Nick223, 2016-08-22:
Another reset:

ets Jan  8 2013,rst cause:4, boot mode:(1,7)
wdt reset

ryonlabaw, 2016-09-26:
ESP8266 + Arducam mini  5mp does work well.

According to the manual, wdt resets are due to not freeing up the processor enough during your procedures. 

I solved my esp8266 wdt reset issues by adding a few delay() and yield() calls at various points (reading/writing). I haven't connected an SD yet. 
Although, I can confirm that the arducam mini 5mp + esp8266 is able to stream any resolution reliably over the network. It is also possible to save to the on-board flash. 

320x240 has about a 1 second delay during network streaming. It may be possible to get better results.
 
Also, I am using the github version of esp8266-arduino library ... and using the spi.transfer() method for reading data.

ArduCAM, 2016-09-26:
@Ryon, thanks for your effort. Would you please provide your example code, so we can update ours.

ryon, 2016-09-26:
@ryonlabaw See above :)

ryonlabaw, 2016-09-28:
@ArduCAM 
Here is the working sketch ... the important part of this is the "serverStream()" function ... that is where I've made the changes to your original script.

Using wget, I am seeing 17KB/s sustained transfer. My goal is to squeeze more performance out of this combo. The esp8266 can transfer data much faster. Obviously the copy spi data from the camera is slowing things down.

See this topic on fast wifi transfer...
[Found a way to upload data VERY fast by wifi](https://github.com/esp8266/Arduino/issues/1853)

As far as results for the sketch below;
I have been able to stream for hours without issue... longest test was about 6hrs before I shut it off to go to bed. 

Hardware: Wemos D1 Mini + Arducam mini 5mp ... 

`// ArduCAM Mini demo (C)2016 Lee
// web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM ESP8266 5MP camera.
// This demo was made for ArduCAM ESP8266 OV5642 5MP Camera.
// It can take photo and send to the Web.
// It can take photo continuously as video streaming and send to the Web.
// The demo sketch will do the following tasks:
// 1. Set the camera to JEPG output mode.
// 2. if server.on("/capture", HTTP_GET, serverCapture),it can take photo and send to the Web.
// 3.if server.on("/stream", HTTP_GET, serverStream),it can take photo continuously as video
//streaming and send to the Web.

// This program requires the ArduCAM V3.4.1 (or later) library and ArduCAM ESP8266 5MP camera
// and use Arduino IDE 1.5.8 compiler or above

// In "memorysaver.h", uncomment the following definition
//  #define OV5642_CAM
//
// and the others are commented out

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"

// Enabe debug tracing to Serial port.
#define DEBUGGING

// Here we define a maximum framelength to 64 bytes. Default is 256.
#define MAX_FRAME_LENGTH 64
//#define MAX_FRAME_LENGTH 256

// Define how many callback functions you have. Default is 1.
#define CALLBACK_FUNCTIONS 1

// set GPIO16 as the slave select :
const int CS = 16;

int wifiType = 0; // 0:Station  1:AP
const char* ssid = "ssidhere"; // Put your SSID here
const char* password = "passhere"; // Put your PASSWORD here

ESP8266WebServer server(80);

ArduCAM myCAM(OV5642, CS);
int status = WL_IDLE_STATUS;

void start_capture() {
  //Flush the FIFO
  myCAM.flush_fifo();
  //Clear the capture done flag
[... 247 lines cut ...]
    Serial.println(WiFi.localIP());
  } else if (wifiType == 1) {
    Serial.println();
    Serial.println();
    Serial.print("Share AP: ");
    Serial.println(ssid);

    WiFi.mode(WIFI_AP);
    WiFi.softAP(ssid, password);
    Serial.println("");
    Serial.println(WiFi.softAPIP());
  }

  // Start the server
  server.on("/capture", HTTP_GET, serverCapture);
  server.on("/stream", HTTP_GET, serverStream);
  server.onNotFound(handleNotFound);
  server.begin();
  Serial.println("Server started");
}
void loop() {

  server.handleClient();
}
`