ArduCAM/Arduino #80: rducam RevC shield _occasionally_ fails to load a valid jpeg

aldobranti, 2016-08-28:
I have a revC on a Mega2560 with OV5642 

About 1/3 of the time it fails to record a valid jpeg file ( see attached )

[image]

looking at the leading bytes of this it clearly doesnt start with an FFD8 group so it seems like we hadn't synchronised up with the start of the jpeg format -- some kind of timing issue ? Why does it sometimes catch a valid jpeg and sometimes it does not ? 

my setup
`//Check if the camera module type is OV5642
  myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
  if ((vid != 0x56) || (pid != 0x42)) {
    Serial.print("Can't find OV5642 module! vid=0x"); Serial.print(vid, HEX); Serial.print(" pid=0x"); Serial.println(pid, HEX);
    //while (1);
  } else
    Serial.println("OV5642 detected");

  myCAM.set_format(JPEG);
  Serial.println("Changed to jpeg capture mode");
  myCAM.InitCAM();
  Serial.println("initialized the OV5642 module ");

  myCAM.OV5642_set_JPEG_size(OV5642_1920x1080);

  myCAM.set_bit(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);

  myCAM.clear_fifo_flag();
  myCAM.write_reg(ARDUCHIP_FRAMES, 0x00);
`
the above code was developed out of the back of issue #72

here is my capture code 

`//Flush the FIFO
  Serial.println("click it: flush");
  myCAM.flush_fifo();
  //Clear the capture done flag
  Serial.println("click it: fifo flag");
  myCAM.clear_fifo_flag();
  //Start capture
  Serial.println("click it: capt start");
  myCAM.start_capture();
  Serial.println("capturing");
  boxState = capturing;
  // Wait for the capture to finish in loop().
`

and here is my storing code, looking for the capture done in loop()

`if ( myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK) ) boxState = storing ;`

and writing to the onboard sd card

`  //Parse JPEG data from FIFO
  while ( (temp != 0xD9) | (temp_last != 0xFF)  ) {
    temp_last = temp;
    temp = myCAM.read_fifo();
    //Write image data to buffer if not full
    if (i < 256)
      buf[i++] = temp;
    else {
      //Write 256 bytes image data to file
      //Serial.print("*");
      outFile.write(buf, 256); buffCnt++;
      i = 0;
      buf[i++] = temp;
    }

  }
  //Serial.print("finished (?) temp=0x"); Serial.print(temp, HEX); Serial.print(" temp_last=0x"); Serial.println(temp_last, HEX);
  //Write the remaining bytes in the buffer
  if (i > 0)
    outFile.write(buf, i);
  //Close the file
  outFile.close();
`

aldobranti, 2016-09-06:
@ArduCAM  this is outrageous, more than half of the capture events today left a corrupt jpeg.
here is only one of dozens 
[image]

this product is definitely a fail

ArduCAM, 2016-09-07:
@aldobranti,
Actually the image is good, please check the attached photo.
I guess you didn't remove the dummy byte at the very beginning, so the header of the JPEG is wrong, and the windows can not open the JPEG file.
I help you remove the dummy byte 0xFF at the JPEG file header, you can see the good result from the attached photo.
Let me know if you need more help.

ArduCAM, 2016-09-09:
@aldobranti,
Failed to upload your photo. Now try again, see below:
[image]

Fred54, 2016-12-09:
I am also having this same problem. I am using a Arducam F Rev.C+ Camera Module Shield with OV2640 for Arduino UNO. https://www.amazon.com/Arducam-Camera-Module-Arduino-Mega2560/dp/B013OBUJ0W/ref=cm_cd_al_qh_dp_t

 It seems there should be a better way than opening a hex editor and fixing each image.

All of the demos that give me JPEGs have the same problem. The one that creates a BMP works fine and working with the program that sends an image straight to windows works fine.

Is there a way to fix the demos to work correctly?
[image]

aldobranti, 2016-12-09:
I suspect that @ArduCAM is running around like a headless chicken without any time to go back and fix up the demos. It's not as if he hasn't a view on the problem #85 though I think that a post edit of a big jpeg, several Mb, is a most ineffective solution
for my part I get acceptable results with the small fix to the demo loader 
`//Parse JPEG data from FIFO
  while ( (temp != 0xD9) | (temp_last != 0xFF)  ) {
    temp_last = temp;
    temp = myCAM.read_fifo();
    //Write image data to buffer if not full
    if (i < 256)
      buf[i++] = temp;
    else {
      //Write 256 bytes image data to file
      outFile.write(buf, 256); buffs++;
      i = 0;
      buf[i++] = temp;
    }
  }`

What I am still waiting for is exposure control

ArduCAM, 2016-12-09:
@aldobranti and @brandan,
Sorry for the bug, we have massive work to do here.
Please be patient, we will tell our engineer to optimize the demos like I mentioned before in the library soon.

workaround below. I suggest you to find the correct combination of FF D8 for JPEG start, and read through all the data until find the FF D9. Then save this blob data as a JPEG image file.
It should be something like this. You can change the code by replace the serial.write with sd.write.
length = myCAM.read_fifo_length();
Serial.println(length, DEC);
if (length >= 524288) //512 kb
{
Serial.println("Over size.");
return 0;
}
if (length == 0 ) //0 kb
{
Serial.println("Size is 0.");
return 0;
}
myCAM.read_fifo(); //read first dummy data
length--;
while ( length-- )
{
temp_last = temp;
temp = myCAM.read_fifo();
Serial.write(temp);
if (is_header == true)
{
  Serial.write(temp);
}
else if ((temp == 0xD8) & (temp_last == 0xFF))  //match the JPEG header
{
  is_header = true;
  Serial.write(temp_last);
  Serial.write(temp);
}
if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
  break;
delayMicroseconds(12);

}
is_header = false;