ArduCAM/Arduino #81: ArduCAM_4CAM Source Code - Trying to port to Windows UWP for IOT

SJUTMP, 2016-08-30:
Hello, 
I am trying to get the source code for the ArduCAM_4CAM project. I will help me review and port this to IOT Windows 10 UWP project.  The project will read the serial output from the Arduino UNO R3 board four camera system to the Raspberry PI with Windows 10 IOT Core system.

Any insights would be help and will share with all on the progress.

ArduCAM, 2016-08-30:
@SJUTMP, please try this example https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_OV2640_4CAM 
It can work with the host application from https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/host_app/ArduCAM_4CAM.exe

SJUTMP, 2016-08-30:
ArduCam Team,

The ArduCAM_4CAM Executable - Is it possible to get the source code for this application?  I reviewing the Arduino code.  It would be very helpful to see the code for the receiving application to expedite my understanding.  Again thank you in advance for your help.

Sincerely,

Steven Ulbrich