ArduCAM/Arduino #85: corrupt jpeg -- FIFO sends a spurious 0xFF before JPEG signature 0xFFD8

aldobranti, 2016-09-09:
Sometimes the sample code using read_fifo() from Arducam does not construct a readable jpeg. One possibility is a random 0xFF byte before the 0xFFD8 pair that signals start of jpeg until we get to 0xFFD9. There are other pathologies here... for instance #80
In an automated setting going through the files to check for and remove this byte is not practical

the following sample code is offered as a solution for this specific failure...

` i = 0;
  
  //Write first image data to buffer
  temp_last = buf[i++] = temp = myCAM.read_fifo();
  // strip occasional spurious FF lead character
  while ( (temp == 0xFF) & (temp_last == 0xFF)  ) temp = myCAM.read_fifo(); // exit on next ! 0xFF
  buf[i++] = temp;
  //Parse JPEG data from FIFO
  while ( (temp != 0xD9) | (temp_last != 0xFF)  ) {
    temp_last = temp;
    temp = myCAM.read_fifo();
    //Write image data to buffer if not full
    if (i < 256)
      buf[i++] = temp;
    else {
      //Write 256 bytes image data to file
      //Serial.print("*");
      outFile.write(buf, 256); 
      i = 0;
      buf[i++] = temp;
    }

  }

  //Write the remaining bytes in the buffer
  if (i > 0)
    outFile.write(buf, i);
  //Close the file
  outFile.close();
`

aldobranti, 2016-09-09:
here are the opening bytes of a jpeg in a recoverable situation:

`$ od -t x2 00000069.JPG | head -5
0000000      ffff    ffd8    00e0    4a10    4946    0046    0101    0001
0000020      0000    0000    ff00    00db    0043    0304    0403    0303`

and a captured jpeg that is fundamentally corrupt

`$ od -t x2 00000064.JPG | head -5
0000000      698b    7529    ab34    c3ab    0520    0922    731c    7d1c
0000020      9de2    d271    4da2    35f4    7b8c    db91    1aea    87e5`

so @ArduCAM is not out of the woods yet...

closed 2016-09-09 by aldobranti

ArduCAM, 2016-09-09:
@Aldobranti,
Thanks for your workaround. I suggest you to find the correct combination of FF D8 for JPEG start, and read through all the data  until find the FF D9. Then save this blob data as a JPEG image file.
It should be something like this. You can change the code by replace the serial.write with sd.write.

 length = myCAM.read_fifo_length();
  Serial.println(length, DEC);

  if (length >= 524288) //512 kb
  {
    Serial.println("Over size.");
    return 0;
  }
  if (length == 0 ) //0 kb
  {
    Serial.println("Size is 0.");
    return 0;
  }

  myCAM.read_fifo(); //read first dummy data
  length--;
  while ( length-- )
  {
    temp_last = temp;
    temp =  myCAM.read_fifo();
    Serial.write(temp);
    
    if (is_header == true)
    {
      Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))  //match the JPEG header
    {
      is_header = true;
      Serial.write(temp_last);
      Serial.write(temp);
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
      break;
    delayMicroseconds(12);
  }

  is_header = false;

93450101, 2018-07-01:
hello. i want rebuild picture by hex data i recievd.what do i do