ArduCAM/Arduino #88: SPI Error with Arduino Uno, OV5642, 5MP ArduCAM mini

ideapod, 2016-09-24:
Hi there,

I have an Arduino Uno and have connected the 5MP ArduCAM mini Rev B via I2C and SPI pins as per the hardware application note. I have used pin 10 as the CS pin, and the code constant has been set to reflect. I keep having an SPI interface error. 

I have modified the #includes in memorysaver.h to reflect I'm using UNO and 5642. I have defined the contstant to say do the bit rotation. 

The SPI check fails, and I cannot work out why. 

`#if defined(__SAM3X8E__)
  Wire1.begin();
#else
  Wire.begin();
#endif
  Serial.begin(38400);
  Serial.println("ArduCAM Start!"); 

  // set the SPI_CS as an output:
  pinMode(CS, OUTPUT);

  // initialize SPI:
  SPI.begin();
   
  checkSPI();`

check SPI says:

`uint8_t temp;
  //Check if the ArduCAM SPI bus is OK
  Serial.println("testing Arducam SPI");
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  Serial.println(temp);
  if(temp != 0x55)
  {
    Serial.println("SPI1 interface Error!");
    //while(1);
  }
  else
    Serial.println("SPI checks out ok");`

'testing Arducam SPI
255
SPI1 interface Error!
'

Any ideas on what could be going wrong?

ArduCAM, 2016-09-26:
@Mark, You don't need any bit rotation on Arduino UNO R3 board. It seems like you didn't power the board properly. 
Would you please show me the photo how you wiring the hardware, especially the VCC pins.

ideapod, 2016-10-03:
Hi. Thanks for responding. It turns out I'm on a UNO prior to R3. I switched my wiring to use ICMP header for SPI, pin 10 for CS. see http://nanosatrevolution.com/hardware-design/connecting-arduino-to-arducam-mini-5mp/. That seemed to work.

ideapod, 2016-10-24:
I'm still having trouble. i setup on a new machine, got the latest code, ran the latest example. it failed to identify the 5642 chip over i2c:
`#else
   //Check if the camera module type is OV5642
    myCAM.wrSensorReg16_8(0xff, 0x01);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
     if((vid != 0x56) || (pid != 0x42))
     Serial.println("Can't find OV5642 module!");
     else
     Serial.println("OV5642 detected.");
  #endif`

said not detected. i bought a bitscope micro, and got a trace of the i2c signals. see attached. 
[image].

i wondered if the levels were bad. but 4.7 volts seems ok.
[image]

any ideas? nothing working still.

ideapod, 2016-10-24:
heres the voltage levels:
[image]

ArduCAM, 2016-10-25:
@Mark, it seems that the I2C commands are all with NAK. Would you please show me your hardware setup?

ideapod, 2016-10-26:
@ArduCAM thanks very much. Here is the pic. It's a bit confusing, because I've got my bitscope connected via the breadboard and my arducam mini directly into the breadboard. I'll add another file for the equivalent fritzing pic

[image]

ideapod, 2016-10-26:
and this is the equivalent fritzing view: [image]

TobGra, 2016-12-01:
Hey guys,

How's it going?

Ideapod, does your camera work now?

I am quite new to the topic Arduino + ArduCam and could need some help. So probably  you guys can help me. I have a Arduino Uno and a ArduCam Mini-5MP-Plus (OV5642).
Now, I simply want to take a picture with the cam and send it via usb to my PC.
I wired it like your fritzing. Exept that I use pin 10 as CS (see picture)

[image]

Is that fine so far?

Then I load the ArduCAM_Mini_5MP_Plus_LowPowerMode.ino (I downloaded it from here: https://github.com/ArduCAM/Arduino)

I changed the ‘memorysaver.h’ to:
	#define OV5642_MINI_5MP_PLUS

And set pin 10 as CS like this:
	const int CS = 10;

Then I uploaded it to the board.

To see and capture pictures I open the host app ‘ArduCAM_host_v1.0’.

When I press ‘Open’ the LED ‘L’ is blinking on the board 

[image]

and the window of the host app looks like that:

[image]

When I then press ‘Capture’ the LED ‘RX’ is blinking very short and the window of the host app looks like that:

[image]

I never see a picture.

Can you help me with that?

Kind regards,
Tobias