ArduCAM/Arduino #90: Arducam deepSleep mode support.

Bitbanger1900, 2016-09-26:
Hi Lee,

I want to run the Arducam with a solar panel and take time laps
photos. Between taking images i want to set the Arduino Uno Esp8266 into
deep sleep.

What i read is that this should be possible by using for example ESP.deepSleep(sleepTimeS * 1000000); 
To wake up the Arduino (according google) i should connect GPIO16 (D0) with the RST.

The result of this is that the Arduino is going to sleep but after waking up it hangs in a loop by
constantly resetting the Arduino. Also GPIO 16 is the CS of the camera.

So the main question is how to set the Arduino UNO ESP8266 into deep sleep mode using
as less power as possible and how to wake it up.

Kind Regards,
Johan

ArduCAM, 2016-09-26:
@Johan, I'm not sure if the ESP can be set into deep sleep mode. Would you please provide more info.

Bitbanger1900, 2016-09-26:
Hi Lee,

When i read the documentation about the esp8266 on 
http://www.esp8266.com/wiki/doku.php?id=esp8266_power_usage 
there should be a Deep Sleep Mode.

quote:
There is a modification to be made – both in hardware and software – to get current consumption low. On the firmware end, the Espressif SDK has made a systemdeepsleep([uint32t timein_us]) function available, which puts the ESP8266 to sleep for a specified number of microseconds. When it wakes up, it begins running the user program from the very beginning. If you’re using the ESP8266 Arduino IDE, they’ve wrapped that function and another into a very nice ESP.deepSleep([microseconds], [mode]) function.
systemdeep_sleep(0): there is no wake up timer; in order to wakeup, connect a GPIO to pin RST, the chip will wake up by a falling-edge on pin RST

Your provided https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp has a 
class bool ESP8266WiFiGenericClass::forceSleepBegin(uint32 sleepUs) what puts the 
ESP8266 in  wifi_fpm_set_sleep_type(MODEM_SLEEP_T);

Of course it might be possible that this board only supports Modem Sleep  and not the Deep Sleep.

Regards,
Johan

lxxl, 2017-03-23:
Let's back to this question, now i have ArduCam EPS8266 UNO V2 +5MP camera, and after connecting GPIO16 to RST its will dance in the loop...
Without link gpio16 to rst ESP.deepSleep(60000000, WAKE_RF_DEFAULT); will do perfect sleep but then after timelimit will do restart and hang ...

Any ideas ?

Regards
lxxl

Easyrunner, 2019-10-02:
Hello,

I'm interesting in this topic as well.
Any progress here?

Regards

ArducamTeam, 2019-10-03:
@Easyrunner 
Hello,
We are having a National Day holiday. We will test and reply you as soon as we return to the company after the holiday.

jyh-cmd, 2019-10-14:
After testing, Arducam_ESP8266_UNO development board does not support deep sleep mode, but Arducam_ESP8266 _Nano_V2 does.If you want to run our camera in deep sleep mode, I recommend using Arducam ESP32S UNO development board.

ArducamTeam, 2019-10-14:
After testing, Arducam_ESP8266_UNO development board does not support deep sleep mode, but Arducam_ESP8266 _Nano_V2 does.If you want to run our camera in deep sleep mode, I recommend using Arducam ESP32S UNO development board.