ArduCAM/Arduino #92: ArduCAM_ESP8266_V2_OV5642_Capture2SD - Capture Error - While Loop

Joel2905, 2016-09-30:
**Problem:** 
_User Error - Recently purchased  a camera and board off of  Amazon. I am trying to do initial set up, but ran into problems. I probably do not have something set up correctly, but after following installation documentation for loading libraries and board manager files, I still seem to be having issues.
Background Mechanical Engineering - Strong Circuits Background, Limited Programming Experience.
Any Help Would Be Much Appreciated_

**The device returns a soft reset over the serial monitor. Seems to be due to a infinite while loop AND/OR miss defined VID & PID camera values. See Appendix for Serial Monitor Print Out.**

**Suspected Code Trouble:**
"while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK)); "
 
**Hardware:** 
ArduCAM ESP8266 UNO V2
OV5642 Rev B Camera Board

**Code:** 
Using Stock Example -  ArduCAM_ESP8266_V2_OV5642_Capture2SD
memorysaver.h -> un-commented  #define OV5642_CAM_BIT_ROTATION_FIXED
_Note: The un-commenting seems to cause the same header file to be included as the normal OV5642._

**Appendix  - Serial Monitor Print Out**

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
ArduCAM Start!
85  // SPI  Chip Test Serial Print - Passes Check
SD Card detected!
239 = vid  // VID value used in OV5642 Camera Modual Comparison-hex conversion
254 = pid // PID value used in OV5642 Camera Modual Comparison-hex conversion
Can't find OV5642 module!
Start Capture // Is entering into the myCAMSaveToSDFile() function
0 // Serial Printing the Argument of the while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK)); 

Soft WDT reset  // Enters soft reset

ctx: cont 
sp: 3ffefb80 end: 3ffeff00 offset: 01b0

>>>stack>>>
3ffefd30:  3ffeeeb0 00000001 3ffeeeb0 40204db9  
3ffefd40:  00000041 3ffeeae0 3ffeeb4c 402025c8  
3ffefd50:  3ffeeeb0 00000000 00000008 4020260c  
3ffefd60:  3fffdad0 00000000 3ffeeeb0 4020269c  
3ffefd70:  feefeffe 00000000 3ffeeae0 4020237b  
3ffefd80:  feefeffe feefeffe feefeffe feefeffe  
3ffefd90:  3ffeeb4c feefeffe feefeffe feefeffe  
3ffefda0:  feefeffe feefeffe feefeffe feefeffe  
3ffefdb0:  feefeffe feefeffe feefeffe 00003118  
3ffefdc0:  00000011 00000200 3ffeeb4c 3ffeeb4c  
3ffefdd0:  00000002 00000000 00000000 402033b2  
3ffefde0:  00000040 3ffeed90 00000000 3ffeeb00  
3ffefdf0:  00000003 00000000 00000000 3ffeeb00  
3ffefe00:  00000003 00000000 00000000 4020129d  
3ffefe10:  00000003 00000001 00000002 402013ad  
3ffefe20:  0000003c 00000000 00000000 3ffeeb00  
3ffefe30:  00000003 00000000 00000000 4020129d  
3ffefe40:  00000003 00000001 00000002 402013ad  
3ffefe50:  0000003c 3ffeeae0 40205000 3ffeeee0  
3ffefe60:  40101180 0008a399 402326f0 00000055  
3ffefe70:  000000c8 00003801 3ffeeafd 40202cc4  
3ffefe80:  3ffe8528 00000000 000003e8 40202cec  
3ffefe90:  40201500 00000064 00000064 402027b0  
3ffefea0:  00000000 00003818 3ffeeae0 40202948  
3ffefeb0:  003d0900 3ffeeae0 3ffeeeb0 402022dc  
3ffefec0:  feefeffe feefeffe feefeffe 3ffeeed4  
3ffefed0:  3fffdad0 00000000 3ffeeecc 402024f0  
3ffefee0:  3fffdad0 00000000 3ffeeecc 4020504c  
3ffefef0:  feefeffe feefeffe 3ffeeee0 40100718  
<<<stack<<<

ArduCAM, 2016-10-03:
@Joel2905. Sorry to let you know that the ArduCAM for ESP8266 has bugs. The 5MP ArduCAM-Mini firmware is changed while the library is not updated yet.
The new ArduCAM-Mini-5MP doesn't need bit rotation operation for now, so you have to use the OV5642_CAM_BIT_ROTATION_FIXED version.
Please try the attached example again. Let me know if you need more help.

Joel2905, 2016-10-05:
@ArduCAM Lee

I attempted the new code but had the same failure as previously noted. I did notice the IDE has some warnings upon finishing compiling. Would these be considered normal?

_WARNING: library Wire claims to run on [esp8266] architecture(s) and may be incompatible with your current board which runs on [ArduCAM_ESP8266_UNO] architecture(s).
WARNING: library SPI claims to run on [esp8266] architecture(s) and may be incompatible with your current board which runs on [ArduCAM_ESP8266_UNO] architecture(s).

Sketch uses 240,169 bytes (22%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 32,568 bytes (39%) of dynamic memory, leaving 49,352 bytes for local variables. Maximum is 81,920 bytes.
Uploading 244320 bytes from to flash at 0x00000000
..............................................................................................................................................................................................................................................._

Joel2905, 2016-11-24:
@ArduCAM 

Hey Lee,

I was curious if the Library had been updated yet.

Thanks!

Joel

supprot, 2017-01-05:
@Joel2905 

Our library has been updated.
Please delete the old version and download our new library to try again.
 Go to File->Preferences and add the http://www.arducam.com/downloads/ESP8266_UNO/package_ArduCAM_index.json
 to Additional Boards Manager URLs. 

Regards,
ArduCAM support team

Joel2905, 2017-01-06:
@ArduCAM 

Thanks for the update, I just tried to download the new ESP8266_UNO package at the above link. I didn't see any new Arducam library/update available. Should I be able to download the Arducam Library from this link?

Thanks,

Joel

supprot, 2017-01-06:
@Joel2905
Hi,
You need not download the library which has been in the ESP8266_UNO package.
Look at the example.pdf we send to you.
[example.pdf](https://github.com/ArduCAM/Arduino/files/688881/example.pdf)

Joel2905, 2017-01-06:
@ArduCAM 

I reinstalled the basic Arduino software, and installed the ESP8266_UNO Board Package with the provided link. 

I was able to successfully upload the example sketch "ArduCAM_ESP8266_V2_OV5642_Capture2SD."
I have un-commented the following files in my memorysaver.h file.

#define OV5642_MINI_5MP_PLUS
#define OV5642_CAM

After upload, I am using the serial monitor and finding that the camera is failing to detect. "Can't find OV5642"

The program continues to execute, Attempts to "star capture" and then drops into a soft reset.

_" Soft WDT reset

ctx: cont 
sp: 3ffefba0 end: 3ffefef0 offset: 01b0

>>>stack>>>
3ffefd50:  3fffdad0 00000000 3ffeeea0 40202630  
3ffefd60:  4020272d 00000000 3ffeead0 40202333  
3ffefd70:  ffffff01 00000002 00000000 fffffffe  
3ffefd80:  ffffffff 3fffc6fc 00000001 3ffeeadc  
3ffefd90:  00005580 00000002 3ffeead0 00000030  
3ffefda0:  feefeffe feefeffe feefeffe feefeffe  
3ffefdb0:  feefeffe feefeffe feefeffe 00003118  
3ffefdc0:  00000011 00000200 3ffeeb3c 3ffeeb3c  
3ffefdd0:  00000002 00000000 00000000 4020331e  
3ffefde0:  00000040 3ffeed80 00000000 3ffeeaf0  
3ffefdf0:  00000003 00000000 00000000 3ffeeaf0  
3ffefe00:  00000003 00000000 00000000 4020129d  
3ffefe10:  00000003 00000001 00000002 402013ad  
3ffefe20:  0000003c 00000000 00000000 4020129d  
3ffefe30:  00000003 00000001 00000002 3ffeead0  
3ffefe40:  000000ff 0000ffff 3ffeeed0 00000001  
3ffefe50:  0000ffff 3ffeead0 40204e8c 3ffeeed0  
3ffefe60:  00000004 00004407 40231bd8 40204f04  
3ffefe70:  3ffe8520 00000000 000003e8 00000000  
3ffefe80:  3ffe8800 3ffeede4 3ffeeed0 3ffeede4  
3ffefe90:  00000000 3ffeeec4 40204e8c 3ffeeed0  
3ffefea0:  402015c2 000003e8 000003e8 402015b7  
3ffefeb0:  003d0900 3ffeeea0 3ffeead0 3ffeeec4  
3ffefec0:  3fffdad0 00000000 3ffeeebc 40202484  
3ffefed0:  3fffdad0 00000000 3ffeeebc 40204ed8  
3ffefee0:  feefeffe feefeffe 3ffeeed0 40100718  
<<<stack<<< "_

Do you think I have a damaged camera, or is this still related to incorrect settings?

Thanks,

Joel

supprot, 2017-01-07:
@Joel2905 
Hi,
Are you sure you are using OV5642_MINI_5MP_Plus instead of OV5642_MINI_5MP ?
According your phenomenon, I guess maybe you have choose error platform,Please try
 OV5642_MINI_5MP.

In addition, is your SD card detected? I suggest you should format your SD card before using it.

Regards,

ArduCAM support team

Joel2905, 2017-01-07:
@ArduCAM 

I have tried both OV5642_MINI_5MP and the 5MP_Plus, both show the same problem.

The SD card has been formatted and is showing as detected.

Thanks,

Joel

supprot, 2017-01-08:
@Joel2905 

Hi,
Can the camera work fine in other board ?

ArduCAM support team.

Joel2905, 2017-01-08:
@ArduCAM 

I only have one ESP8266-Uno board to test with.  Is there any trouble shooting I could run on the board to see if something is damaged?

Thanks,

Joel

supprot, 2017-01-09:
@Joel2905 
Hi,

Do you use our ArduCAM_ESP8266_UNO borad ?
Please take some pictures for us.
Regards,

ArduCAM support team.

Joel2905, 2017-01-10:
@ArduCAM 

I have attached a series of pictures documenting my set up, file configuration, and error.

[Picture Log.zip](https://github.com/ArduCAM/Arduino/files/695014/Picture.Log.zip)

Let me know if you have any difficulties with the files or need them in a different format. (currently are .jpg)

Thanks,

Joel

Joel2905, 2017-01-17:
Just wanted to check in and see if anyone has had a chance to review my pictures. 

Thanks!

Joel

supprot, 2017-01-17:
@Joel2905 

Hi,
The new ArduCAM-Mini-5MP doesn't need bit rotation operation for now, so you have to use the OV5642_CAM_BIT_ROTATION_FIXED version.
you should #define OV5642_MINI_5MP_BIT_ROTATION_FIXED in the memorysaver.h
If the phenomenon is still exist, you can print the temp value in the SPI test code to debug it. Just add "Serial.println(temp,HEX);  behind  the "
myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
temp = myCAM.read_reg(ARDUCHIP_TEST1);"codes.

Regards,
ArduCAM support team.

Joel2905, 2017-01-21:
ArduCAM support,

I have added the serial print as you suggested. The serial monitor prints 0 & 55.

I don't believe it is the SPI values that are failing. The SPI check passes just fine.

My problem, I believe is during the VID and PID value comparison. In addition to those values failing to match up to the !IF statements, the system drops into a soft rest and dumps a large value of hex files. I don't know what may be causing the VID / PID values to be set incorrectly.

Thanks,

Joel

supprot, 2017-01-21:
@Joel2905 
Hi,
What's the value of the VID and PID you have read?

Regards,
ArduCAM support team.

Joel2905, 2017-01-21:
@ArduCAM 

My serial monitor results show:
vid as: 239    (compares it to 86)
pid as: 254   (compares it to 66)

Are these values not hard-flashed onto the chip for identification?

Thanks!

Joel

supprot, 2017-01-22:
@Joel2905 
Hi,
Would you plug the camrea directly to the ESP8266_UNO instead of dupoint line.

Regards,
ArduCAM support team.

Joel2905, 2017-01-23:
@ArduCAM 

That worked perfectly. Thank you for all your support over the past few weeks. You have one happy customer!

[image]

Best Regards,

Joel

supprot, 2017-01-23:
@Joel2905, 
Congratulations!!!
Would you please share your info about how do you over come the issues?

Joel2905, 2017-01-23:
The  connection between the camera and the board was bad. By removing the dupont line and plugging in directly the board, it works great. 

Thanks again, sorry for all the difficulty!

Joel

supprot, 2017-01-23:
@Joel2905,
Thanks for your feedback.
Our camera and ESP8266 UNO board are designed to be plug and play.
After all it finally worked by the simplest solution of removing the bad dupont cable.
Let us know whenever you need further help.

JothaBohemian, 2017-08-14:
Hi all,
although I'm using the ov2640 sensore (ArduCam 2MP Rev.B) I got the same values for pid and vid. 
Plugging the arducam with same jumpers into Arduino UNO R3 will work fine.

- Your solution by using esp-UNO library finally didn't work with nodeMCU1.0(ESP8266-12E). (AVR error: wrong platform).
- I also used all available board file versions for esp8266 at the board manager.
- Trying to set if ((vid != 0xef ) && (( pid != 0xfe ) instead of 0x26 and 0x41 gives no frame and soft reset