ArduCAM/Arduino #94: ArduCAM_OV5642_Digital_Camera - Invalid JPG

huguesbr, 2016-10-06:
I'm trying to run `ArduCAM_OV5642_Digital_Camera` exemple.

The video feedback loop work correctly (I can see the video on the LCD screen), but when I try to capture an image, the captured image seem to be in an invalid format. See attached JPG file.
[image]

Any idea?

Shdwdrgn, 2016-11-07:
See issue #21, I had a similar problem until I changed bufferSize to 2048.

huguesbr, 2016-11-07:
@Shdwdrgn it was due to a different serial port speed between Arduino App / Device :(

Thanks

closed 2016-11-07 by huguesbr

Shdwdrgn, 2016-11-07:
Ah, sounds easy enough, glad you got it figured out.

huguesbr, 2016-11-07:
@Shdwdrgn a bit ashamed 😳...