ArduCAM/Arduino #97: ArduCAM Rev C+ OV7670 and OV7675 error

rsprat, 2016-10-11:
Hello,
I am using ArduCAM Rev C+ in conjunction with Arduino UNO R3 for testing several image sensors from OmniVision (OV7675, OV7670, OV2640).

I have uploaded the corresponding ArduCAM_XXXXXX_Camera_Playback sketch of the latest version of ArduCAM library and tested the setup with different results:

-With OV7675 and OV7670 stores corrupted BMP files to the SD card (see attached image).
-With OV2640 stores valid QVGA images to the SD.
The sensor boards and ArduCAM rev C+ have been purchased at one of your european distributors (botland).
Any ideas of what I am doing wrong? Anything to check?

[image]
(Note that the file has been converted to PNG in order to upload it to GitHub)

Thank you

ArduCAM, 2016-10-13:
@rsprat, Sorry for my late reply.
Did you enable the OV7675 or OV7670 camera module in the memorysaver.h file? If not the camera will not initialize properly.

rsprat, 2016-10-17:
Thanks! That was the issue!

closed 2016-10-17 by rsprat