ArduCAM/Arduino #101: Arducam-Mini-2MP + NodeMCU ESP8266 Data Loss

BuZLuCaY, 2016-11-04:
My connections like [this link](http://www.arducam.com/tag/arducam-esp8266/)
I've tried ArduCAM_ESP8266_OV2640_Capture Example code. But I have data loss [like this](https://postimg.org/image/5hbjxuhv3/)
What can I do?

ArduCAM, 2016-11-04:
It is the know issue with default ESP8266 board support files. Please use our modified libraries for ESP8266, follow this guide http://www.arducam.com/downloads/ESP8266_UNO/ArduCAM_ESP8266_UNO_DS.pdf how to download the board support for ESP8266.

BuZLuCaY, 2016-11-05:
I solved data loss problem with [this examle](https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/tree/master/libraries/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_OV2640_Capture). [This example](https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_OV2640_Capture) isn't true.

torntrousers, 2017-01-22:
I've had the same problem using the standard ESP8266/Arduino support, but it works when doing as described in that ArduCAM_ESP8266_UNO_DS.pdf using the custom boards manager image. Thats not very convenient though and isn't up to date with the latest ESP8266/Arduino code on github. What is it thats done in the custom boards manager image and can we modify the latest ESP8266/Arduino code ourselves to do the same? Or better yet. get the changes made in the ESP8266/Arduino github so it just works out-of-the-box?

Belisarith, 2017-06-23:
I've got the same problem. What makes this issue additionally annoying is that there is no way to use arducam together with more advanced ides like platform.io. 
What exactly are the changes done to the standard library of esp8266? Couldn't this code changes somehow be put into the arducam library itself? Thereby, you could use other ides as well by just including the modified arducam library. Quite annoying. Furtheremore, there are problems if you want to mix arducam with libraries for mqtt like homie.