ArduCAM/Arduino #102: The image cannot be displayed because it contains errors

holcar, 2016-11-06:
### Basic Infos

#### Hardware
Hardware:			ESP8266-12E
Core Version:      	2.2.0
Board: ArduCAM ESP8266 UNO board V2
[image]

Camera module: OV5642 ArduCAM-Mini-5MP rev.B

### Description
When I request image from browser with url: http://192.168.89.115/capture I get corrupted image file from my ESP using OV5642 camera. Everything else is working OK: Camera is detected, when I visit url it says server is running. I use an example ArduCAM_ESP8266_OV5642_Capture and I cant get it work. I just changed the SSID and password and set it to client mode. I also tried camera OV2640 from my kit "ArduCAM-Mini-2MP-V2 and ArduCAM-ESP8266-Nano" but same problem with corrupted image appears again.

Is there problem with ArduCAM ESP8266 UNO board V2 board? Reset button doesn't work either. Is this normal?
[image]

Corrupted image file:
[image]

### Settings in IDE

Module:  			ArduCAM ESP8266 UNO board V2

### Sketch

```cpp
// ArduCAM Mini demo (C)2016 Lee
// web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM ESP8266 5MP camera.
// This demo was made for ArduCAM ESP8266 OV5642 5MP Camera.
// It can take photo and send to the Web.
// It can take photo continuously as video streaming and send to the Web.
// The demo sketch will do the following tasks:
// 1. Set the camera to JEPG output mode.
// 2. if server.on("/capture", HTTP_GET, serverCapture),it can take photo and send to the Web.
// 3.if server.on("/stream", HTTP_GET, serverStream),it can take photo continuously as video 
//streaming and send to the Web.

// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM ESP8266 5MP camera
// and use Arduino IDE 1.5.8 compiler or above
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

//This demo can only work on OV5642_MINI_5MP or OV5642_MINI_5MP_BIT_ROTATION_FIXED
//or OV5640_MINI_5MP_PLUS or ARDUCAM_SHIELD_V2 platform.
#if !(defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) || defined (OV5642_MINI_5MP_PLUS) ||(defined (ARDUCAM_SHIELD_V2) && defined (OV5642_CAM)))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif

// set GPIO16 as the slave select :
const int CS = 16;

//you can change the value of wifiType to select Station or AP mode.
//Default is AP mode.
int wifiType = 0; // 0:Station  1:AP

//AP mode configuration
//Default is arducam_esp8266.If you want,you can change the AP_aaid  to your favorite name
const char *AP_ssid = "arducam_esp8266"; 
[... 235 lines cut ...]

Sketch uses 258,518 bytes (24%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 44,352 bytes (54%) of dynamic memory, leaving 37,568 bytes for local variables. Maximum is 81,920 bytes.
Uploading 262672 bytes from to flash at 0x00000000
..........................................................
```

### Debug Messages from console

```
ArduCAM Start!
OV5642 detected.

Connecting to MyWifi
WiFi connected

192.168.89.115
Server started
CAM Capturing
capture total_time used (in miliseconds):145
CAM Capture Done!
send total_time used (in miliseconds):384
CAM send Done!

```

ArduCAM, 2016-11-07:
@holcar, from your attached image file, it seems that the data is not correct and problem comes from the camera hardware.
Please contact the seller for replacement.

gordolob, 2019-06-09:
Andrej, did you fix this issue replacing the camera?
or was it in the sketch?

AndrejNastran, 2019-06-11:
@gordolob as far as I know first one didn't work, and a replacment didn't either.  I don't know the cause. I had more luck with ArduCAM-mini. It works great.
I would look for camera with ESP32. Is way more powerfull :)