ArduCAM/Arduino #104: Shield V2 + Arduino UNO/MEGA + OV2640: Image not being saved

frank0911, 2016-11-09:
Hello, I recently purchased the shield v2 and I am trying to set it up and up to know I cannot see the images saved on the SD.

I tried using an Arduino UNO or Arduino MEGA R3 with the following examples: 

ArduCAM_Digital_Camera
ArduCAM_Capture2SD

And I do get the OV2460 detected and SD card detected. Also, when I press the shutter button, I do get the messages 'capture time' and 'total time used'.

However when I look in the folder, I do see the files however I am unable to open them. 

Can you please help?

frank0911, 2016-11-09:
Hello,

Moving further into this, I actually found that a dummy header is being placed before the JPEG header. How would this be removed from the code?

ArduCAM, 2016-11-13:
@frank0911, sorry for my late response. Have you download the new library for the Shield V2 support? Actually with the new library you don't have the dummy byte at the beginning.

frank0911, 2016-11-14:
Hi Lee, Thank you for the reply.

I did download the latest library and I even downloaded just now to make sure I did no have an older version. The issue is that when I took the picture, the jpeg file had two extra dummy bytes on it. If I removed the bytes manually, I was able to open the image. 

What I did, was to uncomment the following line from the example:

  //myCAM.read_fifo();

This removed one of the dummy bytes but it still left me with a second one so I placed a second line of the exact code and now I can open the images without any problem.

Just wanted to share in case someone else has this issue.

ArduCAM, 2016-11-15:
@frank0911, thanks for your report, we will look into is bug.

closed 2016-11-22 by frank0911