ArduCAM/Arduino #105: How to select "hardware platform" ?

Nick223, 2016-11-12:
Arduino:1.6.6 (Windows 10), Board:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"
 #error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
  ^
exit status 1
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file

How do I do this ? I've already changed definitions of the camera. What about the board ?

ArduCAM, 2016-11-13:
@Nick223, we update our library recently. Choose the hardware platform depends on which one you are using.

//There are two steps you need to modify in this file before normal compilation
//Only ArduCAM Shield series platform need to select camera module, ArduCAM-Mini series platform doesn't

//Step 1: select the hardware platform, only one at a time
//#define OV2640_MINI_2MP
//#define OV5642_MINI_5MP
//#define OV5642_MINI_5MP_BIT_ROTATION_FIXED
//#define OV5642_MINI_5MP_PLUS
//#define OV5640_MINI_5MP_PLUS

//#define ARDUCAM_SHIELD_REVC 
#define ARDUCAM_SHIELD_V2

//Step 2: Select one of the camera module, only one at a time
#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))
//#define OV7660_CAM
//#define OV7725_CAM
//#define OV7670_CAM
//#define OV7675_CAM
    //#define OV2640_CAM
//#define OV3640_CAM
 #define OV5642_CAM
//#define OV5640_CAM

//#define MT9D111A_CAM
//#define MT9D111B_CAM
//#define MT9M112_CAM
//#define MT9V111_CAM 
//#define MT9M001_CAM 
//#define MT9T112_CAM
//#define MT9D112_CAM
#endif

Nick223, 2016-11-13:
I TOLD YOU I SELECTED CAMERA

Nick223, 2016-11-13:
#ifndef _MEMORYSAVER_
#define _MEMORYSAVER_

//There are two steps you need to modify in this file before normal compilation
//Only ArduCAM Shield series platform need to select camera module, ArduCAM-Mini series platform doesn't

//Step 1: select the hardware platform, only one at a time
//#define OV2640_MINI_2MP
//#define OV5642_MINI_5MP
//#define OV5642_MINI_5MP_BIT_ROTATION_FIXED
//#define OV5642_MINI_5MP_PLUS
#define OV5640_MINI_5MP_PLUS

//#define ARDUCAM_SHIELD_REVC	
//#define ARDUCAM_SHIELD_V2
#define ESP8266

//Step 2: Select one of the camera module, only one at a time
#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))
	//#define OV7660_CAM
	//#define OV7725_CAM
	//#define OV7670_CAM
	//#define OV7675_CAM
    //#define OV2640_CAM
	//#define OV3640_CAM
	//#define OV5642_CAM
	#define OV5640_CAM
	
	//#define MT9D111_CAM
	//#define MT9M112_CAM
	//#define MT9V111_CAM	
	//#define MT9M001_CAM	
	//#define MT9T112_CAM
	//#define MT9D112_CAM
#endif 
#endif	//_MEMORYSAVER_

ArduCAM, 2016-11-13:
@Nick223, you don't need to define ESP8266, because it can be selected automatically by the board manager in Arduino IDE.
The Step 2 is only for ArduCAM SHield RevC/RevC+ and Shield V2 board, you don't need to define anything for OV5640_MINI_5MP_PLUS.

Nick223, 2016-11-13:
"you don't need to define ESP8266, because it can be selected automatically by the board manager in     Arduino IDE."

I know, I know. Then tell me why isn't it working ?

ArduCAM, 2016-11-14:
@Nick223, after you correct the hardware platform setting, what exactly the error message you have that it doesn't work. I have to know your detail problem, then I can help you resolve your problem.

MHz000, 2017-01-06:
I'm too struggeling with the same error message! 
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
I see no further comment from ArduCAM since 14 Nov 2016. That's quite a long time. Pls send a complete and working memsaver.h for OV5642 camera and a generic ESP8266 board.
Maybe Nick223 has a solution and can post it.
Thank you!

dprophet, 2017-01-06:
You need to add a the library to your sketch. Then find the installed memorysaver.h and edit it.

On Mac/Linux do
find /. -type f -name memorysaver.h 

On Windows open a CMD prompt and run
cd \
dir memorysaver.h /s

And edit the memorysaver.h file.

> I see no further comment from ArduCAM since 14 Nov 2016. 
In all fairness this isnt ArduCAM support problem. This is a lack of knowledge of C/C++ compilers and Arduino environment. You could #define what you need directly in your sketch.

MHz000, 2017-01-06:
Erik,
thank you for your fast response. Due to my verry limited knowledge of C++,
Arduino IDE etc I'm still confused.
In my sketch - the unchanged original one that comes with the ArduCAM/ESP
lib - is an #include "memorysaver.h" at line no 24. Also in my additional
lib folder is an ArduCAM folder containing the file memorysaver.h. This one
I've edited, may be the wrong way. Now I would love to get the rigth one
for compiling the sketch for the OV5642 camera and an ESP8266 native board
from you.
Sorry for my dullness but what for do I have to open a CMD promt change to
memorysaver.h /s as I already have canged the file and stored it? Sorry for
that question but I really like to bring up the camera to live.
Thank you!

supprot, 2017-01-07:
@MHz000 

Hi,
If you are in ArduCAM_ESP8266_UNO board, you can go to file ->preference  at the Arduino IDE to quickly find the memorysavery.h for the ArduCAM_ESP8266_UNO board, look at the picture we send.The memorysaver.h file is at the
..\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.1\libraries\ArduCAM path.
[image]

Nick223, 2017-01-07:
No, I don't have any solution yet. Tell me "experts", I changed the memorysaver.h definitions to my camera but you want me to write into my board a sketch compiled for a different board... Why ? My ESP8266 isn't ArduCAM board... Doesn't pinout change ?

MHz000, 2017-01-07:
Many thanks for your hint. However:

1st   in arduino preferences .. there is no ArduCAM folder. See the attaced
screenshot

2nd  as I said before: I'm not using an ArduCAM_ESP8266_UNO board. My
hardware is a "Generic ESP8266 module" as shown in one of your tutorials =>
see
        attached screenshot!

Pls guide me step by step throug the installation of an OV5642 camera in a
WIN10 arduino IDE envieroment for a generic ESP8266 board and the
compilation of the ArduCAM_ESP8266_OV5642_Capture sketch.

To underline the keywords in my request:

*- OV5642 camera*

*- generic ESP8266 board*
*- ArduCAM_ESP8266_OC5642_Capture sketch*

Thank you for patience and support so much!

supprot, 2017-01-07:
@MHz000 

Hi,
You can download our ArduCAM_ESP8266_UNO and ArduCAM library firstly,
If you are using generic ESP8266 board,you can put our ArduCAM library into 
the ..\packages\esp8266\hardware\esp8266\2.3.0\libraries path.

Please follow the steps below to download our ArduCAM_ESP8266_UNO and ArduCAM library：
1、Go to File->Preferences and add the http://www.arducam.com/downloads/ESP8266_UNO/package_ArduCAM_index.json
to Additional Boards Manager URLs. Then click OK.
2、Go to Tools->Board:”xxxx”->Boards Manager…
   Find the ArduCAM_ESP8266_UNO by ArduCAM version 2.2.1 board and install it.

After installing our board. you will find our ArduCAM library in the ..\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.1\libraries folder.

If you use generic ESP8266 board, you can copy the ArduCAM library to  ..\packages\esp8266\hardware\esp8266\2.3.0\libraries  path. The you can
 find our demos about ESP8266.

Regards,
ArduCAM support team.

MHz000, 2017-01-07:
> downloads/ESP8266_UNO/package_ArduCAM_index.json

Hi ArduCAM support team.
Step 1 and step 2 so fare ok! Boardmanager now shows ArduCAM_ESP8266_Uno
and I'm able to select it.   :-)

Because of my limited knowledge of Arduinos IDE I did't find
 ..\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.1\libraries
folder
In C: programme(x86)\arduino there is no packages folder. Where do I have
to look for \packages\esp8266\...

Thank you for your advice
Regards Jochen

MHz000, 2017-01-07:
Hi Nici223,
I got the following from ArduCAM support:
-------------------schnipp-----------------------------

Hi,
You can download our ArduCAM_ESP8266_UNO and ArduCAM library firstly,
If you are using generic ESP8266 board,you can put our ArduCAM library into
the ..\packages\esp8266\hardware\esp8266\2.3.0\libraries path.

Please follow the steps below to download our ArduCAM_ESP8266_UNO and
ArduCAM library：
1、Go to File->Preferences and add the http://www.arducam.com/
downloads/ESP8266_UNO/package_ArduCAM_index.json
to Additional Boards Manager URLs. Then click OK.
2、Go to Tools->Board:”xxxx”->Boards Manager…
Find the ArduCAM_ESP8266_UNO by ArduCAM version 2.2.1 board and install it.

After installing our board. you will find our ArduCAM library in the
..\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.1\libraries
folder.

If you use generic ESP8266 board, you can copy the ArduCAM library to
..\packages\esp8266\hardware\esp8266\2.3.0\libraries path. The you can
find our demos about ESP8266.

------------------------schnapp------------------------

Step 1 and 2 works fine, boardmanager now shows the additional board.

However I do not know wher to find ...packages\esp8266\hardware ... etc to
put the ArduCAM lib in. I'm using an ordinary WIN10 system with no special
settings for arduino installation.

Thank you for any hint

Regards Jochen

supprot, 2017-01-08:
@MHz000 
Hi,
We have already told you how to find the ..\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.1\libraries
folder.
Please refer to the link:
https://cloud.githubusercontent.com/assets/24818249/21737428/778b1afe-d4b3-11e6-87b3-827d877f6658.png

Regards,

ArduCAM support team.

dprophet, 2017-01-09:
@MHz000 I have been talking with Manufacturer of the Arducam hardware. I have been a software engineer for 30 years now. Yes, I struggled with my 1st Arducam IoT hardware project.

Your problem has nothing to do with Arducam hardware. This is purely a software environment issue. For me this is a simple problem to be solved over a TeamView session.

Disclaimer: I do not work at of for Arducam - Perhaps you should consider a premium support issue to get your first camera capture working.

Once the first capture works its easier to get the more advanced usages of the camera & hardware working.

I can tell you I have Arcucam working and integrated with
1) Arducam ESP8266 UNO Board + Arducam Mini OV2640 + SD storage and WIFI streaming
2) Arduino UNO R3 + Arducam Multi Cameras Adapter Board + Arducam Mini OV2640 + WIFI streaming
3) Seeeduino Cloud + Linux OpenWrt + Arducam Mini OV2640 + SD storage + IBM Watson Cloud integration
4) Linkit smart 7688 duo + breakout board + Linux OpenWrt + Arducam Mini OV2640 + SD storage + Environmental Sensors + IBM Watson Cloud integration + Machine Learning

My software development environments are Windows 10 and Apple Mac. @MHz000 This is where your problem is, setting up your software development environment.

Consider paying for 1 premium support issue to achieve your 1st success. It all comes easier after that.

rustya0373, 2020-09-03:
Ok well I see that there has not been any activity on this issue since 2017. I am still having this issue and I have read every forum/blog about this issue and NOBODY has actually recommended a solution. Please don't reply with simply choose your selection in the memorysaver.h file by commenting or uncommenting the camera that you are using. I have done that many times with no success. I am using the OV2640 with an Arducam ESP8266 uno and I have followed every video for setting it up EXACTLY the way that they did it but still the same error. It is as if the memorysaver.h file is not being included so it doesn't know what camera you have selected there for it gives you the error message...#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file. 

Can someone help with this issue with something new or if you got it to work please explain how. I have read here that someone has finally got it to work but refuses to share how!!!!

ArducamSupport, 2020-09-04:
@rustya0373 
Hi,
Please contact support@arducam.com to set up a time to use TeamViewer to help you remotely.Our working hours are from 8.30 am to 5.30 PM Beijing time.

rickycly49, 2020-09-21:
Hello!
I followed every specific details in these comments to get it working but i am still not able to. Still getting a compiler error! Board is sellected for uno.

keithw421, 2020-12-04:
I commented out the error message  i.e. //#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file. ' in the ino im now working an error at the end