ArduCAM/Arduino #107: LCD + CaptureSD

frank0911, 2016-11-22:
Hello Lee,

I am trying to combine the two codes for capture by SD and have it display on the LCD however I am not having a lot of success. Can you please help me with this code?

Thank you!

ArduCAM, 2016-11-23:
@frank0911, you can find ready to use example from : https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/Shield_V2/ArduCAM_Camera_Playback/ArduCAM_Camera_Playback.ino

niluxncsu, 2017-09-16:
Hello, can I display image only on half screen and remaining half screen used for buttons for LED on/OFF. 
Please let me know if you have any suggestions. 
Thanks

supprot, 2017-09-17:
@niluxncsu 
Hi,
You can refer to this code https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/Shield_V2/ArduCAM_Shield_V2_Touch/ArduCAM_Shield_V2_Touch.ino