ArduCAM/Arduino #108: extern triggered multi capture for ArduCAM mini 2MP

gutcZHAW, 2016-11-25:
For my application it's necessary to capture a few images (three or four) in a small time slot.
Since one frame (1600x1200 JPEG) is around 80kB the memory space on the framebuffer should reach for four frames (384kB on the AruduCAM mini 2MP). 

My prefered concept is to trigger the ArduCAM via SPI command up to four times and then read out the framebuffer and store the data on the SD (see the following pseudo code).

```
myCAM1.flush_fifo();                 // flush framebuffer
frame_cnt = 0;                       // reset frame counter
while((frame_cnt < 4) || stop_flag)  // capture up to 4 frames
{
        if(trigger_flag)             // trigger_flag set by interrupt routine
        {
              myCAM1.CS_LOW();
              myCAM1.clear_fifo_flag();
              myCAM1.start_capture();
              trigger_flag = false;
              frame_cnt++;
              while(!myCAM1.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));  // wait till image is buffered
        }
}
for(uint8_t i = 0; i<frame_cnt)
{
       //read till 0xFFD9 and store the data on the sd card
}
```

Unfortunately it seems to overwrite the already buffered frames so that only the last captured frame is stored. Is the Buffer writepointer reseted by the command myCAM1.clear_fifo_flag() or myCAM1.start_capture()?

Also i wondering why the fifo size (myCAM1.read_fifo_length()) isn't equal zero after myCAM1.flush_fifo(). When i look in the ArduCAM library it just reseted the fifo write done flag but also when I add lines to reset fifo read and write pointer the result is still the same.

```
//Reset the FIFO pointer to ZERO
void ArduCAM::flush_fifo(void)
{
  write_reg(ARDUCHIP_FIFO, FIFO_CLEAR_MASK);
  write_reg(ARDUCHIP_FIFO, FIFO_WRPTR_RST_MASK);		// added by gutc
  write_reg(ARDUCHIP_FIFO, FIFO_RDPTR_RST_MASK);		// added by gutc
}
```

There is another vagueness which hasn't an influence to my problem.
In the [Hardware Application note](http://www.arducam.com/downloads/shields/ArduCAM_Mini_5MP_Camera_Shield_Hardware_Application_Note.pdf)  on page 5 there is the following written:

_FIFO control Register
Bit[0]: write ‘1’ to clear FIFO write done flag
Bit[1]: write ‘1’ to start capture
Bit[4]: write ‘1’ to reset FIFO write pointer
Bit[5]: write ‘1’ to reset FIFO read pointer_ 

But in the ArduCam header file there is the following:

```
#define FIFO_RDPTR_RST_MASK     0x10		
#define FIFO_WRPTR_RST_MASK     0x20
```

So these both bits were switched. Which one is correct?

Thank you all for your help :)

gutcZHAW, 2016-12-16:
If there is someone else who have such problems I'll explain my workaround:

First of all I don't know if it's possible to capture multiple frames with the ArduCam mini 2MP.
In the issue #87 some claims it's possible and other says it's not possible. I've never able to capture multiple frames with the ArduCAM mini 2MP. Even if I controlled the Pins `reset read pointer` and `reset write pointer` by the MCU it didn't work stable (But it was possible to store more than one frame in the FIFO).

So I chaged the hardware to the ArduCam-F Shield V2 which has the following advantages:

- flexible change of the camera

- 8MByte FIFO (instead of 384kB)

- on board micro SD-Card slot

- nearly the same power consumption

With this hardware it's possible to capture several frames. But it's still not possible to trigger every particular frame. 
According to the Datasheet of the OV2640 there is a real snapshot mode where a snapshot can be triggered by the pin FREX. Unfortunately this Pin is connected to ground. So I found a real software workaround:

By set the Number of Frames in the Capture Control Register and the command `startCapture()` the ArduCam Shield captures the next NrF (Number of Frames) frames which are output by  the OV2640.
The OV2640 normaly works in the video mode so it displays one frame after the other. During the snapshot mode, the OV2640 doesn't output any frames. So I cofigured the OV2640 for the snapshot mode and the ArduCam Shield for the multiple capture mode. By switch the OV2640 for one frame to the video mode it's possible to control when a frame should be captured and stored in the FIFO.
The following command set's the OV2640 to the video mode and back to the snapshot mode after one frame:
```
// select register bank 1
myCAM.wrSensorReg8_8(0xff, 0x01);

// enable video mode (Register COM3)
myCAM.rdSensorReg8_8(0x0C, &temp);
temp = temp & (~0x01);
myCAM.wrSensorReg8_8(0x0C, temp);

// wait till the OV2640 starts to output the frame
myCAM.CS_LOW();		
while(!myCAM.get_bit(ARDUCHIP_TRIG , VSYNC_MASK));
myCAM.CS_HIGH();

// disable video mode (Register COM3)
myCAM.rdSensorReg8_8(0x0C, &temp);
temp = temp | 0x01;
myCAM.wrSensorReg8_8(0x0C, temp);
```
 This works fine but has one disadvantage. The automatic image control functions such as automatic gain control and automatic exposure control does't work during snapshot mode and the first frames of the video mode are nearly black. So it's necessary to control these values manually.

Besides there is a jitter on the trigger delay between 0ms and 1/15fps (dependent on the frame rate) because the OV2640 seems to capture frames continuous independent of the mode. By switch to the video mode the OV2640 just output the next ready frame. This jitter occurs also in the normal capture mode of the ArduCAM and I suppose it will also occur in the real snapshot mode of the OV2640 (by triggering via pin FREX).

I hope i can help someone with this workaround.
Feel free to ask if something isn't clear.

cheers :)

closed 2016-12-16 by gutcZHAW

ArduCAM, 2016-12-20:
@gutcZHAW, thanks for your awesome workaround for defined multiple frame capture procedure. In our multi-frame capture example only capures continuous frames, your workaround can capture any frames with any time delay, it is a great improvement.