ArduCAM/Arduino #110: corrupt jpeg files

msach281, 2016-11-27:
I have a Shield_V2 VF with a OV5642 camera connected to a Arduino UNO, I am using the example sketch ArduCAM_Digital_Camera.ino
Everything works ok, but when I try to open the .jpg files I get a message "File damaged, or not doesn’t recognize". 
I have "#define ARDUCAM_SHIELD_V2", and "#define OV5642_CAM" in the memorysaver.h file.
I am not sure what I am doing wrong.

I am also seeing error messages with the Arducam Library that I just downloaded. 

/Arduino/libraries/ArduCAM/ArduCAM.h:104:0: warning: "pulse_high" redefined
 #define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);

/Arduino/libraries/UTFT4ArduCAM_SPI/HW_AVR_SPI_defines.h:5:0: note: this is the location of the previous definition
 #define pulse_high(reg, bitmask) sbi(reg, bitmask); UTFT_cbi(reg, bitmask);
 ^
/Arduino/libraries/ArduCAM/ArduCAM.h:105:0: warning: "pulse_low" redefined
 #define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);

/Arduino/libraries/UTFT4ArduCAM_SPI/HW_AVR_SPI_defines.h:6:0: note: this is the location of the previous definition
 #define pulse_low(reg, bitmask) UTFT_cbi(reg, bitmask); sbi(reg, bitmask);
 ^

msach281, 2016-11-27:
I did notice the old post #104 that had the same issue, and was able to fix it by 
"uncomment the following line from the example: //myCAM.read_fifo(); and placed a second line of the exact code"
But I did not have the same luck with the OV5642 camera.

ArduCAM, 2016-11-28:
@msach281, would you please upload your corrupt JPEG files? we will examine what's the problem and get back to you.

msach281, 2016-11-28:
This is was created from the sketch example ArduCAM_Digital_Camera.ino.
[image]

ArduCAM, 2016-11-28:
@msach281, BTW: did you download the latest library?

ArduCAM, 2016-11-28:
There are two dummy bytes at the beginning of  your JPEG file, I have removed these, and you can see you photo below
[image]

msach281, 2016-11-28:
Yes, I downloaded the library and installed it again this morning.
How do I see the dummy bytes on the JPEG files.
How do I prevent the dummy bytes from being saved at the beginning.

ArduCAM, 2016-11-28:
@msach281, You can use hex editor like UltraEdit. 
In order to prevent the dummy bytes, you can use myCAM.read_fifo(); to remove one byte dummy at a time.

msach281, 2016-11-28:
It is working now.
I did try adding myCAM.read_fifo() twice to the sketch yesterday and it didn't work, but today it did.
I must have done something else wrong.
Thanks for your help.

closed 2016-11-28 by msach281