ArduCAM/Arduino #111: Distorted Image

trvlr737, 2016-11-30:
I am new to the ESP8266/ADRUCAM so, I apologize in advance. 
The problem I'm having occurs when I change the size of the image. The image is perfect at a resolution of 160x120. When I try a different resolution, the image doesn't capture/transfer properly. Please, any idea's would be a great help.

[image]
160x120

[image]
640x480

trvlr737, 2016-11-30:
Sorry. I should have added a little more info. 
This is the sketch. I haven't modified it in anyway.

https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_OV2640_Capture/ArduCAM_ESP8266_OV2640_Capture.ino

Hardware: I'm using a LoLin NodeMcu V3 ESP8266 module and a ArduCam Mini 2MP.

ArduCAM, 2016-11-30:
@trvlr737, It is the know issue. The ESP8266 library should be modified to work with our camera. Please use our ESP8266 library package.
http://www.arducam.com/downloads/ESP8266_UNO/ArduCAM_ESP8266_UNO_DS.pdf 
http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/

trvlr737, 2016-11-30:
THANK YOU!! Works perfect!!

closed 2016-11-30 by trvlr737

NVergunst, 2017-04-07:
What is the actual problem that is solved by the modified library? I am not using an Arduino at all, but my picture comes out like yours. I'd really appreciate knowing which magic software settings need to be changed at the I2C/SPI level to get working data...