ArduCAM/Arduino #112: Problem streaming

trvlr737, 2016-12-02:
This is a continuation of Problem #111 "Distorted Images". Your reply worked wonderfully for a single capture. However, when I try to stream, my browser asks me to "Save, Save as, Cancel". If I choose to "Save" or "Save as" a .txt file of 0 length is created and the computer appears to do nothing (I waited 5 minutes then aborted the process). 

I also tried the "Video.html" file located in /AdruCam/examples/ESP8266/AdruCAM_ESP8266_OV2640_Capture/html folder. Nothing was displayed except to image of the UNO board in the selected resolution.

Any help would be greatly appreciated. 
Thank You

ArduCAM, 2016-12-05:
@trvlr737 You can only open the Video.html or index.html at a time, only one connection is allowed.
You can also use host application to connect to the ESP8266 via IP address.
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app/ArduCAM_Host_V2.0_Windows

trvlr737, 2016-12-05:
Sorry. I don't think I made myself clear. Here is a screen shot of what happens when I try to stream. Nothing appears. I connect to the Adrucam_esp8266, then type in the IP address of the ESP8266 (in my case it's 192.168.4.1/stream) and this is a screen shot of what I get. If I enter "192.168.4.1/capture" a single picture appears perfectly. Any ideas? Thank You.

[image]

trvlr737, 2016-12-05:
Here's some more information. I just tried streaming to my Android Phone, and it worked perfectly. I'm using Windows 10 with Edge on my laptop, and I get the blank image shown in my last post. Could Edge be the problem?  If it is, what is the fix? Hope this helps.

Also, I've tried Internet Explorer 11 and get the same blank screen as my last post.
  
Thank You Once Again