ArduCAM/Arduino #115: jpeg corrupted, feature instead of bug. What about AVI?

Fred54, 2016-12-10:
I have been thinking about this https://github.com/ArduCAM/Arduino/issues/80 and I have decided to treat it as a feature instead of a bug. I have had trail cameras stolen before and there is a risk that my current project will be as well. If it is ever stolen I rather the thief not get anything usable. If all the camera takes is corrupted images most thieves would not bother to figure out why.

With that premise I am happy to treat the images that come off of the card as something akin to raw images that need post processing. I can figure out how to open a hex editor and change them one by one. To automate it I am thinking of having the camera save the files with a different extension then once in the computer, run a script to fix all of the pictures and change the extension to JPEG. I am on windows and I am fairly sure Autoit can do it just fine.

Thank you @aldobranti and @ArduCAM between your posts I should be able to figure out a script to fix the JPEGs.

 I would like to be able to fix the AVIs as well. Any thoughts on what I should be looking for in the code on those?

Thanks.

aldobranti, 2016-12-10:
Regards the avi files. There's likely some crap at the first bytes as the fifo wakes up in a mess. Compare with a good one ?

 AVI? (#115)

https://github.com/ArduCAM/Arduino/issues/115

ArduCAM, 2016-12-13:
@aldobranti, We have update the demo sketch and fix the dummy bytes before the JPEG marker FF D8.
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/Shield_V2/ArduCAM_Video2SD