ArduCAM/Arduino #120: SD Card initialization fails on Arducam shield V2 without LCD

ernlzhaw, 2017-01-04:
Hi Lee
I get an error while initializing SD Card inserted to Arducam shield V2 without LCD.
When using exactly the same code with an Arducam shield V2 with LCD it works. I think it could be hardware problem with the spi. Maybe the LCD has some internal pullup/pulldown resistorts which are also needed for SD Card?

I use a sketch which does nothing but initializing the SD-Card. I use a Arduino M0 pro.

Could you please give me some advise
Thank you

[image]
Listing of SPI

supprot, 2017-01-05:
Hi,

Which demo do you use ? We have tested our demos about SD on Shield V2 and anything is normal.

ArduCAM support team

ernlzhaw, 2017-01-05:
I used the example "ArduCAM_Digital_Camera". It worked with an Arducam shield V2 which came with an LCD soldered on the Bottom side. I purchased another Arducam shield V2 with no LCD, because i do not need a Display in my application. The example "ArduCAM_Digital_Camera" does not work on this shield. I get "SD Card Error".
To find the error i deleted everything in the example code except the initialization. It does still give me an "SD Card Error" on the shield with no LCD. But it works on the shield with LCD.

The modified example:
[ArduCAM_Digital_Camera_modified.txt](https://github.com/ArduCAM/Arduino/files/686679/ArduCAM_Digital_Camera_modified.txt)

supprot, 2017-01-05:
@ernlzhaw 

Hi, We have tested  your modified example and it works fine.
Please check the SD card slot and the SD_CS pin as the picture is shown in 
Maybe there are some  bad soldering here.

This is the link of the picture:
[image]

Regards,
ArduCAM support team

ernlzhaw, 2017-01-05:
Hi, thank you for your advise.
I tested the connections of CS, MISO MOSI, SCLK, with an ohmmeter. They have a resistor of 470ohm from Arduino Connector to SD-card-slot. I also measured the supply of the SD-card, it is 3.3V. I do not think it is a soldering problem, because i tested 3 different Arducam shields V2 with no LCD.
Is it correct that the SPI is wired directly to the SD card? I could not find the schematic of the Arducam shield V2. The schematic would help me solving the problem.

supprot, 2017-01-05:
@ernlzhaw 
Hi,
Please provide us your email adress and.
Regards, 
ArduCAM support team

ernlzhaw, 2017-01-05:
Hi, i sent you my email address to admin@arducam.com

mateusz5777, 2017-01-10:
Hi, i have the same problem. Do you know what i have to do?

ernlzhaw, 2017-01-10:
Hi
This shortcut solved the problem for me. I is a shortcut on a 470ohm resistor in the SCLK line. It is needed to have a correct voltage level on the SCLK signal.
I would be interested when you tell me if it also solved your problem.
Best regards
Lukas Erni

mateusz5777, 2017-01-16:
so I have to plug the resistor between SCLK and gnd?

ernlzhaw, 2017-01-16:
No. You have to shortcut the resistor with a wire. Exactly like in the red circle in the attached picture.

mateusz5777, 2017-01-16:
but which resistor? all 4 resistors?

ceritium, 2019-02-07:
Hi, I have the same problem, the SD card is not working, I don't understand where I have to put the wire, the picture show two big red circles.

ArduCAM, 2019-02-07:
Basically the shield is well mate with Arduino UNO R3/ MEGA2560 R3, not sure what kind of board you are using.

ceritium, 2019-02-07:
I am using an Arduino UNO R3 (original), the module is the Arducam Shield v2 with LCD.

ceritium, 2019-02-07:
Ok, I did some tests, the problem is how Mac format the SD cards, everything is working fine, thanks!