ArduCAM/Arduino #122: ESP32 Support

H-LK, 2017-01-09:
I did a proof of concept for connecting an ESP32 with an Arducam.
(https://github.com/H-LK/ESP32_Arducam_Port)

I had to replace some functions which aren't available in the Espressif ESP32 SDK like cbi() and sdi().

Do you plan to add support for the ESP32 in the future? When yes, will there be an own repository like for the ESP8266?

BaEigner, 2017-06-20:
Just got two ESP32s and I'm really interested in using my Arducam Mini with it. Any chance it will happen?

supprot, 2017-06-22:
@H-LK 
Hi, yes, we are developing some demos about arducam with esp32.

Greg-Owens, 2017-06-27:
Hi, 
Can you tell me what you amended.  I can't see any changes in the above code. I assume it should be in ArduCam.h