ArduCAM/Arduino #123: How to check RGB value of pixel.

paekitt, 2017-01-10:
Hi!!
How to check RGB value each pixel of picture ?

ArduCAM, 2017-01-10:
@paekitt, in the camera playback example the image data is saved as RGB565 format.
Each pixel is composed by two consicutive bytes, so you can convert the RGB565 to RGB888 and check the value the way you want.

dprophet, 2017-01-10:
RGB565 is stored in a 16bit number with an extra green bit. Bits are laid out as RRRRRGGGGGGBBBBB

https://s2.www.theimagingsource.com/application-0.0.5623/documentation/ic_imaging_control_class/en_US/images/rgb565.gif

You can check the color bits directly.

I believe that Arduino ATMega is little endian so reverse the image and bits to BBBBBGGGGGGRRRRR

I have some RGB resampling code I wrote some 20 years ago I can dig for if your use case is really that necessary.

paekitt, 2017-01-11:
@ArduCAM @dprophet
Now I use code https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini/ArduCAM_Mini_Video_Streaming
 
And Host App V.2 
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app/ArduCAM_Host_V2.0_Windows

I want to show R G B  value on Host App V.2 ( Serial.print(); ) in mode streaming or capture image.

supprot, 2017-01-11:
@paekitt 
The ArduCAM_Mini_Video_Streamimg can't support showing RGB value on Host App V.2.
We will update the sample to support this function as soon as possible.

Regards,
ArduCAM support team.

supprot, 2017-01-11:
@paekitt 
Hi,
You can try this example and ArduCAM_Host_V2
If you want to show RGB value, just to choose 'SetToBMP'
 at the ArduCAM_Host_V2.

// ArduCAM Mini demo (C)2016 Lee
// web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini camera, and can run on any Arduino platform.
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It needs to be used in combination with PC software.
// It can take photo continuously as video streaming.
//
// The demo sketch will do the following tasks:
// 1. Set the camera to JEPG output mode.
// 2. Read data from Serial port and deal with it
// 3. If receive 0x00-0x08,the resolution will be changed.
// 4. If receive 0x10,camera will capture a JPEG photo and buffer the image to FIFO.Then write datas to Serial port.
// 5. If receive 0x20,camera will capture JPEG photo and write datas continuously.Stop when receive 0x21.
// 6. If receive 0x30,camera will capture a BMP  photo and buffer the image to FIFO.Then write datas to Serial port.
// 7. If receive 0x11 ,set camera to JPEG output mode.
// 8. If receive 0x31 ,set camera to BMP  output mode.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.5.2 compiler or above
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
//This demo can only work on OV2640_MINI_2MP or OV5642_MINI_5MP or OV5642_MINI_5MP_BIT_ROTATION_FIXED platform.
#if !(defined OV5642_MINI_5MP || defined OV5642_MINI_5MP_BIT_ROTATION_FIXED || defined OV2640_MINI_2MP)
  #error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};
// set pin 7 as the slave select for the digital pot:
const int CS = 7;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;

#if defined (OV2640_MINI_2MP)
ArduCAM myCAM( OV2640, CS );
#else
ArduCAM myCAM( OV5642, CS );
#endif
uint8_t read_fifo_burst(ArduCAM myCAM);

void setup() {
  // put your setup code here, to run once:
  uint8_t vid, pid;
  uint8_t temp;
#if defined(__SAM3X8E__)
  Wire1.begin();
  Serial.begin(115200);
#else
  Wire.begin();
  Serial.begin(921600);
#endif

  Serial.println("ACK CMD ArduCAM Start!");

  // set the CS as an output:
[... 351 lines cut ...]
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
	  Serial.println("ACK IMG");
      Serial.write(temp_last);
      Serial.write(temp);
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
      break;
    delayMicroseconds(15);
  }
  myCAM.CS_HIGH();
  is_header = false;
  return 1;
}

Regards,
ArduCAM support team

paekitt, 2017-01-12:
@supprot  @ArduCAM @dprophet 
Hi , 
I try your example (@supprot) and choose 'SetToBMP' but don't show RGB value. 

And I try convert the RGB565 to RGB888.

But Host App V.2 show Number command all the same.

I would like to show and collect RGB data from a snapshot. But I do not get on how to access this data, without storing the image or whatever. 

I want RGB value to plot graph in my work.

lbtt0423, 2017-01-12:
@paekitt 
Hi, 
We have tested this example and if can work fine. If you want to save the IMG，just choose 'SaveImage'
Please take some picture to us, and let us know your detail problem.

Regards,
ArduCAM support team

lbtt0423, 2017-01-12:
@paekitt 
Hi,
We have tested that example we send to you and it can work fine,
Please take some picture to us and let us know  your detail problems.

Regards,
Arducam support team.

dprophet, 2017-01-12:
@paekitt 

> I would like to show and collect RGB data from a snapshot. But I do not get on how to access this data, without storing the image or whatever.

Use a Bitmap (.BMP) or a compressed Bitmap (.PNG). You can use libpng to decompress the pixels and access the RGB.

I dont know your use case but, IMO, but what problem are you trying to solve? I think you are going about the problem wrong.

I dont think you need to build anything yourself. I use the ArduCAM outputs to feed others systems. In the case of image analysis you can use IBM Watson cloud.
https://visual-recognition-demo.mybluemix.net/

If you want to analyze it yourself use R. They have a nice image statistical package.

ttps://dahtah.github.io/imager/imager.html

OpenCV has a very nice python interface that allows you to do some great analysis of Images.
http://opencv.org/

If you just want to view the RGB values at the host, use something like:
http://instant-eyedropper.com/

Just accessing RGB values isnt really solving any problem that I can see. What you are trying to accomplish has likely already been done and you can use those tools.

supprot, 2017-01-13:
@paekitt 
Hi,
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/Shield_V2/ArduCAM_Camera_Playback/ArduCAM_Camera_Playback.ino

Using this example can read RGB data,
Please refer to the relevant code in the example:

 //Read 320x240x2 byte from FIFO
  //Save as RGB565 bmp format
  for (i = 0; i < 240; i++)
    for (j = 0; j < 320; j++)
    {
      VH = myCAM.read_fifo();
      VL = myCAM.read_fifo();
      buf[k++] = VL;
      buf[k++] = VH;

regards,
ArduCAM support team.

paekitt, 2017-01-25:
@supprot  @ArduCAM 
[image]

RGB value from capture , I try print (Serial.print();) but RGB value indirect image ?

[image]
[image]

AliSyech, 2018-02-10:
What is the tool for getting RGB value ?
Can camera OV7670 is used ?

93450101, 2018-07-01:
I want get data of an arducam image and rebuild image with data I received. I cant speak English very well. where can I find useful example