ArduCAM/Arduino #124: possible to change hue,saturation on Arducam-Mini OV2640

oric-dan, 2017-01-10:
I'm using the Mini with OV2640 very successfully for image capture and display, and am very happy all in all. However, I notice from the d/s that all of the color,hue,saturation,etc adjustments are in "reserved" registers.  

Is it possible to get info on these registers. I would like to make some hue and saturation adjustments. A primary issue is that orange comes out of the FIFO as RGB565=FC10h rather than FC00h, so has too much "blue", and looks like salmon-pink. Note I've tried both bright Led and tungsten lighting and get the same result. See attached, cap atop green bottle to the right. I'm also attaching what it looks like using my Sanyo digital camera.

Also, I am assuming the camera default uses Advanced AWB mode, adjustments based upon color temperature. Yes?

[image]

[image]

H-LK, 2017-01-10:
Just convert the code from here (page 15) and insert it into ov2640_regs.h
http://www.dragonwake.com/download/camera/ov2640/OV2640%20Camera%20Module%20Software%20Application%20Notes.pdf

For example I changed brightness and contrast with these options:

   /* JPG 1600x1200 */
const struct sensor_reg OV2640_1600x1200_JPEG[] PROGMEM =
{

	//Set Brightness + 2
	{0xff, 0x00},
	{0x7c, 0x00},
	{0x7d, 0x04},
	{0x7c, 0x09},
	{0x7d, 0x40},
	{0x7d, 0x00},

	//Set Contrast + 2
	{0xff, 0x00},
	{0x7c, 0x00},
	{0x7d, 0x04},
	{0x7c, 0x07},
	{0x7d, 0x20},
	{0x7d, 0x28},
	{0x7d, 0x0c},
	{0x7d, 0x06},

.......

rflmota, 2017-01-10:
Hi @H-LK ,

Is it possible for you to share the same kind of application note for _OV5642_ module?

H-LK, 2017-01-10:
Just google for "OV5642 Software PDF"
http://forum.motofan.ru/index.php?act=attach&type=post&id=256766

ArduCAM, 2017-01-13:
@Hendrik, You can change the color saturation, brightness, constrast for ArduCAM mini 2MP (OV2640)

supprot, 2017-01-13:
@oric_dan, it is more advanced topic about the settings of the sensor. Would you please let me know your email address , then we can discuss more about it.

paekitt, 2017-01-16:
Hi @oric-dan ,
How to fixed exposure time ?

oric-dan, 2017-01-16:
"oric_dan" <oric333@yahoo.com>

--------------------------------------------

 

 
 
 
 
 
 
 
 
 
 
 it on GitHub, or mute
 the thread.

mosazahany, 2022-03-03:
Strange problems in adjusting the saturationand the brightness in ov2640
These bugs do not actually exist and are a problem of the available documents
When you execute the following commands
set_brightness(s,2);
set_contrast(s,2);
set_brightness(s,2);
The last command is always active and the previous commands are canceled.
To solve this problem, you must activate the desired effect and this is done with the following command

Run the following command at the end of all settings.

unsigned char enable_effect=	1(saturation and hue enable)+
				2(aditional saturation enable)+
				4(brighnes and contrast enable)+
				8(spital effect1 enable green -> blue)+
				16(spital effect1 enable gray -> read)+
				32(remove (UV) in YUV color system)+
				64(enable negative effect)+
				128(remove (Y) in YUV color system)  ;
write_SCCB(0xff, 0x00);
write_SCCB(0x7c, 0x00);
write_SCCB(0x7d,enable_effect);

command 
write_SCCB(0x7c, ADRESS);
Preparing the address for data injection

ofter that use
The following command to inject data
write_SCCB(0x7d,byte_0);
write_SCCB(0x7d,byte_1);
write_SCCB(0x7d,byte_2);
.
.
.

The memory structure is as follows
byte_0 =enable_effect  
         byte_0->bit0  =saturation and hue 
         byte_0->bit1  =saturation enable
         byte_0->bit2  =brighnes and contrast enable
         byte_0->bit3  =green -> blue spitial effect (Antique and blunish and greenish and readdish and b&w) enable
         byte_0->bit4  =gray  -> read spitial effect (Antique and blunish and greenish and readdish and b&w) enable
         byte_0->bit5  =remove (UV) in YUV color system
         byte_0->bit6  =negative enable
         byte_0->bit7  =remove (Y) in YUV color system
byte_1 =saturation1  0-255
byte_2 =hue 0-255
byte_3 =saturation2  0-255
byte_4 =reenter saturation2 in documents
byte_5 =spital effect green -> blue   0-255
byte_6 =spital effect gray  -> read   0-255
byte_7 =contrast uper  eadge  0-255
byte_8 =contrast lower eadge  0-255
byte_9 =brighnes 0-255
byte_10= if byte_10==4 contrast effective 

mosazahany@gmail.com

با تشکر