ArduCAM/Arduino #128: Support for platformio y

goetzmoritz, 2017-01-16:
I there,

I'd like to use platformio for development but it seems like for my generic ESP8266 module or for my nodeMCU I have to use your custom esp8266-core-library as with the regular esp8266 library that comes with platformio I only get messed up images and in another post you mention that for regular esp8266-users one has to use your image. So, how do I proceed with my issue? I just failed when I tried to import your customized esp8266-core-library into platformio, is there a workaround for me? Thanks.

dprophet, 2017-01-17:
If you are looking for internet connectivity I use ArduCAM with
https://www.seeedstudio.com/LinkIt-Smart-7688-Duo-p-2574.html
I think you will find out that if you add too much code in the Arduino side you will run out of 32k space. Plenty of space in the OpenWrt Linux side and a lot better Internet connectivity.

goetzmoritz, 2017-01-17:
Erik,

thank you for your hint. Unfortunatly I'm bound to ESP8266.
Is it possible for us to get an updated version of the ESP8266-Arducam-branch? That would be so great.

buschnyj, 2018-01-17:
Hello @ArduCAM, can you please weigh in on this? I am attempting to use the 2mp camera module. There is no option on the PlatformIO Library Manger: [http://docs.platformio.org/en/latest/librarymanager/](url)

Attempting to save the Arduino folder under `/Platformio_Projcet_Name/lib` does not link to it. 

Thank you.

ArduCAM, 2018-01-18:
The founder of PlatformIO Ivan Kravets just create a pull request for supporting this, please check:
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/library.json

buschnyj, 2018-01-18:
Noted. Thank you