ArduCAM/Arduino #129: Connecting Camera OV2640 Mini Module to ethernet shield

wpanduka, 2017-01-19:
Need help in connecting camera module ov2640 to Ethernet shield in order to capture images and upload to database. i am not clear what pins i have to plug in to in the Ethernet Shield. and how to capture the saving images on the SD card in the Ethernet Shield ( devices i m using Ardino uno R3, OV2640 ArduCAM Mini Module, Arduino Ethernet shield R3 with micro SD connector)

supprot, 2017-01-20:
@wpanduka 
Hi,
Please refer to the exhibit hardware connection .
[image]

Regards,
ArduCAM support team.

wpanduka, 2017-01-20:
Hi , Thank you so much for your help, greatly appreciate it. i have manage to connect and the camera is working ( i have check the example ArduCam_Mini_Capture2SD) and there is a massage on the serial panel "The fifo length is :8388607 Over size" . 
so how to solve the oversize problem? 
and another thing can you please tell me where is this images save on the SD card and how to find it on the folder?
thank you once again for your help

supprot, 2017-01-20:
@wpanduka 
Hi,
Our 2MP's FIFO size is 0x5ffff.
We think maybe  your hardware connection is not stable, you can check the hardware connection firstly.
if it can work fine, you will see "start Capture"、"Capture Done"in the serial panel,and the image's size can't over 0x5ffff.
Another way, you can use our ArduCAM_Mini_Video_Streaming demo and our ArduCAM_Host_V2 to check if your camera can work fine.

When you capture succesfully,the images is under the root directory of the SD card, so you just need open your SD card, you will see the image file.

Let us know if you need more help,

Best regards,
ArduCAM support team.

wpanduka, 2017-01-20:
Hi , i have connected the exactly the way as you mention in the diagram , when i run the "ArduCAM_Mini_Video_Streaming" example there are weired letters appear on the serial display  and nothing happens, when i run the ArduCam_Mini_Capture2SD it shows 

ArduCAM Start!
rface Error!
SD Card Error
OV2640 detected.
ArduCAM Start!
SPI1 interface Error!
SD Card Error
OV2640 detected.
star Capture
Capture Done!
The fifo length is :8355711
Over size.
 

dont know what do , please tell me what to do , thanks

supprot, 2017-01-21:
@wpanduka 
@wpanduka erf
Hi,
Your SPI interface is error, you should check it firstly.
Another, your SD card init is error too. In our code ,the SD CS is 9 and you should change the SD CS to 4 .
this is the arduino-Ethernet-Shield2-V2-sch.
[arduino-Ethernet-Shield2-V2-sch.pdf](https://github.com/ArduCAM/Arduino/files/720785/arduino-Ethernet-Shield2-V2-sch.pdf)

Let us know if you need more help.

Regards,
ArduCAM suport team.

supprot, 2017-01-21:
@wpanduka 
Hi,
Your SPI interface is error, you should check it firstly.
Another, your SD card init is error too. In our code ,the SD CS is 9 and you should change the SD CS to 4 .
this is the arduino-Ethernet-Shield2-V2-sch.
arduino-Ethernet-Shield2-V2-sch.pdf

Let us know if you need more help.

Regards,
ArduCAM suport team.

supprot, 2017-01-21:
@wpanduka 
Hi,
Your SPI interface is error, you should check it firstly.
Another, your SD card init is error too. In our code ,the SD CS is 9 and you should change the SD CS to 4 
this is the arduino-Ethernet-Shield2-V2-sch.
[arduino-Ethernet-Shield2-V2-sch.pdf](https://github.com/ArduCAM/Arduino/files/720789/arduino-Ethernet-Shield2-V2-sch.pdf)

Let us know if you need more help.

Regards,
ArduCAM suport team.

cfeng19, 2017-05-04:
I also seem to have a similar issue here. 

Using a w5100 ethernet shield, OV2640 mini 2mb, and implementing the same connections shown in the hardware connection diagram, with SD CS as 4, and SPI CS as 7, I get SPI Interface Error running Arducam_Mini_Capture2SD. 

Surprisingly, all the other functions work fine if I comment out the while loop after //initialize SPI;. Any ideas how I can get the SPI initialized, I am pretty sure my connections are done correctly.

cfeng19, 2017-05-04:
Actually, I moved SDA and SCL to the PC4 and PC5 on the digital header side, and ran the code commenting out the iniitalize SPI, and I can capture images on my SD card. Problem solved.

supprot, 2017-05-05:
@cfeng19 

Glad to hear you have solved your problems.

Regards,
ArduCAM support team.

gregnsk, 2020-05-25:
I'm trying to connect ArduCAM-2MP-Plus to Ethernet shield attached to Mega2560 R3. I followed the schematics as shown above. When I'm trying to run ArduinoCAM_Mini_2MP_Plus_Multi_Capture2SD (or any other example related to 2MP-Plus), the code fails with SPI Interface Error. If, as @cfeng19 recommended, I comment out the initialize SPI part (//SPI.begin()) - the code gets stuck in the following line (myCAM.write_reg(0x07, 0x80);

What should I do differently?