ArduCAM/Arduino #130: Serial monitor problem while connecting camera with wifi

happyfear, 2017-01-24:
This is what my serial monitor shows. Should i be able to change my wifi name and password from there. I connected arduino to putty and then it gave me text "ACK CMD switch to OV2640_160x120" and "ACK CMD switch to OV5642_320x240"
[image]

happyfear, 2017-01-24:
```
#include <ArduCAM.h>
#include <memorysaver.h>
#include <mt9d111_regs.h>
#include <mt9d112_regs.h>
#include <mt9m001_regs.h>
#include <mt9m112_regs.h>
#include <mt9t112_regs.h>
#include <mt9v111_regs.h>
#include <ov2640_regs.h>
#include <ov3640_regs.h>
#include <ov5640_regs.h>
#include <ov5642_regs.h>
#include <ov7660_regs.h>
#include <ov7670_regs.h>
#include <ov7675_regs.h>
#include <ov7725_regs.h>

// ArduCAM Mini demo (C)2016 Lee
// web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini 2MP/5MP camera, and can run on any Arduino platform.
// This demo was made for ARDUCAM_SHIELD_V2.
// It needs to be used in combination with PC software.
// It can take photo continuously as video streaming.
//
// The demo sketch will do the following tasks:
// 1. Set the camera to JEPG output mode.
// 2. Read data from Serial port and deal with it
// 3. If receive 0x00-0x08,the resolution will be changed.
// 4. If receive 0x10,camera will capture a JPEG photo and buffer the image to FIFO.Then write datas to Serial port.
// 5. If receive 0x20,camera will capture JPEG photo and write datas continuously.Stop when receive 0x21.
// 6. If receive 0x30,camera will capture a BMP  photo and buffer the image to FIFO.Then write datas to Serial port.
// 7. If receive 0x11 ,set camera to JPEG output mode.
// 8. If receive 0x31 ,set camera to BMP  output mode.
// This program requires the ArduCAM V4.0.0 (or later) library and ARDUCAM_SHIELD_V2
// and use Arduino IDE 1.5.2 compiler or above
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
//This demo can only work on ARDUCAM_SHIELD_V2  platform.
#if !(defined (ARDUCAM_SHIELD_V2)&&(defined (OV5640_CAM) ||defined (OV5642_CAM)||defined (OV2640_CAM)))

#endif
#define BMPIMAGEOFFSET 66
#define ARDUCAM_SHIELD_V2
#define OV2640_CAM
#if defined(ESP8266)
 const int SPI_CS = 16;
#else 
 const int SPI_CS =10;
#endif
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;

const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};

#if defined (OV2640_CAM)
ArduCAM myCAM(OV2640, SPI_CS);
#elif defined (OV5640_CAM)
ArduCAM myCAM(OV5640, SPI_CS);
[... 383 lines cut ...]
  while ( length-- )
  {
    #if defined (ESP8266)
      yield();
    #endif
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      Serial.write(temp_last);
      Serial.write(temp);
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
      break;
    delayMicroseconds(15);
  }
  myCAM.CS_HIGH();
  is_header = false;
}
```

supprot, 2017-01-25:
@happyfear
Hi,
This demo has nothing to with WIFI,but you should our Host V2 to show the image.

Let us know if you need more help,
Regards,
Arducam support team.

happyfear, 2017-01-25:
How can i access wifi configuration then? I would like to see my camera stream trought wifi. When i plug in ArduCAM i get wifi named "test" but i dont know the password for it and it's protected.

happyfear, 2017-01-25:
So now it shares wifi. I can change wifi name/password trought the code. But doesn't show the camera picture on "Arduino\arduino-1.6.5-r5\libraries\ArduCAM\examples\ESP8266\ArduCAM_Mini_OV2640_websocket_server\html". When i try to connect to camera from there the Arduino IDE serial monitor tells me "QL change to: 0".

I have ArduCAM ESP8266 UNO V2 and QV2640 2MP camera.

I followed this tutorial http://www.arducam.com/arducam-supports-esp8266-arduino-board-wifi-websocket-camera-demo/

happyfear, 2017-01-25:
[image]

[image]

Hardware:	ESP8266
Core Version: 2.2.0
Board: ArduCAM ESP8266 UNO board V2
Camera module: 2MP image sensor OV2640

supprot, 2017-01-25:
Hi @happyfear 

Your images datas contains some errors. We think it maybe exist some bad connetions and you should try replug your mini2mp to your ESP8266 board.

Let us know if you need more help.
Regards,
Arducam support team.

happyfear, 2017-01-26:
I tryed repluging the camera and no luck. 

[image]

Also if i am connected to ArduCAM wifi it shows yellow triangle on my computer.

[image]

enabled ov2640 cam and arducam_shield_v2 in memorysaver.ch

https://gyazo.com/4bd061482b61d305c406aa48cc2be1e1

ArduCAM,ESP8266-websocket,UTFT4ArduCAM_SPI are in Arduino folder

[image]

I tryed it in I.E

[image]

supprot, 2017-01-27:
@happyfear 
Hi,
What's the mode do you set ? The mode can be set in our code.If you are working at AP mode, have you connect the ESP8266's wireless hostpot.If you are working at station mode, you should connect to your own router .Can you detected the OV2640? Is the SPI OK? If they all work fine, so, the problem is exist in data transmission, please ensure your wifi signal is good.Another, you can print the image data to the serial port. Please ensure are there 0xff,0xd8 in the header firstly.

Let us know if you need more help.
Regards,
Arducam suppprt team.

happyfear, 2017-01-28:
Im starting to think that my ArduCAM-Mini-2MP OV2640 doesn't even work. It's connected into it's place but it doesn't take any pictures. Today i tryed to take pictures to my SD card but no luck.
Also i got my wifi working with LEDs so i could turn my LEDs on/off trought wifi.

[image]

```
// ArduCAM demo (C)2016 Lee
// web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with a supported camera modules, and can run on any Arduino platform.
// This demo was made for Omnivision OV2640 2MP sensor.
// It will run the ArduCAM ESP8266 2MP as a real 2MP digital camera, provide both JPEG capture.
// The demo sketch will do the following tasks:
// 1. Set the sensor to JPEG mode.
// 2. Capture and buffer the image to FIFO every 5 seconds 
// 3. Store the image to Micro SD/TF card with JPEG format in sequential.
// 4. Resolution can be changed by myCAM.set_JPEG_size() function.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM ESP8266 2MP shield
// and use Arduino IDE 1.5.2 compiler or above
#include <ArduCAM.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

//This demo can only work on OV2640_MINI_2MP or ARDUCAM_SHIELD_V2 platform.
#if !(defined (OV2640_MINI_2MP)||(defined (ARDUCAM_SHIELD_V2) && defined (OV2640_CAM)))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
// set GPIO16 as the slave select :
const int CS = 16;
//Version 2,set GPIO0 as the slave select :
const int SD_CS = 0;
ArduCAM myCAM(OV2640, CS);

void myCAMSaveToSDFile(){
  char str[8];
  byte buf[256];
  static int i = 0;
  static int k = 0;
  static int n = 0;
  uint8_t temp, temp_last;
  File file;
  //Flush the FIFO
  myCAM.flush_fifo();
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  //Start capture
  myCAM.start_capture();
  Serial.println("star Capture");
 while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
 Serial.println("Capture Done!");  

 //Construct a file name
 k = k + 1;
 itoa(k, str, 10);
 strcat(str, ".jpg");
 //Open the new file
 file = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
 if(! file){
  Serial.println("open file faild");
  return;
 }
 i = 0;
 myCAM.CS_LOW();
 myCAM.set_fifo_burst();
#if !(defined (ARDUCAM_SHIELD_V2) && defined (OV2640_CAM))
SPI.transfer(0xFF);
#endif
 //Read JPEG data from FIFO
 while ( (temp !=0xD9) | (temp_last !=0xFF)){
  temp_last = temp;
  temp = SPI.transfer(0x00);

  //Write image data to buffer if not full
  if( i < 256)
   buf[i++] = temp;
   else{
    //Write 256 bytes image data to file
    myCAM.CS_HIGH();
    file.write(buf ,256);
    i = 0;
    buf[i++] = temp;
    myCAM.CS_LOW();
    myCAM.set_fifo_burst();
   }
   delay(0);  
 }
 
 //Write the remain bytes in the buffer
 if(i > 0){
  myCAM.CS_HIGH();
  file.write(buf,i);
 }
 //Close the file
 file.close();
  Serial.println("CAM Save Done!");
}

void setup(){
  uint8_t vid, pid;
  uint8_t temp;
  Wire.begin();
  Serial.begin(115200);
  Serial.println("ArduCAM Start!");

  //set the CS as an output:
  pinMode(CS,OUTPUT);

  //initialize SPI:
  SPI.begin();
  SPI.setFrequency(4000000); //4MHZ

  delay(1000);
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if (temp != 0x55){
    Serial.println("SPI1 interface Error!");
    while(1);
  }
  
    //Initialize SD Card
  if(!SD.begin(SD_CS)){
    Serial.println("SD Card Error");
  }
  else
  Serial.println("SD Card detected!");
  

  
//Check if the camera module type is OV2640
  myCAM.wrSensorReg8_8(0xff, 0x01);
  myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
  if ((vid != 0x26 ) && (( !pid != 0x41 ) || ( pid != 0x42 )))
   Serial.println("Can't find OV2640 module!");
   else
   Serial.println("OV2640 detected.");
   myCAM.set_format(JPEG);
   myCAM.InitCAM();
}

void loop(){
  delay(5000);
  myCAMSaveToSDFile();
}

```

happyfear, 2017-01-30:
I am having so much trouble with ArduCAM.

- Camera picture wont appear.
-ArduCAM wont take power in from 9V battery(barrel wire)
-ArduCAM wont take power in from any motorshield(tryed multiple). 9V battery or Li-Po battery(11.1V)
-Servos crash when powered from USB. Especially from D8 port.

ArduCAM, 2017-01-31:
@happyfear , sorry for so much trouble, we are now on Chinese New Year holidays. We will get back to you when our engineers back from holidays.
ESP8266 is more complex than standard Arduino board, there are some restrictions when using different GPIO pins.

supprot, 2017-02-01:
@happyfear 
Hi,
We are very sorry for reply you so late because of our New Year holiday.
Please try this demo on ArduCAM ESP8266 UNO board.

// ArduCAM demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with a supported camera modules, and can run on any Arduino platform.
// This demo was made for Omnivision  2MP/5MP sensor.
// It will run the ArduCAM ESP8266 2MP/5MP as a real 2MP/5MP digital camera, provide both JPEG capture.
// The demo sketch will do the following tasks:
// 1. Set the sensor to JPEG mode.
// 2. Capture and buffer the image to FIFO every 5 seconds 
// 3. Store the image to Micro SD/TF card with JPEG format in sequential.
// 4. Resolution can be changed by myCAM.set_JPEG_size() function.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM ESP8266 2MP/5MP shield
// and use Arduino IDE 1.6.8 compiler or above
#include <ArduCAM.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

//This demo can work on OV2640_MINI_2MP/OV5640_MINI_5MP_PLUS/OV5642_MINI_5MP_PLUS/OV5642_MINI_5MP_PLUS/
//OV5642_MINI_5MP_BIT_ROTATION_FIXED/ ARDUCAM_SHIELD_V2 platform.
#if !(defined (OV2640_MINI_2MP)||defined (OV5640_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP_PLUS) \
    || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) \
    ||(defined (ARDUCAM_SHIELD_V2) && (defined (OV2640_CAM) || defined (OV5640_CAM) || defined (OV5642_CAM))))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
// set GPIO16 as the slave select :
const int CS = 16;
//Version 2,set GPIO0 as the slave select :
const int SD_CS = 0;

#if defined (OV2640_MINI_2MP) || defined (OV2640_CAM)
ArduCAM myCAM(OV2640, CS);
#elif defined (OV5640_MINI_5MP_PLUS) || defined (OV5640_CAM)
ArduCAM myCAM(OV5640, CS);
#elif defined (OV5642_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) ||(defined (OV5642_CAM))
ArduCAM myCAM(OV5642, CS);
#endif
void myCAMSaveToSDFile(){
  char str[8];
  byte buf[256];
  static int i = 0;
  static int k = 0;
  uint8_t temp = 0, temp_last = 0;
  uint32_t length = 0;
  bool is_header = false;
  File outFile;
  //Flush the FIFO
  myCAM.flush_fifo();
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  //Start capture
  myCAM.start_capture();
  Serial.println(F("Star Capture"));
 while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
 Serial.println(F("Capture Done."));  

 length = myCAM.read_fifo_length();
 Serial.print(F("The fifo length is :"));
 Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //8M
  {
[... 118 lines cut ...]
   else
   Serial.println(F("OV5642 detected."));
  #endif
    //Change to JPEG capture mode and initialize the OV2640 module
  myCAM.set_format(JPEG);
  myCAM.InitCAM();
  #if defined (OV2640_MINI_2MP) || defined (OV2640_CAM)
    myCAM.OV2640_set_JPEG_size(OV2640_320x240);
  #elif defined (OV5640_MINI_5MP_PLUS) || defined (OV5640_CAM)
    myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
    myCAM.OV5640_set_JPEG_size(OV5640_320x240);
  #elif defined (OV5642_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) ||(defined (OV5642_CAM))
    myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
    myCAM.OV5642_set_JPEG_size(OV5642_320x240);  
  #endif
}

void loop(){
  delay(5000);
  myCAMSaveToSDFile();
}

Let us know if you need more help.
Regards,
ArduCAM support team.

happyfear, 2017-02-02:
I got my camera working trought wifi after i replaced my ams1117 5.0 dn628 voltage regulator chip. I replaced it with Arduino UNO voltage regulator chip. ArduCAM chip burned out somehow(Vout). I never put in 11.1V in my ArduCAM. Some sources suggest that you should never put more thant 7V in ams1117 5.0 dn628 voltage regulator chip.

happyfear, 2017-02-04:
I tryed your code to capture to SD card. Every single code gives me pretty much the same serial monitor "error".  
[image]

supprot, 2017-02-05:
@happyfear 
Hi,

Please try format your SD card.

Let us know if you need more help.
ArduCAM support team.

happyfear, 2017-02-05:
Hello,

i can't belive that formating did the trick and it actually works now. The SD card was totally empty before.
Thank you! I let you know if i have any more questions. 

Best regards,
Kevin

supprot, 2017-02-05:
@happyfear 
Hi,
Congratulations.
Let us know if you have any problems.

Regards,
ArduCAM support team.

happyfear, 2017-02-09:
Hello,

How can i change that how long clip does it record to my SD card. Currently it records .avi files that are 1 minute long but i would like to change it to 10 seconds.

Best regards,
Kevin

supprot, 2017-02-10:
@happyfear 
Hi,
For the first question:
Our video is encoded by 200 images which is determined by the pic_num variable. If you want to short the video time ,you can change the pic_num value. If you want to increase the playback speed,you just change the rate variable which can determine the playback speed.

For the second queston:
Our demo is time to record the video and if you want change ,you should change our demo such as through the serial command to control record video.

Let us know if you need more help.
ArduCAM support team.

happyfear, 2017-02-11:
What is the reason behind the repeating pins? When i connect OV2640 camera to the 8 pins meant for the camera i cant use d5,d6,d7,d2,d1 pins in the right pin row. Is there any way i can bypass it?

[image]

supprot, 2017-02-12:
@happyfear 
Hi,
Our ArduCAM use the SPI and I2C interfaces which is  D1, D2, D5, D6, D7.
Look at the SPI.cpp file, they are defined SPECIAL mode when you SPI.begin();
If you want to change them as a general IO,you should change their mode to INPUT or OUTPUT.
[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

inininked, 2018-05-21:
@supprot 
Hi,
Can you explain more about the SPI setting? After I change above parts to OUTPUT, it will show SPI1 Interface Error. What other step I should do?

I'm using ArduCAM ESP8266 UNO board and OV5642 5MP Plus Camera. After I use the example code, I cannot even control the BUILTIN_LED. 

I want to use ArduCAM to connect both OV5642 and L298N, is there any way to know how to setup the pin?

You mentioned "change them as general IO". Does it mean, if I want to connect to multiple components, I need to change all OV5642's pin to general IO? 

Thanks for any advice, I'm not very familiar with engineering.

UCTRONICS, 2018-05-21:
@inininked 
Of course.
In the demo, we use hardware spi which means you just need to config special function register.
After you have enable the hardware spi by configuring the spi register, it will not work as general IO port.
If you want to use it as general IO port, you must reconfigure it as a general port.At the same way, if it work as a general io port, it will not work as a hardware  SPI signal.

The Arducam ESP8266 doesn't have enough gpio source, if you want to driver the L293D, I advice you use our Arducam esp32 UNO which has enough gpio source for use.http://www.uctronics.com/arducam-esp32-uno-board-for-arducam-mini-camera-module-compatible-with-arduino-uno-r3.html

inininked, 2018-05-23:
@UCTRONICS 
Thanks for your reply! 

Buying  Arducam esp32 UNO may not be my option, since local retailer does not have this type and I need it in urgent. 

I've been thinking the possibility of connecting L298N via SPI (I'm still googling how to do it), do you think it is doable? And ArduCAM should be able to support two SPI interface modules, right? Do I need to modify the SPI.cpp/SPI.h? Or I can simply add one more pin in .ino file?  

Thanks

UCTRONICS, 2018-05-23:
@inininked 
Dear sir,
But Arducam ESP8266 UNO doesn't have enough pin source, You don't have other gpio to driver L298N .
Maybe you can use Arduino UNO to di that.