ArduCAM/Arduino #134: Error compiling for board ArduCAM ESP8266 UNO

Darakarn, 2017-02-02:
Hi, Yesterday I update board Arduino 1.6.13 but error compiling for board ArduCAM ESP8266 UNO ?

Help me!
[image]
[image]

supprot, 2017-02-02:
@Darakarn 
Hi, 
We have update our ArduCAM ESP8266 UNO board .
Please delete the old version and download the version 2.2.2.

Let us know if you need more help.
Regards,
ArduCAM support team

happyfear, 2017-02-02:
Delete C:\Users\YourUserName\AppData\Local\Arduino15\packages\ArduCAM_ESP8266_UNO

and

C:\Users\YourUserName\AppData\Local\Arduino15\staging\packages\ArduCAM_ESP8266_UNO.zip

Darakarn, 2017-02-03:
@supprot 
Thank, I have a problem.

Why upload code ArduCAM_Shield_V2_Capture2SD, Arduino UNO board "SPI interface Error!", but ArduCAM ESP8266 UNO board "SPI interface OK!" .

supprot, 2017-02-03:
@Darakarn 
Hi,

When you upload the code about Shield_V2, you should use our Shield_V2 and nano esp8266 board instead of ArduCAm ESP8266 UNO V2 board.Because the pin definition has changed.

Let us know if you need more help.
Regards,
Arducam Support team.