ArduCAM/Arduino #135: ArduCAM_ESP8266_UNO Board Missing

trvlr737, 2017-02-02:
Hopefully someone can help. I had the OV2640 cam working perfectly. I've changed something in my Arduino 1.6.9 settings and the ArduCAM_ESP8266_UNO Board is no longer listed. The add-on is still on the computer at the installed location (C:\Users\Your computer name\AppData\Local\Arduino15\) but, I do not know how to get the  ArduCAM_ESP8266_UNO Board relisted. I tried to reinstall the entire package, but I get a "CRC" error. 

Please help.

happyfear, 2017-02-02:
I had the same problem yesterday. 
remove arducam packages in Arduino IDE.

then

Delete C:\Users\YourUserName\AppData\Local\Arduino15\packages\ArduCAM_ESP8266_UNO

and

C:\Users\YourUserName\AppData\Local\Arduino15\staging\packages\ArduCAM_ESP8266_UNO.zip

and install again.

trvlr737, 2017-02-03:
Thank You happyfear!! Your answer fixed the problem.

closed 2017-02-03 by trvlr737