ArduCAM/Arduino #136: Low Power mode and wake-up

lorenberg, 2017-02-03:
Hello,

my previous topic was ignored, so I try to open a new one with the current status :)
I have the Arducam Mini with ESP8266, and I need to switch everything off sometimes, so that I consume only 60mA. 
I do this by calling the instruction myCAM.set_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK);
This works entirely fine. Then, I put the ESP.deepSLEEP, and after a few seconds I wake up the circuit (the ESP resets automatically).
The problem is that when the circuit resets, the OV2640 module is not detected anymore!
On the software application note of the Arducam, I found this line saying " User should reinitialize the sensor when power up the sensor circuit again." So I'm probably missing this part.

But how should I reinitialize the sensor? I tried to call ArduCAM:InitCAM(), I tried to make sensor_addr and sensor_model public, so I can write on them in the setup() method of the Arduino, but still no luck.

Do you have any suggestion on how I can reinitialize the sensor from the Arduino code?

Thank you very much

supprot, 2017-02-03:
@lorenberg 

Hi,
When you  do operate on camera, you  should use myCAM.clear_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK); 
Then do  myCAM.set_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK);

Let us know if you need more help.
Regards,
ArduCAm support team.

lorenberg, 2017-02-03:
Thanks for your help, I'll try it. Unfortunately now I have another problem....
All of a sudden my Arduino IDE is giving me this error
Board ArduCAM_ESP8266_UNO (platform ArduCAM_ESP8266_UNO, package ArduCAM_ESP8266_UNO) is unknown
Error compiling for board ArduCAM ESP8266 UNO.

I just went away for lunch and when I got back I cannot compile anything anymore! Do you know what is wrong?

lorenberg, 2017-02-03:
Ok, problem fixed. I needed to update to the libraries 2.2.2 :)

supprot, 2017-02-03:
@lorenberg 

Congratulations ,let us know if you need more help.

Regards,
ArduCAM support team.

Schnulli87, 2017-04-10:
Hi,
I have an additional question regarding the usage of low power mode (GPIO_PWDN_MASK):
I have modified the standard web-sketch to put the camera into sensor standby after sending a photo and wake it up, when a http-request for a picture is made.
Here is what I have discovered:
If there is no delay between standby Exit and starting a capture, I receive a completely white image.
If I add a delay of 100ms, the image is almost black with some areas recognizable where there should be bright objects in the photo.
The longer the delay after standby is, the better the image gets, but the whole image is shifted towards green. I have tried with delays up to 2 seconds, but the image still did not have the correct colors/white balance.
What can/must I do to improve this behavior? Fixed white balance? If yes: how?
How long does the sensor need after leaving standby, until the white balance is correct?
In the sensor datasheet, I could not find any information regarding this issue.
If i keep the sensor awake, every image has the correct colors, independent of the type of light source used.

Thank you in advance!