ArduCAM/Arduino #137: crashing on capture

hutchij, 2017-02-05:
I'm using the ov5642 camera shield with this ESP8266 board - "Makerfocus D1 Mini NodeMcu 4M Bytes Lua WIFI Development Board ESP8266 ESP-12F"

I'm running the sketch called Arducam_ESP8266_OV5642_capture.ino. 

I added this code to pass the startup test
  SPI.setDataMode (SPI_MODE2); //

I believe I have the latest ARDUCAM libraries from head. 

I then get this output resulting in a crash when I hit 10.0.0.13/capture

Any help? 
Thanks, John

ArduCAM Start!
OV5642 detected.

Connecting to cottage
...WiFi connected

10.0.0.13
Server started
CAM Capturing

Soft WDT reset

ctx: cont 
sp: 3fff2230 end: 3fff2520 offset: 01b0

>>>stack>>>
3fff23e0:  00000041 3fff0a30 3fff13d8 40206060  
3fff23f0:  0001740b 3fff0a30 00000008 40206094  
3fff2400:  00000010 3fff2460 3fff14c0 402060b0  
3fff2410:  3fff3ae4 00000001 3fff14c0 4020296e  
3fff2420:  3fff14a0 000001b7 000001b7 40204540  
3fff2430:  00000001 00000001 3fff3a04 40207776  
3fff2440:  00000000 3fffc6fc 3fff3a04 40204536  
3fff2450:  3fff3a04 3fff0a7c 3fff3a04 40204572  
3fff2460:  00000000 00000000 00000000 40206fbc  
3fff2470:  3fff3a04 3fff0a7c 3fff0a3c 40204605  
3fff2480:  3fff3ae4 0000000f 00000008 402035c0  
3fff2490:  3fff0a7c 3fff0a3c 3fff1500 00000001  
3fff24a0:  00000001 40203b7c 0000000f 3fff1500  
3fff24b0:  00000000 00000000 3fff0a3c 3fff14ec  
3fff24c0:  00000001 3fff0a60 3fff0a3c 4020483f  
3fff24d0:  3ffea6e0 00000000 000003e8 4020776c  
3fff24e0:  00000000 3fff3abc feefeffe feefeffe  
3fff24f0:  3fffdad0 00000000 3fff14e4 402027a4  
3fff2500:  3fffdad0 00000000 3fff14e4 402075a4  
3fff2510:  feefeffe feefeffe 3fff1500 40100718  
<<<stack<<<

supprot, 2017-02-05:
@hutchij 
Hi,
We have update our demo about ESP8266 board.
In our demo, the SPI mode is mode 0,Please try this demo
below:

// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM ESP8266 2MP/5MP camera.
// This demo was made for ArduCAM ESP8266 2MP/5MP Camera.
// It can take photo and send to the Web.
// It can take photo continuously as video streaming and send to the Web.
// The demo sketch will do the following tasks:
// 1. Set the camera to JPEG output mode.
// 2. if server.on("/capture", HTTP_GET, serverCapture),it can take photo and send to the Web.
// 3.if server.on("/stream", HTTP_GET, serverStream),it can take photo continuously as video 
//streaming and send to the Web.

// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM ESP8266 2MP/5MP camera
// and use Arduino IDE 1.6.8 compiler or above

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

//This demo can only work on OV2640_MINI_2MP or ARDUCAM_SHIELD_V2 platform.
#if !(defined (OV2640_MINI_2MP)||defined (OV5640_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP_PLUS) \
    || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) \
    ||(defined (ARDUCAM_SHIELD_V2) && (defined (OV2640_CAM) || defined (OV5640_CAM) || defined (OV5642_CAM))))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
// set GPIO16 as the slave select :
const int CS = 16;

//you can change the value of wifiType to select Station or AP mode.
//Default is AP mode.
int wifiType = 1; // 0:Station  1:AP

//AP mode configuration
//Default is arducam_esp8266.If you want,you can change the AP_aaid  to your favorite name
const char *AP_ssid = "arducam_esp8266"; 
//Default is no password.If you want to set password,put your password here
const char *AP_password = "";

//Station mode you should put your ssid and password
const char *ssid = "SSID"; // Put your SSID here
const char *password = "PASSWORD"; // Put your PASSWORD here

static const size_t bufferSize = 4096;
static uint8_t buffer[bufferSize] = {0xFF};
uint8_t temp = 0, temp_last = 0;
int i = 0;
bool is_header = false;

ESP8266WebServer server(80);
#if defined (OV2640_MINI_2MP) || defined (OV2640_CAM)
  ArduCAM myCAM(OV2640, CS);
#elif defined (OV5640_MINI_5MP_PLUS) || defined (OV5640_CAM)
  ArduCAM myCAM(OV5640, CS);
#elif defined (OV5642_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) ||(defined (OV5642_CAM))
  ArduCAM myCAM(OV5642, CS);
[... 281 lines cut ...]
Serial.println(AP_ssid);
Serial.print(F("The password is: "));
Serial.println(AP_password);

WiFi.mode(WIFI_AP);
WiFi.softAP(AP_ssid, AP_password);
Serial.println("");
Serial.println(WiFi.softAPIP());
}

// Start the server
server.on("/capture", HTTP_GET, serverCapture);
server.on("/stream", HTTP_GET, serverStream);
server.onNotFound(handleNotFound);
server.begin();
Serial.println(F("Server started"));
}

void loop() {
server.handleClient();
}

Let us know if you need more help.
Regards,
ArduCAM support team.

hutchij, 2017-02-07:
Thanks a lot! It's working now. I notice one typo in your sketch

else if ((temp == 0xD8) & (temp_last == 0xFF))

should be 

else if ((temp == 0xD8) && (temp_last == 0xFF))

closed 2017-02-07 by hutchij

supprot, 2017-02-07:
@hutchij 
Hi,
Congratulations.
Very thanks for your advice,
Regards,
ArduCAM support team.

hutchij, 2017-02-08:
Hi, do you have any sketches showing how to access the RAW data from the sensor? I

supprot, 2017-02-08:
@hutchij 
Hi,
We are developling sketches about RAW data on ArduCAM MINI 5MP Plus.
When we finished we will give you a reply.
Now, only 5MP_Plus sopport big resolution RAW.

Regards,
ArduCAM support team.

hutchij, 2017-02-14:
Hi, I'm happy to hear that you are working on this. Any idea when it will be ready? Should I wait or try to develop this myself? 

Thanks again for the great support! John

supprot, 2017-02-15:
@hutchij 
Hi,
We are developing on the 5MP Plus and we have can capture a 2592x1944 image in raw format.
this is our config :+1: 
const struct sensor_reg ov5642_RAW[] PROGMEM =
{
{0x3103,0x03},
{0x3008,0x82},
{0x3017,0x7f},
{0x3018,0xfc},
{0x3810,0xc2},
{0x3615,0xf0},
{0x3000,0x00},
{0x3001,0x00},
{0x3002,0x00},
{0x3003,0x00},
{0x3011,0x08},
{0x3010,0x30},
{0x3604,0x60},
{0x3622,0x08},
{0x3621,0x17},
{0x3709,0x00},
{0x4000,0x21},
{0x401d,0x02},
{0x3600,0x54},
{0x3605,0x04},
{0x3606,0x3f},
{0x3c01,0x80},
{0x300d,0x21},
{0x3623,0x22},
{0x5000,0xcf},
{0x5001,0xFF},
{0x5020,0x04},
{0x5181,0x79},
{0x5182,0x00},
{0x5185,0x22},
{0x5197,0x01},
{0x5500,0x0a},
{0x5504,0x00},
{0x5505,0x7f},
{0x5080,0x08},
{0x300e,0x18},
{0x4610,0x00},
{0x471d,0x05},
{0x4708,0x06},
{0x3710,0x10},
{0x370d,0x06},
{0x3632,0x41},
{0x3702,0x40},
{0x3620,0x37},
{0x3631,0x01},
{0x370c,0xa0},
{0x3808,0x0a},
{0x3809,0x20},
{0x380a,0x07},
{0x380b,0x98},
{0x380c,0x0c},
{0x380d,0x80},
{0x380e,0x07},
{0x380f,0xd0},
{0x5000,0x06},
{0x501f,0x03},
{0x3503,0x07},
{0x3501,0x73},
{0x3502,0x80},
{0x350b,0x00},
{0x3818,0xc0},
{0x3621,0x27},
{0x3801,0x8a},
{0x3a00,0x78},
{0x3a1a,0x04},
{0x3a13,0x30},
{0x3a18,0x00},
{0x3a19,0x7c},
{0x3a08,0x12},
{0x3a09,0xc0},
{0x3a0a,0x0f},
{0x3a0b,0xa0},
{0x3004,0xff},
{0x350c,0x07},
{0x350d,0xd0},
{0x3a0d,0x08},
{0x3a0e,0x06},
{0x3500,0x00},
{0x3501,0x00},
{0x3502,0x00},
{0x350a,0x00},
{0x350b,0x00},
{0x3503,0x00},
{0x3030,0x2b},
{0x3a02,0x00},
{0x3a03,0x7d},
{0x3a04,0x00},
{0x3a14,0x00},
{0x3a15,0x7d},
{0x3a16,0x00},
{0x3a00,0x78},
{0x3a08,0x09},
{0x3a09,0x60},
{0x3a0a,0x07},
{0x3a0b,0xd0},
{0x3a0d,0x10},
{0x3a0e,0x0d},
{0x3620,0x57},
{0x3703,0x98},
{0x3704,0x1c},
{0x589b,0x00},
{0x589a,0xc0},
{0x3633,0x07},
{0x3702,0x10},
{0x3703,0xb2},
{0x3704,0x18},
{0x370b,0x40},
{0x370d,0x02},
{0x3620,0x52},
{0x5000,0x06},
{0x5001,0xff},
{0x5005,0x00},
{0x3818,0x80},
{0x3621,0x17},
{0x3801,0xb4},
{0x3001,0x40},
{0x3002,0x1c},
{0x3810,0x00},
{0x3818,0x00},
{0x460c,0x20},
{0x501f,0x03},
{0x4300,0xf8},

{0xffff,0xff},	
};
Let us know if you have more suggestions.
Regards,
ArduCAM support team.

taf2, 2020-09-28:
For anyone using platform.io memorysaver.h unfortunately hard codes the #define OV5642_MINI_5MP_PLUS

I found i have to copy the files into my project make the necessary #define change and then it works... very frustrating but very rewaring once it works