ArduCAM/Arduino #138: arducam 0v02640 with multicamera adapter board not working

Kiki-Sarpong1, 2017-02-06:
Hi 
i recently bought the arducam ovo2640 camera and the multi camera adapter board but it doesnt seem to work. i downloaded the pc software, looked online for help and made some adjustments but i still get no video feed no matter what piece of code from the arducam library i use .  I normally use code from arducam mini but the pc software posts some kind of gibberish (to me) after "port opened".
for instance , if I run (arduca_mini_video_streaming), I get(image posted below)   

[image]

supprot, 2017-02-07:
@Kiki-Sarpong1 
Hi,
We have update our host_V1to host_V2 please use our ArduCAM_Mini_QT which is at ../ArduCAM/example/host_app/ArduCAM_Host_V2_Windows folder.

Let us know if you need more help.
Best regards,
ArduCAM support team.

Kiki-Sarpong1, 2017-02-07:
excuse me , but i really don't understand what any of that means

supprot, 2017-02-07:
@Kiki-Sarpong1 
Hi,
Please to the link and downlad the Host_V2.0_Windows.
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app/ArduCAM_Host_V2.0_Windows
Let us know if you need more help.
ArduCAM support team.

Kiki-Sarpong1, 2017-02-07:
i downloaded it but anytime i try to run it, i keep getting these messages
i tried every solution i found online but it never works
[image]

supprot, 2017-02-08:
@Kiki-Sarpong1 
Hi,
You should download the entire ArduCAM_Host_V2.0_Windows  folder,
Please use this link : https://github.com/ArduCAM/Arduino.git
Let us know if you need more help.
Regards,
ArduCAM support team.

Kiki-Sarpong1, 2017-02-08:
I'm still getting "SPI INTERFACE ERROR"
even with the new pc software. 
[image]

Kiki-Sarpong1, 2017-02-08:
[image]

supprot, 2017-02-09:
@Kiki-Sarpong1 
For the first problem:
Please ensure your SPI CS firstly.In our demo we set the SPI CS 7 which means you should connect the camera to the Arducam mini 4in the adapter .Look at the picture
[image]
For the second problem:
About our ArduCAM_MINI_4CAM_VideoStreaming, we are developing our host for it. If you want to use ArduCAM_Host_V2 in the ArduCAM_MINI_4CAM_VideoStreaming demo, you should change  case 0x15: to case 0x10: in the ArduCAM_MINI_4CAM_VideoStreaming demo.
Look at the picture:
[image]

Kiki-Sarpong1, 2017-02-09:
Here are my results after  applying all the changes. It says this but still doesn't show any feed
thanks for all your help by the way .
[image]

supprot, 2017-02-09:
@Kiki-Sarpong1 
Hi,
 You should use the ArduCAM_Host_V2 host.
Please look at the picture we tested.
[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

Kiki-Sarpong1, 2017-02-09:
Thanks soo much i finally got it to work but why does it say this 
[image]

supprot, 2017-02-10:
@Kiki-Sarpong1 
Hi,
Your image datas is corrupt.Please try plug your camera to our adapter again.
If the phenomenon is still exist, you can print the image datas to the serial port 
and check if there are 0xff 0xd8 at the header of the image.

Let us know if you need more help.
ArduCAM support team.

Kiki-Sarpong1, 2017-02-15:
i tried plugging it in the adapter again and that didn't help
how do i print the image datas ?
where in the code do i add "Serial.println"

supprot, 2017-02-15:
@Kiki-Sarpong1 
Hi,
Can the camera work in other demo?
You can add Serial.println(temp,HEX); 
behind   temp =  SPI.transfer(0x00);//read a byte from spi.

Let us know if you need more help.
Regards,
ArduCAM support team.

UMecamat, 2017-02-20:
Hi, 
I am running ubuntu 16.04 LTS. Can you please provide additional info on how to run ArduCAM_Host_V2 host in linux system? Where to download/extract the folder? I am trying to launch the  ArduCAM_Mini.QT as a standalone application (and, yes, I do have installed Qt creator... just in case)
Thank you