ArduCAM/Arduino #139: ArduCAM Mini with OV5642 - constant 2V level on MISO

GornAlex, 2017-02-06:
Hello,

My ArduCCAM module keeps MISO at the constant 2V level if it is not used. The camera module itself works fine. I can take pictures with the host application. However, I can't work with my SD card module because in my case all SPI MISO signals mixed together through the OR gate (SN74VLC1G332), so the high level of ArduCAM's MISO blocks all other devices. Is it the normal situation or I have a defective unit? Here is the picture of MISO signal from my oscilloscope: https://1drv.ms/i/s!AliVreQcv-vnjp8917Nodzoxx1OpSg

supprot, 2017-02-07:
@GornAlex
We have level shift circuit using MOSFET on the camera board, if the camera CS is high, all the SPI signal are floated and pull up by a 10K OHM resistor.
You can connect the SPI signal directly to other peripherals like SD card without any gate circuits. 
Hope this help.

donfelipex, 2017-03-03:
Yeah, I had this problem. I attempted to get an answer, here: 
https://github.com/ArduCAM/Arduino/issues/50
The designer seems to be oblivious to this flaw or completely unwilling to offer solutions. I made a workaround by adding buffer to filter this noise out running the camera at 2.8V. If you run it at 3.3V or lower this behavior is minimized, but the camera in general adds completely unacceptable noise on SPI channels. The other flaw would be that it won't function higher than 4MHz(though the documentation never states this) and the Arduino code uses the SD library, which slows the image capture down even further. Good Luck