ArduCAM/Arduino #140: ArduCAM-M-2MP ESP8266 Evaluation Kit doesn't work

muhlmmau, 2017-02-07:
Please help me , I followed your document "ArduCAM-M-2MP ESP8266 Evaluation Kit User Guide" to install and try to run arduCAM_ESP8266_Capture when compile shows this message:

/* 
ATENCIÓN: la librería ESP8266WiFi pretende ejecutarse sobre arquitectura(s) [esp8266] y puede ser incompatible con tu actual tarjeta la cual corre sobre arquitectura(s) [ArduCAM_ESP8266_UNO].
ATENCIÓN: la librería ESP8266WebServer pretende ejecutarse sobre arquitectura(s) [esp8266] y puede ser incompatible con tu actual tarjeta la cual corre sobre arquitectura(s) [ArduCAM_ESP8266_UNO].
ATENCIÓN: la librería Wire pretende ejecutarse sobre arquitectura(s) [esp8266] y puede ser incompatible con tu actual tarjeta la cual corre sobre arquitectura(s) [ArduCAM_ESP8266_UNO].
ATENCIÓN: la librería SPI pretende ejecutarse sobre arquitectura(s) [esp8266] y puede ser incompatible con tu actual tarjeta la cual corre sobre arquitectura(s) [ArduCAM_ESP8266_UNO].
*/

I think the problem is when change de memorysaver , should i change for my hardware this?:

#ifndef _MEMORYSAVER_
#define _MEMORYSAVER_
//Only when using raspberry,enable it
//#define RASPBERRY_PI
//There are two steps you need to modify in this file before normal compilation
//Only ArduCAM Shield series platform need to select camera module, ArduCAM-Mini series platform doesn't
//Step 1: select the hardware platform, only one at a time
#define OV2640_MINI_2MP
//#define OV5642_MINI_5MP
//#define OV5642_MINI_5MP_BIT_ROTATION_FIXED
//#define OV5642_MINI_5MP_PLUS
//#define OV5640_MINI_5MP_PLUS
//#define ARDUCAM_SHIELD_REVC	
//#define ARDUCAM_SHIELD_V2
//Step 2: Select one of the camera module, only one at a time
#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))
	//#define OV7660_CAM
	//#define OV7725_CAM
	//#define OV7670_CAM
	//#define OV7675_CAM
    //#define OV2640_CAM
	//#define OV3640_CAM
	 //#define OV5642_CAM
	//#define OV5640_CAM 
	//#define MT9D111A_CAM
	//#define MT9D111B_CAM
	//#define MT9M112_CAM
	//#define MT9V111_CAM	
	//#define MT9M001_CAM	
	//#define MT9T112_CAM
	//#define MT9D112_CAM
#endif 
#endif	//_MEMORYSAVER_

the preferences was install here :

C:\Users\Mauricio\AppData\Local\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.2\libraries\ArduCAM

The next are the other configurations :

[image]
[image]

supprot, 2017-02-07:
@muhlmmau 
Hi,
Just do not care those warnings which is because The official ESP8266 architecture is not compatible with other esp8266 board.
If you are using the MINI_2MP , just enable the OV2640_MINI_2MP in the memorysaver.h file.

Let us know if you need more help.
Regards,
ArduCAM support team.

muhlmmau, 2017-02-08:
I use this devices ; aren't your 8266 board and your camera Mini 2MP?

[image]
[image]

[image]

supprot, 2017-02-09:
@muhlmmau 
Hi,
Yes, what I mean is the official ESP8266 architecture is not compatible with us.
Let us know if you need more help,
Regards,
ArduCAM support team.