ArduCAM/Arduino #141: SPI Interface Error Arducam Mini 2MP

beeRitu, 2017-02-10:
Hi I have wired up my cam as shown below 
[image]
[image]
However I get SPI interface error. I am trying to run the low power mode example for Arducam mini and I want to read the images to my PC without using the SD card. I am using Arduino 101 board. Please help to troubleshoot.

supprot, 2017-02-10:
@beeRitu 
Hi,
Please ensure your SPI CS. In our demo we set the SPI CS 7 instead of 10.
Let us know if you need more help.

Regards,
ArduCAM support team.

beeRitu, 2017-02-10:
Hi,

Thanks for a quick response. I also tried connecting Arducam as shown in response to another issue https://cloud.githubusercontent.com/assets/24818249/22140195/cd3570e0-df25-11e6-8d73-2f9ebbb5abb1.jpeg. However, I still get the same output. What else could I be doing wrong?

Thanks.

supprot, 2017-02-10:
@beeRitu 
Hi,

Don't worry and we will test it and give you a reply as soon as quickly.

Regards,
ArduCAM support team.

supprot, 2017-02-10:
@beeRitu 
Hi,
We have tested it on Genuino 101 board,Please refer to our test:
[image]
[image]

supprot, 2017-02-10:
@beeRitu 
Hi,
We have tested it on Genuino 101 board.Please refer to our test:
[image]
[image]
Let us know if you need more help.
Regards,
ArduCAM support team.

beeRitu, 2017-02-10:
Hi,

Thank you for a quick response again. Please see the screenshot of my program (on the right, yours is on the left). It looks the same to me. Also shown are my pin connections. The only difference I see is that I am trying to see the output on serial monitor (with baud rate 9600, I don't see an option for baudrate 921800 as used in your host application). I tried to download the host application but is it only for windows platform? I am working on MAC.
[image]
[image]
[image]

Really appreciate your help with troubleshooting.

Thanks.

supprot, 2017-02-11:
@beeRitu 
Hi,

We also met this kind of situation, please  check the i2c interface hardware connection.
Restart 101 board,wait for a while,after waiting for serial interface stability  and give it a try.
It is recommended to use our adapter board.
Let us lnow if you need more help.

Regards,
ArduCAM support team.

beeRitu, 2017-02-13:
Hi,

Could you explain how exactly to check the i2c interface hardware connection? Also, what do you mean by using your adapter board?

As I showed you in the picture, I think my wiring connections are same as yours. I did try resetting the board but that does not help.

I do notice that the program ArduCAM_Mini_LowPowerMode sets the baudrate as below: 

#if defined(__SAM3X8E__)
  Wire1.begin();
  Serial.begin(115200);
#else
  Wire.begin();
  Serial.begin(921600);
#endif

However, Arduino 101 page https://www.arduino.cc/en/Guide/Arduino101 says  "The SoftwareSerial library allows serial communication on any of the 101's digital pins up to 57600 bps". Could that be a problem?

Kindly help.
Thanks.

supprot, 2017-02-13:
@beeRitu 
Hi,
Yes ,it seems the i2c is error at 101 board easily.But from our test we found that repower and wait about five second,it can work fine. Maybe because of the virtual serialport needing some time to stable.
We check I2c by reading the sensor id register .
About the adapter plate,please refer to the link:
http://www.arducam.com/arduino-panorama-photography-arducam/arducam-mini-multicam-adapter1/

Let us know if you need more help.
Rggards,
ArduCAM support team.

beeRitu, 2017-02-13:
Hi,

I have following questions for you:

1. I am curious how in your demo you used baud rate of 921800 when Arduino 101 spec says that the SoftwareSerial library allows serial communication on any of the 101's digital pins up to 57600 bps
2. I see the following lines in the code
#if defined (OV2640_MINI_2MP)
  while(1){
    //Check if the camera module type is OV2640
    myCAM.wrSensorReg8_8(0xff, 0x01);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
    if ((vid != 0x26 ) && (( pid != 0x41 ) || ( pid != 0x42 ))){
      Serial.println(F("ACK CMD Can't find OV2640 module!"));
      delay(1000);continue;
    }
    else{
      Serial.println(F("ACK CMD OV2640 detected."));break;
    } 
  }
However when I print the values on vid and pid they are 0x01 and 0xA8. If I change the values to these, it says OV2640 is detected. Is there a register spec sheet that I can have to go over the code line by line? For example, I want to know why we write 0x01 to 0xff. And why my vid and pid values are 0x01 and 0xA8 when expected to be 0x26 and 0x41 or 0x42. Please help to provide detailed documentation for me to debug the issue.

Thanks.

supprot, 2017-02-13:
@beeRitu 
Hi,
You can download our ArduCAM Software_Application_Note here:
http://www.arducam.com/downloads/shields/ArduCAM_Software_Application_Note.pdf
You can download the host application for MAC here:
arducam.com/downloads/app/ArduCAM_Host_V2.0_Mac.app.zip
At 101 board, the serialport is a vertial serialport which means it can not be limited by the baud rate.
You can change the baud to any value.
We tested the i2c timing sequence and found that when we open the serial without delay any time,the i2 timing sequence is not ack signal look at the picture:
[image]

But when we do some delay after power the board,it can work fine.Look at the picture:
[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

beeRitu, 2017-02-15:
Hi,

I am still having issues with this. However, I figured if I restart the Arduino 101 everytime (power off and on) then I get the right pin id. But now I am having issues with this part of the code
```

void loop() {
uint8_t temp = 0xff, temp_last = 0;
uint8_t start_capture = 0;
temp = Serial.read();
delay(1000);         //Ritu
Serial.println(temp);//Ritu
switch(temp)
{
  case 0:
```

It seems like I am not getting anything on serial read as the value of temp remains 0xff. What could be going on here?

supprot, 2017-02-15:
@beeRitu 
Hi,
Thank you for your suggestions.
Usually,we use the serial like this:
while (Serial.available()>0)
  {
    data= Serial.read();
}
Regards,
ArduCAM support team.

bilal697, 2021-07-29:
I got an error when I am trying to make the interface of ArduCam with Arduino Nano 33 BLE Sense. SPI Interface ERROR