ArduCAM/Arduino #142: OV2640 and infrared LED

lorenberg, 2017-02-10:
Hello Lee,
thanks for your help on my previous problem. Now I have another :)

I saw on your website http://www.arducam.com/camera-modules/2mp-ov2640/ that it is possible to take night photos using an IR led 940nm. I am actually trying to do this, but I still see only black images.
Do I need a special setting for using the OV2640 with infrared light?

Thank you for helping me,

Lorenzo

supprot, 2017-02-10:
@lorenberg 
Hi,
You do not have to do any other configurations,but you should use the lens without filter.

Let us know if you need more help.
Regards,
ArduCAM support team.

lorenberg, 2017-02-10:
Hi,
thanks for the reply...Do you have a suggestion on how I can remove the filter without breaking something?
Thank you,
Lorenzo

lorenberg, 2017-02-11:
Actually I have bought the module with esp8266 included, precisely this one 
http://www.uctronics.com/arducam-2mp-v2-mini-camera-shield-with-esp8266-nano-esp-12f-evaluation-kits.html

Does this supports the IR vision? I cannot find any filter to remove....

lorenberg, 2017-02-23:
Hi again,
can you please help me on how to remove the IR filter?

supprot, 2017-02-23:
@lorenberg 
Hi,
The default delivery is with the filter, cannot be removed.
You need to purchase and leave a message to our seller.
Let us know if you need more help.
Regards,
Arducam support team.

lorenberg, 2017-02-23:
Hello,
thanks a lot. Your seller is uctronics? Or should I contact another one?
Do you have a contact for this? Their contact form online does not work. Thank you

supprot, 2017-02-23:
@lorenberg 
Hi,
Please contact  sales@uctronics.com
Let us know if you need more help.
Regards,
ArduCAM support team.

wconnect, 2017-11-17:
IR　フィルターの無いArduino UNO Mega2560の品番と価格をお知らせ下さい

wconnect, 2017-11-17:
再送
IRフィルターのないArduino UNO Mega2560の品番と価格をお知らせ願います

UCTRONICS, 2017-11-17:
@wconnect
Please connect sales@arducam.com