ArduCAM/Arduino #143: Arducam Mini 2P and SPI

khaliddecroix, 2017-02-14:
Hello,

I'm trying to build an application on an STM32F207 using the Arducam documentation. I have wired the camera and verified the timing of the SPI signals using an oscilloscope. The camera does not output the proper values when reading from registers such as 0x00 or 0x40. MISO is often asserted as 3-4 bits during the address phase and whatever it asserts in the address phase is what it outputs for the output phase as well. This behavior persists irregardless of which SPI phase/polarity is asserted.

 I have read through the software and hardware datasheets, following their specifications, and have been unable to resolve this issue. 

`SPI spi(PB_5, PB_4, PA_5); // mosi, miso, sclk
DigitalOut cs(PA_4);

Serial serial(USBTX, USBRX); 

int main() {
    spi.frequency(100000);
    spi.format(8,0);
    int data = 0;
    
    while(1) {
        cs = 0;
        spi.write(0xff); wait_us(50); spi.write(0x00); spi.write(0x00);
        cs = 1;
        wait_ms(5);
        
    }
}`

supprot, 2017-02-14:
@khaliddecroix 

Hi,
Please send us your project? We will help you check it in detail.
Regards,
ArduCAM support team.

khaliddecroix, 2017-02-14:
That's pretty much all the code right there as we're only verifying communications at this point. All we need to do is take a JPEG really, but the SPI has been exhibiting the behavior previously mentioned.

khaliddecroix, 2017-02-15:
After further testing, it appears that the SPI bus is asleep or something and requires a dummy cycle before it will allow writing? I am not sure if that is the case, but the issue was resolved by using a dummy read cycle on the test register. So for now everything appears to be operating normally.