ArduCAM/Arduino #144: ESPNano OV2640 free pins

lorenberg, 2017-02-14:
Hello again,

in my project I would need to connect a few IR Leds to the EPSNano module's free pins for night vision.
At the same time, I would need a free pin to connect to a mosfet for automatic shut down after some time.

Can you please tell me if there is any free pin I can use for this purpose? I tried checking on the documentation, but pins 10,11 (indicated as free GPIO) do not work as they should.
Is it maybe a limitation of current on the pins output?
If this is not possible, the ESP8266 ARDUINO UNO module would have more free pins I could use?
Thank you

supprot, 2017-02-14:
@lorenberg 

Hi, there is not free pin at ESp8266 platform.

Regards,
ArduCAM support team.

lorenberg, 2017-02-14:
Ok....on the ESP8266 ARDUINO UNO module neither?

supprot, 2017-02-15:
@lorenberg 
Hi,

On the ESP8266 ARDUINO UNO module, 
except GPIO15 and NC pins,other pins can be used .

Let us know if you need more help.
ArduCAM support team.