ArduCAM/Arduino #145: ArduCam Mini - Arduino Zero/M0 Error

KoRba88, 2017-02-17:
When compiling example i get this error (newest library and IDE)

C:\Program Files (x86)\Arduino\libraries\ArduCAM/ArduCAM.h:431:2: error: 'regtype' does not name a type

  regtype *P_CS;

  ^

C:\Program Files (x86)\Arduino\libraries\ArduCAM/ArduCAM.h:432:2: error: 'regsize' does not name a type

  regsize B_CS;

  ^

exit status 1

supprot, 2017-02-19:
@KoRba88 
Hi,
Very sorry to hear your problem.
Our ArduCAM is not compatible with with the Arduino /Genuino zero platform temporarily.
We will optimize the code and give you a reply as soon as possible.

Let us konw if you need more help.
Regards,
ArduCAM support team.

KoRba88, 2017-02-19:
Thanks for the fast replay.

I use arducam mini for scientific project at my university so I will be very glad if you solve this issue soon

Also on the ESP32 I have the same error code. Mayby you will fix it too at one time ;] ?

supprot, 2017-02-20:
@KoRba88 
Hi,
Please try adding the code which is compatible with the arm platform in the ArduCAM.h file.

#if defined(__arm__)

#define cbi(reg, bitmask) *reg &= ~bitmask
#define sbi(reg, bitmask) *reg |= bitmask
#define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);
#define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);

#define cport(port, data) port &= data
#define sport(port, data) port |= data

#define swap(type, i, j) {type t = i; i = j; j = t;}

#define fontbyte(x) cfont.font[x]  

#define regtype volatile uint32_t
#define regsize uint32_t
#define PROGMEM

    #define pgm_read_byte(x)        (*((char *)x))
//  #define pgm_read_word(x)        (*((short *)(x & 0xfffffffe)))
    #define pgm_read_word(x)        ( ((*((unsigned char *)x + 1)) << 8) + (*((unsigned char *)x)))
    #define pgm_read_byte_near(x)   (*((char *)x))
    #define pgm_read_byte_far(x)    (*((char *)x))
//  #define pgm_read_word_near(x)   (*((short *)(x & 0xfffffffe))
//  #define pgm_read_word_far(x)    (*((short *)(x & 0xfffffffe)))
    #define pgm_read_word_near(x)   ( ((*((unsigned char *)x + 1)) << 8) + (*((unsigned char *)x)))
    #define pgm_read_word_far(x)    ( ((*((unsigned char *)x + 1)) << 8) + (*((unsigned char *)x))))
    #define PSTR(x)  x
  #if defined F
    #undef F
  #endif
  #define F(X) (X)
#endif	

Let us know if you need more help.
Regards,
ArduCAM support team.

KoRba88, 2017-02-20:
I added this code and nothing, the same error.

supprot, 2017-02-21:
@KoRba88 
Hi,
Please use this ArduCAM.h we send to you .
www.uctronics.com/download/ArduCAM.h
We have tested it and it can work fine.
Please look at the  pictures below:
[image]
[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

KoRba88, 2017-02-21:
Compiling witch 0 errors but serial monitor display only 
ACK CMD SPI1 interface Error!
ACK CMD SPI2 interface Error!
ACK CMD SPI3 interface Error!
ACK CMD SPI4 interface Error!

supprot, 2017-02-22:
@KoRba88 
Hi,
What's your SPI CS? 
In our demo,the default SPI CS is 4,5,6,7.
Let us know if you need more help.

Regards,
ArduCAM support team.

KoRba88, 2017-03-17:
I have CS pin ok, i tried 4,5,6,7 and nothing. Also I tried Arduino M0 and Sparkfun boards and the same error.

supprot, 2017-03-18:
@KoRba88 
Hi,
there is a vitual serial port which is turn from USB in 
the M0 board.After power on,please try to wait a moment.
Let us know if you need more help.
Regards,
ArduCAM support team.

ahmadrezarazian, 2018-02-21:
Hi dude
I use STM32F103C  , but I have same errors yet.
I'm so pleased if you fix it for this.

good luck, TanX

supprot, 2018-02-22:
@ahmadrezarazian 
Hi,
Maybe your SPI driver exist some errors.
Please send us your email address and we will send you our camera driver on STM32 platform.