ArduCAM/Arduino #146: OV2640 photo capture works but not video

rmcncal, 2017-02-21:
Hi 

I'm using  an OV2640 with an ArduCam ESP8266.  Example ArduCAM_ESP8266_V2_OV2640_Capture2SD sketch works fine to upload photos to the micro SD card but video for ArduCAM_ESP8266_V2_OV2640_Video2SD is consistently corrupted. Similarly ArduCAM_ESP8266_OV2640_Capture example sketch uploads photos over wifi but video is blank.

Thanks 

RMcN

supprot, 2017-02-21:
@rmcncal 
Hi,
We have update our esp8266 demo.
Please try our new example:

// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM ESP8266 2MP/5MP camera, and can run on any Arduino platform.
// This demo was made for ArduCAM ESP8266  2MP/5MP Camera.
// It can shoot video and store it into the SD card
// The demo sketch will do the following tasks
// 1. Set the camera to JPEG output mode.
// 2. Capture a JPEG photo and buffer the image to FIFO
// 3.Write AVI Header
// 4.Write the video data to the SD card
// 5.More updates AVI file header
// 6.close the file
//The file header introduction
//00-03 :RIFF
//04-07 :The size of the data
//08-0B :File identifier
//0C-0F :The first list of identification number
//10-13 :The size of the first list
//14-17 :The hdr1 of identification
//18-1B :Hdr1 contains avih piece of identification 
//1C-1F :The size of the avih
//20-23 :Maintain time per frame picture

// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM ESP8266 2MP/5MP camera
// and use Arduino IDE 1.6.8 compiler or above

#include <SD.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

//This demo can only work on OV2640_MINI_2MP or ARDUCAM_SHIELD_V2 platform.
#if !(defined (OV2640_MINI_2MP)||defined (OV5640_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP_PLUS) \
    || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) \
    ||(defined (ARDUCAM_SHIELD_V2) && (defined (OV2640_CAM) || defined (OV5640_CAM) || defined (OV5642_CAM))))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
// Version 2, set pin 0 as the slave select for SD:
#define SD_CS 0
#define pic_num 200
#define rate 0x05
//set pin 16 as the slave select for SPI:
const int CS = 16;
#define AVIOFFSET 240
unsigned long movi_size = 0;
unsigned long jpeg_size = 0;
const char zero_buf[4] = {0x00, 0x00, 0x00, 0x00};
const char avi_header[AVIOFFSET] PROGMEM ={
  0x52, 0x49, 0x46, 0x46, 0xD8, 0x01, 0x0E, 0x00, 0x41, 0x56, 0x49, 0x20, 0x4C, 0x49, 0x53, 0x54,
  0xD0, 0x00, 0x00, 0x00, 0x68, 0x64, 0x72, 0x6C, 0x61, 0x76, 0x69, 0x68, 0x38, 0x00, 0x00, 0x00,
  0xA0, 0x86, 0x01, 0x00, 0x80, 0x66, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00,
  0x64, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x84, 0x00, 0x00, 0x00,
  0x73, 0x74, 0x72, 0x6C, 0x73, 0x74, 0x72, 0x68, 0x30, 0x00, 0x00, 0x00, 0x76, 0x69, 0x64, 0x73,
  0x4D, 0x4A, 0x50, 0x47, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x01, 0x00, 0x00, 0x00, rate, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0A, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x73, 0x74, 0x72, 0x66,
  0x28, 0x00, 0x00, 0x00, 0x28, 0x00, 0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00,
  0x01, 0x00, 0x18, 0x00, 0x4D, 0x4A, 0x50, 0x47, 0x00, 0x84, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00,
[... 208 lines cut ...]
  #elif defined (OV5640_MINI_5MP_PLUS) || defined (OV5640_CAM)
    myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
    myCAM.OV5640_set_JPEG_size(OV5640_320x240);
  #elif defined (OV5642_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) ||(defined (OV5642_CAM))
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
    myCAM.OV5642_set_JPEG_size(OV5642_320x240);  
  #endif

  //Initialize SD Card
  if (!SD.begin(SD_CS))
  {
    //while (1);    //If failed, stop here
    Serial.println(F("SD Card Error"));
  }
  else
    Serial.println(F("SD Card detected!"));
}
void loop(){  
   Video2SD();
  delay(5000);
}

Let us know if you need more help.
Regards,
Arducam support team.

rmcncal, 2017-02-22:
Hi

I tried your new code but I get the same issue, it records multiple AVI files on micro SD card but each one of the files I open is corrupt.

supprot, 2017-02-22:
@rmcncal 
Hi,
Vey sorry to hear your problem and we will test it and give you a reply 
as soon as quickly.
Regards,
ArduCAM support team.

supprot, 2017-02-23:
@rmcncal 
Hi,
We have tested the ArduCAM_ESP8266_V2_ Video2SD demo and it can work fine.
Please refer to the pictures and a AVI format video file  we send to you .
This is the AVI format video link:
www.arducam.com/downloads/support

Please refer to the pictures:
[image]
[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

rmcncal, 2017-02-25:
Yes that's the example I used. The code example on my arducam works the same way as in your pictures, it writes AVI files but each one is still corrupt. I tried a different micro SD card but it didn't work either.

supprot, 2017-02-26:
@rmcncal 
Hi,
Can your camera work fine when using other demos?
If so,please try to format your SD card and test again.

Let us know if you need more help,
ArduCAM dupport team.

rmcncal, 2017-02-27:
Yes camera works fine running Arducam_ESP8266_V2_OV2640_Capture2SD, it writes JPEG files to the micro SD and they all work ok. The only weird thing  I have noticed is the date of the files (both video and JPEG) are always 1/1/2000 01:00AM. I tried doing a quick reformat of the micro SD but same problem with the AVI files.

supprot, 2017-02-27:
@rmcncal 
Hi,
How about changing a video player?
We use the storm video player.
Let us know if you need more help.
Regards,
Arducam support team.

rmcncal, 2017-03-02:
I tried VLC player and it works but it says the AVI file index is broken. It wasn't working in Windows media player. Is there a way to fix this issue?

supprot, 2017-03-02:
@rmcncal 
Hi,

Very sorry for we have not find a way to solve this problem 
and we suggest you use the storm video player to try.

Let us know if you need more help.
Regards,
ArduCAM support team.

shibshib, 2017-03-06:
I have the same issue with the ArduCAM_Shield_V2_Video2SD demo using the ArduCAM Shield V2. 

AVI files are always corrupted when I try to open them after recording. I've tried using the Storm player as well and the problem still persists. 

Please advise.

supprot, 2017-03-07:
@shibshib 
Hi, 
Can your camera work fine on other platform?
ArduCAM support team.

shibshib, 2017-03-07:
@supprot It takes pictures no problem.

I've fixed the problem by adapting the updated ESP8266 code to the VIDEO2SD demo. I also use ffmpeg to fix the corrupted video but it looks like this:

[image]

This is when I used a higher frame rate than 5fps. Is there any way to increase it? It says in the Arducam catalogue that it is possible to do QVGA with 120fps. 

Thanks.

supprot, 2017-03-08:
@shibshib 
Hi,
Can you provide us with your codes and we will test it for you.
Best regards, 
ArdyCAM support team.

shibshib, 2017-03-08:
Yeah this is what I have:

//The file header introduction
//00-03 :RIFF
//04-07 :The size of the data
//08-0B :File identifier
//0C-0F :The first list of identification number
//10-13 :The size of the first list
//14-17 :The hdr1 of identification
//18-1B :Hdr1 contains avih piece of identification 
//1C-1F :The size of the avih
//20-23 :Maintain time per frame picture
// This program requires the ArduCAM V4.0.0 (or later) library and ARDUCAM_SHIELD_REVC
// and use Arduino IDE 1.6.8 compiler or above

#include <SD.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"

#define SD_CS 4
const int SPI_CS =6;

#define rate 0x06
// set the num of picture
#define num_frames 200

#define AVIOFFSET 240
bool is_header = false;
unsigned long movi_size = 0;
unsigned long jpeg_size = 0;
const char zero_buf[4] = {0x00, 0x00, 0x00, 0x00};
const char avi_header[AVIOFFSET] PROGMEM ={
  0x52, 0x49, 0x46, 0x46, 0xD8, 0x01, 0x0E, 0x00, 0x41, 0x56, 0x49, 0x20, 0x4C, 0x49, 0x53, 0x54,
  0xD0, 0x00, 0x00, 0x00, 0x68, 0x64, 0x72, 0x6C, 0x61, 0x76, 0x69, 0x68, 0x38, 0x00, 0x00, 0x00,
  0xA0, 0x86, 0x01, 0x00, 0x80, 0x66, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00,
  0x64, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x84, 0x00, 0x00, 0x00,
  0x73, 0x74, 0x72, 0x6C, 0x73, 0x74, 0x72, 0x68, 0x30, 0x00, 0x00, 0x00, 0x76, 0x69, 0x64, 0x73,
  0x4D, 0x4A, 0x50, 0x47, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x01, 0x00, 0x00, 0x00, rate, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0A, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x73, 0x74, 0x72, 0x66,
  0x28, 0x00, 0x00, 0x00, 0x28, 0x00, 0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00,
  0x01, 0x00, 0x18, 0x00, 0x4D, 0x4A, 0x50, 0x47, 0x00, 0x84, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54,
  0x10, 0x00, 0x00, 0x00, 0x6F, 0x64, 0x6D, 0x6C, 0x64, 0x6D, 0x6C, 0x68, 0x04, 0x00, 0x00, 0x00,
  0x64, 0x00, 0x00, 0x00, 0x4C, 0x49, 0x53, 0x54, 0x00, 0x01, 0x0E, 0x00, 0x6D, 0x6F, 0x76, 0x69,
};

ArduCAM myCAM( OV5642, SPI_CS );

void print_quartet(unsigned long i,File fd){
  fd.write(i % 0x100);  i = i >> 8;   //i /= 0x100;
  fd.write(i % 0x100);  i = i >> 8;   //i /= 0x100;
  fd.write(i % 0x100);  i = i >> 8;   //i /= 0x100;
  fd.write(i % 0x100);
}

void Video2SD(){
char str[8];
File outFile;
byte buf[256];
static int i = 0;
static int k = 0;
uint8_t temp = 0, temp_last = 0;
unsigned long position = 0;
uint32_t length = 0;
uint16_t frame_cnt = 0;
[... 162 lines cut ...]
  while(1){
    //Check if the camera module type is OV5642
    myCAM.wrSensorReg16_8(0xff, 0x01);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
    if((vid != 0x56) || (pid != 0x42)){
      Serial.println("Can't find OV5642 module!");
      delay(1000);continue;
    } else{
      Serial.println("OV5642 up in da hood."); break;
    }
  }
  
  myCAM.set_format(JPEG);
  myCAM.InitCAM();
  
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5642_320x240);
  
  delay(1000);
}
void loop(){
  Video2SD();
  delay(5000);
}

shibshib, 2017-03-09:
Thanks guys. My question is how do i change the frame rate? Currently its at 5 fps and i need to go up to ~10fps...

 Changing the "rate" variable only created a faster moving video, as if it was fast forwarding.

supprot, 2017-03-09:
@shibshib 
Hi,

The frame rate is mainly subject to writting SD card time 
and frame rate ot the camera itsef.
You can not change it by just changing the "rate"varible.
If you want to change the frame rate, you should optimize the writting 
SD card speed.

Let us know if you need more help.
Regards,
ArduCAM support team.

shibshib, 2017-03-09:
Can you give me an example on how to do this?

supprot, 2017-03-10:
@shibshib 
Hi,
It has a lot to do with the performance of the processor. 
At Arduino platform, it's very hard to get high frame rate.
At STM32 platform, our 2MP's frame rate can up to 15pfs.
Let us know if you need more help.
Regards,
ArduCAM support team.

hanyoung918, 2019-05-22:
hi, i've bought the Arducam ESP8266 WiFi Camera -2M Pixel and SD card.
i have a arduino mega 2560 and yet, i don't know where to put these SD card socket and my arducam in order to make the code work. if possible, can you let me know where to put pins in?