ArduCAM/Arduino #147: ArduCAM Host app with Ubuntu 16.04

UMecamat, 2017-02-21:
Hi,
I am running ubuntu 16.04 LTS. Can you please provide additional info on how to run ArduCAM_Host_V2 host in linux system? Is there any mandatory directory/folder where to download/extract the app folder? I am trying to launch the ArduCAM_Mini.QT as a stand-alone application (and, yes, I do have installed Qt creator... just in case)
Thank you

supprot, 2017-02-22:
@UMecamat 
Hi,
Please refer to the link
www.arducam.com/downloads/app//arducam.com/downloads/app
to download the host_V2 which can work on linux system.

But you should run this command first:
1. Open a terminal and into the software in the folder
2. Input: sudo chmod + x run. Sh / / to run. Sh scripts run permissions
3. Input: sudo. / run / / run the script to the software running depend on file configuration link
4. Input: sudo chmod u + x ArduCAM_Host_V2. 0 (software) / / to the software operation permissions
5. Input: sudo chmod u + x ArduCAM_Host_V2. 0. Sh / / to the software scripts run permissions
6. Input: sudo. / ArduCAM_Mini_QT. Sh / / running software

Let us know if you need more help.
Regards,
ArduCAM support team.

UMecamat, 2017-02-22:
Hi,
Thank you for the response, I finally was able to run the host_V2 for linux. However, I still have problems with both ArduCAM_MINI_4CAM_Capture2SD and ArduCAM_MINI_4CAM_VideoStreaming examples. I am using a Mega + Arducam adapter board + 2 ArduCAM MINI_5MP (in both sketches I excluded CAM3 and CAM4, just to be sure not to have the SPI not detected error...).  
As for the 2SD sketch, when opening the ArduCAM_Mini_QT.sh using linux terminal (sudo ./ArduCAM_Mini_QT.sh) the following happen:

1) the host_V2 opens
2) but in linux terminal I get the following message: 
`./ArduCAM_Mini_QT.sh: 5: [: .: unexpected operator
QMetaObject::connectSlotsByName: Connecting slot on_comboBox_pix_currentIndexChanged() with the first of the following compatible signals: ("currentIndexChanged(int)", "currentIndexChanged(QString)")
QMetaObject::connectSlotsByName: Connecting slot on_comboBox_function_currentIndexChanged() with the first of the following compatible signals: ("currentIndexChanged(int)", "currentIndexChanged(QString)")
QMetaObject::connectSlotsByName: Connecting slot on_comboBox_type_currentIndexChanged() with the first of the following compatible signals: ("currentIndexChanged(int)", "currentIndexChanged(QString)")`

After setting parameters in host_V2, (same baudrate, single mode, ov5642) I push the open button I get the following message in linux terminal:
`bool QSerialPort::setDataTerminalReady(bool): device not open`

and in host_V2 terminal always the same "SD Card Error" (I also tried 3 different SD cards...)

As for the streaming sketch, things seems better (but I always get the message at point 2 above when opening the host_V2). As you can see in the screenshot below, in single mode I can take pictures. However, only the picture from CAM2 appears on the pop-up window, and the files are stored with .jpeg extension, but are in fact plain text documents... 
[image]

Trying to connect to the IP assigned automatically I always get a "request fail !" message. 
Any idea about all these issues (SD failing, streaming failing, only 1 cam capturing, jpeg are plain text files)?

Finally, an additional info for other users that might be interested in run ArduCAM with linux: do please note that the host_V2 for linux is not available at the link http://www.arducam.com/downloads/app/arducam.com/downloads/app you suggested, but can still be found at http://www.arducam.com/downloads/app/ArduCAM_Mini_V2.0_Linux_x86_64bit.zip 
Thank you very much in advance

supprot, 2017-02-23:
@UMecamat 
Hi,
Thank you for pointing out our problem and we have update our Host_V2 at http://www.arducam.com/downloads/app/ArduCAM_Mini_V2.0_Linux_x86_64bit.zip
link.
But at present our Host_v2 can not support ArduCAM_4CAM_VideoStreaming demo.
You can try ArduCAM_Mini_Video_Streaming demo .
We have tested it and Please refer the picture we send to you:
[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

supprot, 2017-02-23:
@UMecamat 
Hi,
Before running the softwae do the command firstly:
2. Input: sudo chmod + x run. Sh / / to run. Sh scripts run permissions
3. Input: sudo. / run. Sh / / run the script to the software running depend on file configuration link
4. Input: sudo chmod u + x ArduCAM_Mini_QT (software) / / to the software operation permissions
5. Input: sudo chmod u + x ArduCAM_ Mini_QT. Sh / / to the software scripts run permissions
6. Input: sudo. / ArduCAM_Mini_QT. Sh / / running software

Regards,
ArduCAM support team.

UMecamat, 2017-02-23:
Hi, 
thank you very much for these explanations about using Host_v2 with linux os and streaming options. 
However, 
1) I still get the "SD Card Error" when trying the  ArduCAM_MINI_4CAM_Capture2SD example
2) and - surprisingly - I get the "open file failed" if using ArduCAM_Mini_5MP_4CAM_TimeElapse2SD_LowPower example (from former examples now not uploaded on your GitHub rep, why?)

In both cases I try with two different ArduCAM adapter board and using different SD cards. Any help?
Thank you as always

UMecamat, 2017-02-24:
update: tried 2 OV2640 instead of 2 OV5642, still getting "SD Card Error" independently from microSD or ArduCAM Adapter Board used... very frustrating...

supprot, 2017-02-26:
@UMecamat 
Hi,
Very sorry to reply you so late because of busying other things.
Please don't worry and we will test it for you tomorrow.  
Regards,
ArduCAM suport team.

supprot, 2017-02-27:
@UMecamat 
Hi,
We have tested it and it can work fine.
Please refer to the picture to learn more details.
[image]

Let us know if you need more help.
ArduCAM support team.

UMecamat, 2017-02-27:
Hi again, 
thank you for replying me. However, as I tried to explain above, configuration is not the point: SD_CS pinout is fine, as well as memorysaver option, wiring, etc... Independently from the micro SD card I am using, and trying with two different but identical ArduCAM Adapter boards, as well as two pairs of 2MP and 5MP rev B, the result is always the same and it concerns a failure of SD card initialization... 
I'll let this thread open just in case, but I will open another one hopefully waiting for some other user that could have solve this issue

dilpreetdan, 2017-05-13:
[image]

I tried all 6 steps to run the linux version as stated later, but I QXcbConnection: Could not connect to display 

Please suggest.

shyam1992atri, 2017-06-26:
shyam@shyamprasadvatri:~/Downloads/ArduCAM_Mini_V2shyam@shyamprasadvatri:~/Downloads/ArduCAM_Mini_V2nux_x86_64bit$ sudo chmod +x run.Sh
[sudo] password for shyam: 
chmod: cannot access 'run.Sh': No such file or directory
shyam@shyamprasadvatri:~/Downloads/ArduCAM_Mini_V2.0_Linux_x86_64bit$ sudo chmod +x run.sh
shyam@shyamprasadvatri:~/Downloads/ArduCAM_Mini_V2.0_Linux_x86_64bit$ sudo ./run.sh
shyam@shyamprasadvatri:~/Downloads/ArduCAM_Mini_V2.0_Linux_x86_64bit$ sudo chmod u+x ArduCAM_Mini_QT
shyam@shyamprasadvatri:~/Downloads/ArduCAM_Mini_V2.0_Linux_x86_64bit$ sudo chmod u+x ArduCAM_Mini_QT
.sh
shyam@shyamprasadvatri:~/Downloads/ArduCAM_Mini_V2.0_Linux_x86_64bit$ sudo ./ArduCAM_Mini_QT.sh
./ArduCAM_Mini_QT.sh: 5: [: .: unexpected operator
./ArduCAM_Mini_QT: error while loading shared libraries: libQt5SerialPort.so.5: cannot open shared object file: No such file or directory
shyam@shyamprasadvatri:~/Downloads/ArduCAM_Mini_V2.0_Linux_x86_64bit$

shyam1992atri, 2017-06-26:
can anyone help debugging this thing?

UCTRONICS, 2017-06-28:
@shyam1992atri 
Hi,
Have you solved your problems?

dilpreetdan, 2017-06-28:
Can anyone assist in the problem I received in the post above? 

> QXcbConnection: Could not connect to display

UCTRONICS, 2017-06-29:
@shyam1992atri 

Hi,
We have add run.sh please try it again.
For ArduCAM_Mini_V2.0_Linux_x86_64bit, Please refer to this link:
www.arducam.com/downloads/app/ArduCAM_Mini_V2.0_Linux_x86_64bit.zip

Step1: Open the terminal and enter the folder of the software
Step2: Input：sudo chmod +x run.sh
Step3: Input：sudo ./run.sh
Step4: Input：sudo chmod u+x ArduCAM_Mini_QT（name of software）
Step5: Inout：sudo chmod u+x ArduCAM_Mini_QT.sh
Step6: Inout：sudo ./ArduCAM_Mini_QT.sh

JoustingLion, 2018-12-01:
When I do sudo ./run.sh, I get this
ln: target 'libQt5Core.so.5' is not a directory
ln: target 'libQt5Widgets.so.5' is not a directory
ln: target 'libQt5Gui.so.5' is not a directory
ln: target 'libQt5Network.so.5' is not a directory
ln: target 'libQt5SerialPort.so.5' is not a directory
ln: target 'libicudata.so.56' is not a directory
ln: target 'libicui18n.so.56' is not a directory
ln: target 'libicuuc.so.56' is not a directory
ln: target 'libQt5XcbQpa.so.5' is not a directory
ln: target 'libQt5DBus.so.5' is not a directory
./run.sh: 13: export: (1):: bad variable name

and no files are created

UCTRONICS, 2018-12-02:
Don't worry, tomorrow we will try it again.

atekgul, 2019-09-07:
Hello,
What is "（name of software）" in the shell file?

UCTRONICS, 2019-09-10:
@atekgul 
For ArduCAM_Mini_V2.0_Linux_x86_64bit, Please refer to this link:
www.arducam.com/downloads/app/ArduCAM_Mini_V2.0_Linux_x86_64bit.zip

Step1: Open the terminal and enter the folder of the software
Step2: Input：sudo chmod +x run.sh
Step3: Input：sudo ./run.sh
Step4: Input：sudo chmod u+x ArduCAM_Mini_QT（name of software）
Step5: Inout：sudo chmod u+x ArduCAM_Mini_QT.sh
Step6: Inout：sudo ./ArduCAM_Mini_QT.sh

atekgul, 2019-09-10:
Hi,
I have already downloaded ArduCAM_Mini_V2.0_Linux_x86_64bit.zip and applied the steps.
I run "sudo chmod u+x ArduCAM_Mini_QT" in step 4.
But when i run the shell file, i got an error message like "./ArduCAM_Mini_QT.sh: 5: [: .: unexpected error".
That is why i am asking what the term "**name of software**" means to check if I made something wrong.

russiancosmo, 2020-01-24:
Hi everyone,
just to make some steps clear:
1. you need to download and install the **libQt5SerialPort** libraries to Ubuntu. Do so by typing the following in a terminal window: **sudo apt-get install libqt5serialport5**

2. You need to edit the **run.sh** in a text editor and change the versions of the files ending with ".so.x.x.x", to whatever version you downloaded of the libraries. Check your version by doing a search in the file manager in your hard drive.
Example:
 ln -s $PWD/libQt5Core.so.5.**12.4** libQt5Core.so.5
ln -s $PWD/libQt5Widgets.so.5.**12.4** libQt5Widgets.so.5
ln -s $PWD/libQt5Gui.so.5.**12.4** libQt5Gui.so.5
ln -s $PWD/libQt5Network.so.5.**12.4** libQt5Network.so.5
ln -s $PWD/libQt5SerialPort.so.5.**12.4** libQt5SerialPort.so.5
ln -s $PWD/libicudata.so.**63.2** libicudata.so.56
ln -s $PWD/libicui18n.so.**63.2** libicui18n.so.56
ln -s $PWD/libicuuc.so.**63.2** libicuuc.so.56
ln -s $PWD/libQt5XcbQpa.so.5.**12.4** libQt5XcbQpa.so.5
ln -s $PWD/libQt5DBus.so.5.**12.4** libQt5DBus.so.5

export set LD_LIBRARY_PATH=$PWD:$LD_LIBRARY_PATH

3. Then run the following commands. AND FORGET THE "name of software" or "software". It's just confusing.
Step1: Open the terminal and enter the folder of the software
Step2: Input：sudo chmod +x run.sh
Step3: Input：sudo ./run.sh
Step4: Input：sudo chmod u+x ArduCAM_Mini_QT
Step5: Inout：sudo chmod u+x ArduCAM_Mini_QT.sh
Step6: Inout：sudo ./ArduCAM_Mini_QT.sh

4. The next time you need to run the SW just go to the path of the SW and type. You'll be asked for your root passwd everytime you need to change the permissions:
Step5: Inout：sudo chmod u+x ArduCAM_Mini_QT.sh
Step6: Inout：sudo ./ArduCAM_Mini_QT.sh

ralfleggett, 2020-10-04:
Is this issue now closed? I'm looking to buy a stereo camera HAT, but for me its mandatory for it to work on Ubuntu. Thanks

russiancosmo, 2020-10-05:
Hi, I did not receive any resolution from this. Maybe it was solved after
all of this time...