ArduCAM/Arduino #148: Arducam OV5642 Mini Rev. B detection issue

Mschindl301, 2017-02-25:
Hello,

I have the Arducam OV5642 Mini 5 MP Rev. B camera. I have connected to my Arduino Mega 2560, and have uploaded the most recent example code, ArduCAM_Mini_Video_Streaming. I had to change the baud rate from 921600 to 115200, but those were the only changes made. When I open the serial monitor, it states that SPI is ok, but then continuously runs the loop that OV2640 cannot be found. Any recommendations?

Thanks for the help!!

supprot, 2017-02-26:
@Mschindl301 
Hi,
please refer to the pictures below.
We think maybe it is due to existing bad hardware connection.
[image]
[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

Mschindl301, 2017-02-26:
I doubled checked all the connections, used both the standard pins and the ISP header to connect the camera, and even connected in to an Arduino Uno, and still had the same issue. A note when I do compile the sketch it compiles correctly, but I do get these messages. I don't know if that is affecting anything.

Build options changed, rebuilding all
C:\Users\Matthew\AppData\Local\Temp\arduino_modified_sketch_248897\ArduCAM_Mini_Video_Streaming.ino:36:1: warning: narrowing conversion of '240' from 'int' to 'const char' inside { } [-Wnarrowing]

 };

 ^

C:\Users\Matthew\AppData\Local\Temp\arduino_modified_sketch_248897\ArduCAM_Mini_Video_Streaming.ino:36:1: warning: narrowing conversion of '196' from 'int' to 'const char' inside { } [-Wnarrowing]

C:\Users\Matthew\AppData\Local\Temp\arduino_modified_sketch_248897\ArduCAM_Mini_Video_Streaming.ino:36:1: warning: narrowing conversion of '196' from 'int' to 'const char' inside { } [-Wnarrowing]

C:\Users\Matthew\AppData\Local\Temp\arduino_modified_sketch_248897\ArduCAM_Mini_Video_Streaming.ino:36:1: warning: narrowing conversion of '248' from 'int' to 'const char' inside { } [-Wnarrowing]

C:\Users\Matthew\AppData\Local\Temp\arduino_modified_sketch_248897\ArduCAM_Mini_Video_Streaming.ino:36:1: warning: narrowing conversion of '224' from 'int' to 'const char' inside { } [-Wnarrowing]

Sketch uses 10,280 bytes (4%) of program storage space. Maximum is 253,952 bytes.
Global variables use 405 bytes (4%) of dynamic memory, leaving 7,787 bytes for local variables. Maximum is 8,192 bytes.
Invalid library found in C:\Users\Matthew\Documents\Arduino\libraries\OV7670FIFO: C:\Users\Matthew\Documents\Arduino\libraries\OV7670FIFO
Invalid library found in C:\Users\Matthew\Documents\Arduino\libraries\OV7670FIFO: C:\Users\Matthew\Documents\Arduino\libraries\OV7670FIFO

buschnyj, 2018-01-05:
@Mschindl301 did you figure out your error? I am running into the same issue with Arducam setup.

supprot, 2018-01-06:
@Mschindl301 @buschnyj 
Do you configure your hardware platform in the memorysaver.h file?
Your camera is 5642, You shuld enable the  #define ARDUCAM_SHIELD_REVC	 and #define OV5642_CAM
What's more ,How do you connect your i2c interface of the camera to the mega2560?

buschnyj, 2018-01-08:
@supprot Yes, I correctly configured the memorysaver.h file. 

The code actually runs despite the compiling errors. 

The only change I made is using Digital pin 10 instead of 7:
`const int CS = 10`

Is that what you mean regarding your i<sup>2</sup>C question?

supprot, 2018-01-09:
@buschnyj
Hi,
The pin 7 and pin 10 have nothing to do with the i2c interface.
In the code, it detecetd the camera sensor by i2c interface.So, if you 
can't find the camera ,It is likely that the i2c interface is in trouble.
We advice you print the value of the pid and vid to check.

while(1){
    //Check if the camera module type is OV5642
    myCAM.wrSensorReg16_8(0xff, 0x01);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
Serial println(vid, HEX);Serial println(pid, HEX);
    if((vid != 0x56) || (pid != 0x42)){
      Serial.println(F("Can't find OV5642 module!"));
      delay(1000);continue;
    }
    else{
      Serial.println(F("OV5642 detected."));break;
    } 
  }

Regards,