ArduCAM/Arduino #149: SD Card Initialization failure with ArduCAM Adapter Board

UMecamat, 2017-02-27:
Hi all, 
This is my set up:
Arduino Mega and ArduCAM 4 Cameras Adapter Board to run two ArduCAM Mini rev B. 
Independently from the Adapter Board I use (I got 2), from the ArduCAM Mini I use (I tried with both 2 and 5MP cameras), from the microSD card (I tried with 4-5 different microSDs), I always get the same "SD Card Error" that concerns the failure of SD card initialisation. This failure happens always, independently of the example sketch I am using (even with former ArduCAM examples no longer in the github repo).
I noticed that without SDcard mounted the Adapter Board led blink properly, whilst with SDcard inserted, the led light is very (very very) low. Does anyone experienced the same very frustrating error?
Thanks

jwerbs1217, 2017-03-24:
Having same issue

supprot, 2017-03-24:
@UMecamat @jwerbs1217 
Hi,
Don't worry,what's the phenomenon when using only one camera with our adapter board on Arduino Mega platform?
What's your SD CS is ? 
Can you try to format your SD card?
Let us know if you need more help.
Regards,
ArduCAM support team.

UMecamat, 2017-03-24:
Hi, 
- The adapter board is inserted directly to the Arduino Mega, and the SD_CS is 9, as for your examples;
- Tried with example "ArduCam_Mini_Capture2SD", one ArduCAM Mini rev B 5MP, and it works attached to CS4 (SPI_CS = 7); however, a new, very frustrating proble occurs, as the saved images are all and always 320x240, despite changing resolution in code (and I tried each possible resolution given in OV5642.h file)!!

- Also tried example "ArduCam_Mini_4CAM_Capture2SD" example with 2 ArduCAM Mini rev B 5MP. Result is again "SD card error". In the code, I excluded CAM3 and CAM4 as follows: 

`#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
//This demo can only work on OV2640_MINI_2MP and  OV5642_MINI_5MP and OV5642_MINI_5MP_BIT_ROTATION_FIXED platform.
#if !(defined OV5642_MINI_5MP || defined OV5642_MINI_5MP_BIT_ROTATION_FIXED || defined OV2640_MINI_2MP)
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif

#define SD_CS 9

// Set pin 4,5,6,7 as the slave select for SPI:
const int CS1 = 4;
const int CS2 = 5;
//const int CS3 = 6;
//const int CS4 = 7;

bool CAM1_EXIST = false; 
bool CAM2_EXIST = false;
//bool CAM3_EXIST = false;
//bool CAM4_EXIST = false;

#if defined (OV2640_MINI_2MP)
ArduCAM myCAM1(OV2640, CS1);
ArduCAM myCAM2(OV2640, CS2);
//ArduCAM myCAM3(OV2640, CS3);
//ArduCAM myCAM4(OV2640, CS4);
#else
ArduCAM myCAM1(OV5642, CS1);
ArduCAM myCAM2(OV5642, CS2);
//ArduCAM myCAM3(OV5642, CS3);
//ArduCAM myCAM4(OV5642, CS4);
#endif

void setup() {
  // put your setup code here, to run once:
  uint8_t vid, pid;
  uint8_t temp;
  Wire.begin(); 
 Serial.begin(115200);
 Serial.println(F("ArduCAM Start!")); 
// set the CS output:
pinMode(CS1, OUTPUT);
pinMode(CS2, OUTPUT);
//pinMode(CS3, OUTPUT);
//pinMode(CS4, OUTPUT);
pinMode(SD_CS, OUTPUT);
// initialize SPI:
SPI.begin(); 
  //Check if the 4 ArduCAM Mini 2MP Cameras' SPI bus is OK
  while(1){
  myCAM1.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM1.read_reg(ARDUCHIP_TEST1);
  if(temp != 0x55)
  {
    Serial.println(F("SPI1 interface Error!"));
  }else{
      CAM1_EXIST = true;
      Serial.println(F("SPI1 interface OK."));
    }
  myCAM2.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM2.read_reg(ARDUCHIP_TEST1);
  if(temp != 0x55)
[... 156 lines cut ...]
        buf[i++] = temp;
        else
        {
          //Write 256 bytes image data to file
          myCAM.CS_HIGH();
          outFile.write(buf, 256);
          i = 0;
          buf[i++] = temp;
          myCAM.CS_LOW();
          myCAM.set_fifo_burst();
        }        
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      buf[i++] = temp_last;
      buf[i++] = temp;   
    } 
  } 
}`

The same result if I use your original example without excluding CAM3 and CAM4 (which is obviously giving me a SPI3 and 4 error, but SPI1 and 2 are OK. 

This are, again, a very frustrating obstacles. Hope you will give some support without waiting so much time (25 days...)
All the best

supprot, 2017-03-24:
@UMecamat 
Hi,we wil test it Next Mondayand give you a reply.
Regards,

jwerbs1217, 2017-03-25:
So I opened the serial monitor and it looks like it's not getting past SD
card initialization. How do I set up my SD card? I've never done a project
with one? But more importantly for me, is there a way to have it send
information somewhere without an SD card first? For example, I'm trying to
send the photo through an XBEE to a remote controller. How might I go about
that? Thanks

UMecamat, 2017-03-25:
@jwerbs1217 
Hi, perhaps we can help each other waiting for some support from ArduCam guys. 
As far as I understand, and assuming your are using one of the sketches provided by ArduCam and the adapter board, the SD card is initialised by the 'SD.begin(SD_CS)' command. This initialization can fails if one or more of the following occur: 
- wrong wiring (by default the SD pin is 9, as defined in '#define SD_CS 9'); if you are using the adapter board directly connected to Arduino UNO or Mega, this should be automatic; otherwise, if you are using cables, make make sure SD goes to 9;
- SD card fails because not properly formatted: this can occur especially if your pc is running on Linux distro (as is mine). I burnt 2 micro SD before to understand this...; and, as far as I know, each time you insert the SD card into the adapter board, new images from ArduCam overwrite the old ones...

As for the XBEE, I have not experience about it... but, I could suggest you to have a look to the ArduCam ESP8266 mcu which is perhaps the simplest way to send images... or try to use an ESP8266 coupled to your Arduino (I recently tried a Node MCU with a single ArduCam mini 5MP to send images to a server - both local and remote - and it worked fine, just google it; also tried Arduino UNO/Mega coupled with a very simple ESP8266-05 and ArduCam mini 5MP, which was also fine). The only drawback for all these options is that apparently you can use just one ArduCam mini 5MP; no way to use two cameras together, unless you find out how to double the ISP. Next days I will try to use an Arduino Uno or Mega with the ArduCam adapter borad and an ESP8266-05 module acting as WiFi station. If you are interested, I'll let you know. 
Hope this can help you 
Best

supprot, 2017-03-28:
@UMecamat 
@jwerbs1217 
Hi,
We have tested the ArduCAM_MIni_4CAM_Capture2SD demo 
on MEGA 2560 board and it works fine.
We attach some picture for you.
Please check it again.
[image]
[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

UMecamat, 2017-03-28:
C'mon guys! One month waiting for some support and you send me such a ridiculous response? It really seems that you have not done any effort to check my previous message... Obviously the code pictured by you is the same I am using; obviously the Prints are different as when I use 2 cameras it does not work. Even more: as I said in ALL my messages, SD card error occur always and independently from the arduino (Uno or mega), from the SD card (I used a lot...), or adapter board (I have two). But all these issues were already explained in my previous comments. Do I should expect a real support from you? Any issue on SD cards perhaps not compatibles? Any problem with Linux users (I use Ubuntu)?

ArduCAM, 2017-03-28:
@UMecamat
From your previous message, I'm not sure if the SD card worked or not.
> Tried with example "ArduCam_Mini_Capture2SD", one ArduCAM Mini rev B 5MP, and it works attached to CS4 (SPI_CS = 7); however, a new, very frustrating proble occurs, as the saved images are all and [...]
If sd card works with ArduCam_Mini_Capture2SD, it should work with ArduCam_Mini_4CAM_Capture2SD example.
As you know the SD card is nothing to do with the camera, it should work alone even if no camera plugged.
Further more, you can use addtional sd card breakout board to your MEGA board without the 4camera adapter board.
Would you please show me some photos of your hardware setup?
Thanks.

UMecamat, 2017-03-28:
Hi, thanks for reply
I have not the hardware with me right now, but the hardware set up is as follow, as already explained:
> The adapter board is inserted directly to the Arduino Mega, and the SD_CS is 9, as for your examples;
so... no wiring problems. 
I understand that is not a SD problem. However, the SD is initialised and run only when 1 ArduCam Mini is plugged into the 4 cameras adapter board. When I plug another camera, I always receive a SD failure message. Another problem occurs when running just one camera: even if setting a new resolution, the images saved on the SD card are only 320x240 resolution  (and obviously I do not know if this occurs also when more than one camera is plugged on the adapter board). Last, thank your for the advice on additional sd card breakout, but having bought 2 ArduCam adapter board I would like to us their sd card slots, not adding more hardware. 
Thanks in advance

UMecamat, 2017-03-28:
sorry, wrong quoting in last message

ArduCAM, 2017-03-28:
@UMecamat,
Our engineer will look into the resolution issue tomorrow, it must be a bug.
In terms of the SD card problem, make sure each of the camera works perfectly with sd card before plugging two cameras. More cameras means more load to the SPI bus, if one of the camera hold the bus, the SPI bus will be occupied.

UMecamat, 2017-03-28:
Thanks, I'll check later if each cameras works with sd card before plugging 2 cameras, and wait for some fix about resolution problem.
Thanks

supprot, 2017-03-29:
@UMecamat 
Hi,
We have tested it about resolution problem and it can work fine.
How do you change it's resolition? If you are using 5MP , you just
 change  myCAM1.OV5642_set_JPEG_size(OV5642_320x240) in the demo to get the
 resolution you want.

If your problem still exist, please check your memorysaver.h file to ensure
 you have enable OV5642_MINI_5MP and disable others . Then check the 
ArduCAM.cpp file and try replace the 
void ArduCAM::OV5642_set_JPEG_size(uint8_t size) funtion  with
void ArduCAM::OV5642_set_JPEG_size(uint8_t size)
{
#if defined(OV5642_CAM) || defined(OV5642_CAM_BIT_ROTATION_FIXED)|| defined(OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_PLUS)
  uint8_t reg_val;

  switch (size)
  {
    case OV5642_320x240:
      wrSensorRegs16_8(ov5642_320x240);
      break;
    case OV5642_640x480:
      wrSensorRegs16_8(ov5642_640x480);
      break;
    case OV5642_1024x768:
      wrSensorRegs16_8(ov5642_1024x768);
      break;
    case OV5642_1280x960:
      wrSensorRegs16_8(ov5642_1280x960);
      break;
    case OV5642_1600x1200:
      wrSensorRegs16_8(ov5642_1600x1200);
      break;
    case OV5642_2048x1536:
      wrSensorRegs16_8(ov5642_2048x1536);
      break;
    case OV5642_2592x1944:
      wrSensorRegs16_8(ov5642_2592x1944);
      break;
    default:
      wrSensorRegs16_8(ov5642_320x240);
      break;
  }
#endif
}

[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

closed 2018-03-26 by UMecamat