ArduCAM/Arduino #151: compile library

SaraRua, 2017-03-07:
Hello,

I have a problem, i cant compile the examples because arduino ide doesnt recognize the library! I tried uploading the repository, and it says: 'The specified folder or ZIP file does not contain a valid library'.
Then i tried adding it manually, and when i compile it the ide says: 'Invalid library found in C:\(...)\ ArduCam'.

What could be the problem here?

Thanks in advance

supprot, 2017-03-07:
@SaraRua 
Hi,
You should put our ArduCAM library in the libraries folder which is under the path where your arduino ide is installed.
for example we put it in the path D:\Program Files (x86)\arduino3\libraries 

Let us know if you need more help.
Regards,
ArduCAM support team.

SaraRua, 2017-03-07:
I tried that already, it says the library is not valid. I have the arduino ide 1.8.1.

supprot, 2017-03-07:
@SaraRua 
Hi,
Please refer to the two pictures we attached for you.
[image]
[image]

Let us know if you need more help.
Regards,
ArduCAM support team.

SaraRua, 2017-03-07:
I did that. I guess something is missing, because the IDE doesnt recognize the files as a library!
Adding it manually (copy and dropping the folder directly to the library folder of arduino) gives this error

[image]

Trying to add the library directly in the IDE, by adding the zip files, gives this error.

[image]

The only way i could make it work was by adding the files manually in the .ino folder, like this 

[image]

supprot, 2017-03-08:
@SaraRua 
Hi,
Do you look at the first picture we attached carefully?
You should put ArduCAM library instead of Arduino-master in the libraries path.
The examples are in the example folder .
Please try again.
Let us know if you need more help.
Regards,
ArduCAM support team.