ArduCAM/Arduino #154: Arduino UNO Arducam 4 cameras adapter board problem

WilliamJohnsson, 2017-03-12:
Hello, 

Im trying to get 2 Arducam mini 2mp running on an Arduino UNO together with the Adapter board. Unfortunately I can't get any feed in the host app running on my macbook. Ive tried the ArduCam_Mini_Video_Streaming example and the ArduCam_Mini_4CAM_VideoStreaming. The memory saver.h looks fine and I guess the wiring is given since I'm using the adapter board straight on to the Arduino board. It mostly say something like this: 

"COM is open!

���瑓
囂�
 
ArduCAM single
"L'�
Lg"

Any clues? 

Best regards,

supprot, 2017-03-13:
@WilliamJohnsson 
Hi,

Very sorry to hear it.
Our host for Arducam 4cam has some bugs 
and we will optimize it as soon as possible.

Regards,
ArduCAM support team.

WilliamJohnsson, 2017-03-13:
Yeah I coult tell. It reminds me of this issue that I think was solved (?): 
https://github.com/ArduCAM/Arduino/issues/138

Maybe the hostapp for Mac is still the old one (1.0) ? 

Regards,