ArduCAM/Arduino #155: Arducam 2MP + ESP8266 Esp-12F Nano Crash

dduenker, 2017-03-13:
Hi there.

I have recently treated myself to a Arducam 2MP mounted on the ESP8266 Esp-12F Nano Board.
It worked at first, but while i played around with the sample code, to increase the resolution, i must have done something wrong, as the ESP8266 crashes now everytime, i try to take a picture.
I uploaded the original sketch for capturing and forgot to save my changes, so the code which caused it is lost.
I get a stack trace during the crash, but i have no idea, how to make sense of it currently. Any help concerning this issue would be appreciated.
Thanks.

`ArduCAM Start!
OV2640 detected.

Share AP: arducam_esp8266
The password is: 

192.168.4.1
Server started
CAM Capturing

Soft WDT reset

ctx: cont 
sp: 3fff1770 end: 3fff1a70 offset: 01b0

>>>stack>>>
3fff1920:  00000001 0000000d 3fff0a10 40206a19  
3fff1930:  00000041 3ffeff88 3fff0928 402060d8  
3fff1940:  001d87b1 3ffeff88 00000008 40206100  
3fff1950:  00000001 00000001 3fff0a10 4020611c  
3fff1960:  3fff1b6c 00000001 3fff0a10 402029d6  
3fff1970:  3fff09f0 0000011a 0000011a 402045a8  
3fff1980:  00000001 00000001 3fff2254 4020761a  
3fff1990:  00000000 00000000 3fff2254 4020459e  
3fff19a0:  3fff2254 3ffeffd4 3fff2254 402045da  
3fff19b0:  00000000 00000000 00000000 40206f00  
3fff19c0:  3fff2254 3ffeffd4 3ffeff94 40204669  
3fff19d0:  3fff1b6c 0000000f 00000008 40203680  
3fff19e0:  3ffeffd4 00000000 3fff0a50 00000001  
3fff19f0:  00000001 40203c10 00000010 3fff0a50  
3fff1a00:  00000000 00000000 3ffeff94 3fff0a3c  
3fff1a10:  00000001 3ffeffb8 3ffeff94 402048cf  
3fff1a20:  3ffe9510 00000000 000003e8 40207610  
3fff1a30:  00000000 3fff262c feefeffe feefeffe  
3fff1a40:  3fffdad0 00000000 3fff0a34 40202758  
3fff1a50:  3fffdad0 00000000 3fff0a34 402074e8  
3fff1a60:  feefeffe feefeffe 3fff0a50 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)`

supprot, 2017-03-14:
@dduenker 
Hi,

Please try to add some yield() when sending image by the wifi.
Let us know if you need more help.
Regards,
ArduCAM support team.

dduenker, 2017-03-14:
@supprot 
Thank you for your answer, this has not completely solved my problem, but now i have progressed from crashing to producing an infinite loop.

```c
void serverCapture(){
yield();
start_capture();
Serial.println(F("CAM Capturing"));
yield();
int total_time = 0;

total_time = millis();

Serial.println(F("while start"));
while (!myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK)){
  yield(); //<-- this never ends
}
Serial.println(F("while end"));
```

So the condition
```c
myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK)
```
never gets true.

Any hints on what to try next?
I tried setting the quality to a different setting, which sadly changed nothing so far.

Thank you.

supprot, 2017-03-14:
@dduenker 
Hi,
Don't worry, please send us your code. 
We will test it  for you and give you a 
reply as soon as possible.

Best regards,
ArduCAM support team.

dduenker, 2017-03-14:
Dearest support Team,

the code which produced the problem in the first place was overwritten by me, in an attempt to set everything back to normal, so it is lost for now.
I can give you the code  i am using now, but is just the example sketch with some added yields by now:

```c
// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM ESP8266 2MP/5MP camera.
// This demo was made for ArduCAM ESP8266 2MP/5MP Camera.
// It can take photo and send to the Web.
// It can take photo continuously as video streaming and send to the Web.
// The demo sketch will do the following tasks:
// 1. Set the camera to JPEG output mode.
// 2. if server.on("/capture", HTTP_GET, serverCapture),it can take photo and send to the Web.
// 3.if server.on("/stream", HTTP_GET, serverStream),it can take photo continuously as video 
//streaming and send to the Web.

// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM ESP8266 2MP/5MP camera
// and use Arduino IDE 1.6.8 compiler or above
#define OV2640_MINI_2MP
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

//This demo can only work on OV2640_MINI_2MP or ARDUCAM_SHIELD_V2 platform.
#if !(defined (OV2640_MINI_2MP)||defined (OV5640_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP_PLUS) \
    || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) \
    ||(defined (ARDUCAM_SHIELD_V2) && (defined (OV2640_CAM) || defined (OV5640_CAM) || defined (OV5642_CAM))))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
// set GPIO16 as the slave select :
const int CS = 16;

//you can change the value of wifiType to select Station or AP mode.
//Default is AP mode.
int wifiType = 1; // 0:Station  1:AP

//AP mode configuration
//Default is arducam_esp8266.If you want,you can change the AP_aaid  to your favorite name
const char *AP_ssid = "arducam_esp8266"; 
//Default is no password.If you want to set password,put your password here
const char *AP_password = "";

//Station mode you should put your ssid and password
const char *ssid = "SSID"; // Put your SSID here
const char *password = "PASSWORD"; // Put your PASSWORD here

static const size_t bufferSize = 4096;
static uint8_t buffer[bufferSize] = {0xFF};
uint8_t temp = 0, temp_last = 0;
int i = 0;
bool is_header = false;

ESP8266WebServer server(80);
#if defined (OV2640_MINI_2MP) || defined (OV2640_CAM)
  ArduCAM myCAM(OV2640, CS);
#elif defined (OV5640_MINI_5MP_PLUS) || defined (OV5640_CAM)
  ArduCAM myCAM(OV5640, CS);
#elif defined (OV5642_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) ||(defined (OV5642_CAM))
  ArduCAM myCAM(OV5642, CS);
[... 292 lines cut ...]
Serial.print(F("Share AP: "));
Serial.println(AP_ssid);
Serial.print(F("The password is: "));
Serial.println(AP_password);

WiFi.mode(WIFI_AP);
WiFi.softAP(AP_ssid, AP_password);
Serial.println("");
Serial.println(WiFi.softAPIP());
}

// Start the server
server.on("/capture", HTTP_GET, serverCapture);
server.on("/stream", HTTP_GET, serverStream);
server.onNotFound(handleNotFound);
server.begin();
Serial.println(F("Server started"));
}

void loop() {
server.handleClient();
}
```

Thank you.

supprot, 2017-03-15:
@dduenker 
Hi,

We have tested our demo with 1600x1200 resolution 
on our ESp8266 UNO board and it can work fine.
We attached a picture for you.
[image]
You stop at while (!myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK)); Maybe your camera work is not stable.Please ensure your hardware connection is stable and reset again.

Let us know if you need more help.
Regards,
ArduCAM support team.

dduenker, 2017-03-15:
Thanks again, i found what i did wrong: i forgot to set the 
```c
#define OV2640_MINI_2MP
``` 
in memorysaver.h

It works now. 😀
I have no idea, why it worked in the first place without this.

Best Regards,
Daniel

closed 2017-03-15 by dduenker

supprot, 2017-03-15:
@dduenker 
Hi,
So glad to hear you have solved the problem.
Best regards,
ArduCAM support team.