ArduCAM/Arduino #158: ArduCam Mini 5MP Plus and Raspberry PI

KoRba88, 2017-03-19:
Do ArduCam Mini 5MP Plus support Raspberry PI ? And if yes, can you give a code to take pictures from 4 camera modules?

supprot, 2017-03-19:
@KoRba88 
Hi,
Yes, you can download our ArduCAM library 
which can run on raspberry PI platform. 
You should enable RASPBERRY_PI in the memorysaver.h

Let us know if you need more help.
Regards,
ArduCAM support team.

KoRba88, 2017-04-07:
Hi

I have tested ArduCAM Mini 5mp and ArduCAM Mini 5MP PLUS. And PLUS version dont work. Could you update library and add PLUS support to raspberry pi ?

supprot, 2017-04-08:
@KoRba88 
Hi,
Our ArduCAM4Pi library has not update and please download our ArduCAM library.
Please run below command:
1. sudo apt-get install wiringpi
2. sudo apt-get install i2c-tools
3. sudo apt-get install libi2c-dev
4. sudo apt-get install python-smbus
Then 
cd /home/pi/ArduCAM/example/RaspberryPi
sudo make
sudo ./ov5642_capture -c test.jpg 320x240

Let us know if you need more help.
Regards,
ArduCAM support team.

VitalCoder, 2018-07-02:
Hi, 
I am getting issues with wiringpi 

When executing I am getting 

pinMode: You have not called one of the wiringPiSetup
  functions, so I'm aborting your program before it crashes anyway.