ArduCAM/Arduino #159: Single capture mode?

WilliamJohnsson, 2017-03-22:
Hello, 
I am now running 2 Arducam 2mp's on an adapter board together wth the Arduino uno. Im using the ArduCam_Mini_4CAM_Capture2SD example seen below. My end game is to implement a sonar based trigger but first of all I want to learn more about the software. As I run the example today, it starts to burst pictures. Just by booting it up and provide the USB-power it starts to take pictures and I can't find where the trigger is looped? I suppose it should go into the "read_reg" method with the argument "ARDUCHIP_TEST1", but what does the argument here really say? 

Bast regards

 `// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini camera, and can run on any Arduino platform.
//
// This demo was made for ArduCAM Mini Camera.
// It needs to be used in combination with PC software.It can take 4 photos at the same and save them to the SD card
// The demo sketch will do the following tasks:
// 1. Set the 4 cameras to JPEG output mode.
// 2. Read data from Serial port and deal with it
// 3. Save the images to SD card.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM Mini camera
// and use Arduino IDE 1.5.2 compiler or above

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
//This demo can only work on OV2640_MINI_2MP and  OV5642_MINI_5MP and OV5642_MINI_5MP_BIT_ROTATION_FIXED platform.
#if !(defined OV5642_MINI_5MP || defined OV5642_MINI_5MP_BIT_ROTATION_FIXED || defined OV2640_MINI_2MP)
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif

#define SD_CS 9

// Set pin 4,5,6,7 as the slave select for SPI:
const int CS1 = 4;
const int CS2 = 5;
const int CS3 = 6;
const int CS4 = 7;

bool CAM1_EXIST = false; 
bool CAM2_EXIST = false;
bool CAM3_EXIST = false;
bool CAM4_EXIST = false;
  

#if defined (OV2640_MINI_2MP)
ArduCAM myCAM1(OV2640, CS1);
ArduCAM myCAM2(OV2640, CS2);
ArduCAM myCAM3(OV2640, CS3);
ArduCAM myCAM4(OV2640, CS4);
#else
ArduCAM myCAM1(OV5642, CS1);
ArduCAM myCAM2(OV5642, CS2);
ArduCAM myCAM3(OV5642, CS3);
ArduCAM myCAM4(OV5642, CS4);
#endif

void setup() {
  // put your setup code here, to run once:
  uint8_t vid, pid;
  uint8_t temp;
  Wire.begin(); 
 Serial.begin(115200);
 Serial.println(F("ArduCAM Start!")); 
// set the CS output:
pinMode(CS1, OUTPUT);
pinMode(CS2, OUTPUT);
pinMode(CS3, OUTPUT);
pinMode(CS4, OUTPUT);
pinMode(SD_CS, OUTPUT);
// initialize SPI:
SPI.begin(); 
  //Check if the 4 ArduCAM Mini 2MP Cameras' SPI bus is OK
  while(1){
  myCAM1.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM1.read_reg(ARDUCHIP_TEST1);
  if(temp != 0x55)
  {
    Serial.println(F("SPI1 interface Error!"));
  }else{
      CAM1_EXIST = true;
      Serial.println(F("SPI1 interface OK."));
    }
  myCAM2.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM2.read_reg(ARDUCHIP_TEST1);
  if(temp != 0x55)
  {
    Serial.println("SPI2 interface Error!");
  }else{
      CAM2_EXIST = true;
      Serial.println(F("SPI2 interface OK."));
    }
  myCAM3.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM3.read_reg(ARDUCHIP_TEST1);
  if(temp != 0x55)
  {
    Serial.println(F("SPI3 interface Error!"));
  }else{
      CAM3_EXIST = true;
      Serial.println(F("SPI3 interface OK."));
    }
  myCAM4.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM4.read_reg(ARDUCHIP_TEST1);
  if(temp != 0x55)
  {
    Serial.println(F("SPI4 interface Error!"));
  }else{
      CAM4_EXIST = true;
      Serial.println(F("SPI4 interface OK."));
    }
    if(!(CAM1_EXIST||CAM2_EXIST||CAM3_EXIST||CAM4_EXIST)){
    delay(1000);continue;
    }else
    break;
  }
  //Initialize SD Card
  while(!SD.begin(SD_CS)){
    Serial.println(F("SD Card Error"));delay(1000);
  }
  Serial.println(F("SD Card detected."));
    
  #if defined (OV2640_MINI_2MP)
  while(1){
      //Check if the camera module type is OV2640
     myCAM1.wrSensorReg8_8(0xff, 0x01);
     myCAM1.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
     myCAM1.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
     if ((vid != 0x26 ) && (( pid != 0x41 ) || ( pid != 0x42 ))){
      Serial.println(F("Can't find OV2640 module!"));
      delay(1000);continue;
      }else{
         Serial.println(F("OV2640 detected."));break;
        }
    } 
  #else
  while(1){
   //Check if the camera module type is OV5642
    myCAM1.wrSensorReg16_8(0xff, 0x01);
    myCAM1.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM1.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
     if((vid != 0x56) || (pid != 0x42)){
       Serial.println(F("Can't find OV5642 module!"));
       delay(1000);continue;
      }else{
        Serial.println(F("OV5642 detected."));break;
     }   
  }
   #endif
    myCAM1.set_format(JPEG);
    myCAM1.InitCAM();
  #if defined (OV2640_MINI_2MP)
    myCAM1.OV2640_set_JPEG_size(OV2640_1600x1200);
  #else
    myCAM1.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
    myCAM2.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
    myCAM3.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
    myCAM4.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
    myCAM1.OV5642_set_JPEG_size(OV5642_320x240);
 #endif
  delay(1000);
  myCAM1.clear_fifo_flag();
  myCAM2.clear_fifo_flag();
  myCAM3.clear_fifo_flag();
  myCAM4.clear_fifo_flag();
}
void loop() {
  if(CAM1_EXIST)
    myCAMSaveToSDFile(myCAM1);
    if(CAM2_EXIST)
    myCAMSaveToSDFile(myCAM2);
    if(CAM3_EXIST)
    myCAMSaveToSDFile(myCAM3);
    if(CAM4_EXIST)
    myCAMSaveToSDFile(myCAM4);
    delay(5000);
  }
void myCAMSaveToSDFile(ArduCAM myCAM){
  char str[8];
  byte buf[256];
  static int i = 0;
  static int k = 0;
  uint8_t temp = 0,temp_last=0;
  uint32_t length = 0;
  bool is_header = false;
  File outFile;
  //Flush the FIFO
  myCAM.flush_fifo();
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  //Start capture
  myCAM.start_capture();
  Serial.println("start Capture");
 while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
 Serial.println(F("Capture Done."));  
 length = myCAM.read_fifo_length();
 Serial.print(F("The fifo length is :"));
 Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //384K
  {
    Serial.println(F("Over size."));
    return ;
  }
    if (length == 0 ) //0 kb
  {
    Serial.println(F("Size is 0."));
    return ;
  }
 //Construct a file name
 k = k + 1;
 itoa(k, str, 10);
 strcat(str, ".jpg");
 //Open the new file
 outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
 if(!outFile){
  Serial.println(F("File open faild"));
  return;
 }
myCAM.CS_LOW();
 myCAM.set_fifo_burst();
while ( length-- )
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    //Read JPEG data from FIFO
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
    {
        buf[i++] = temp;  //save the last  0XD9     
       //Write the remain bytes in the buffer
        myCAM.CS_HIGH();
        outFile.write(buf, i);    
      //Close the file
        outFile.close();
        Serial.println(F("Image save OK."));
        is_header = false;
        i = 0;
    }  
    if (is_header == true)
    { 
       //Write image data to buffer if not full
        if (i < 256)
        buf[i++] = temp;
        else
        {
          //Write 256 bytes image data to file
          myCAM.CS_HIGH();
          outFile.write(buf, 256);
          i = 0;
          buf[i++] = temp;
          myCAM.CS_LOW();
          myCAM.set_fifo_burst();
        }        
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      buf[i++] = temp_last;
      buf[i++] = temp;   
    } 
  } 
}`

supprot, 2017-03-23:
@WilliamJohnsson 
Hi,
What do you mean about "I can't find where the trigger is looped? I suppose it should go into the "read_reg" method with the argument "ARDUCHIP_TEST1", but what does the argument here really say?"
Our ArduCam_Mini_4CAM_Capture2SD demo just is timing capture and save it to the SD card.
What do you mean about ARDUCHIP_TEST1 and argument? Can you describe it in detail?

Let us know if you need more help.
Regards,
ArduCAM support team.

WilliamJohnsson, 2017-03-23:
Ok! So its just taking pictures with a delay? I thought the demo was connected to the "capture" button on the adapter board. Makes sense though since it should be Used for panorama pictures.

If I want it to take pictures after pressing the capture button, what part code should i modify?

Best regards

supprot, 2017-03-23:
@WilliamJohnsson 
Hi,
You can refer to our ArduCAM_Shield_V2_Camera_Playback demo which is to start capture by pressing the capture button.
// ArduCAM demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with a supported camera modules.
//This demo can only work on ARDUCAM_SHIELD_V2 platform.
//This demo is compatible with ESP8266
// This demo was made for Omnivision OV2640/OV5640/OV5642/ sensor.
// It will turn the ArduCAM into a real digital camera with capture and playback functions.
// 1. Preview the live video on LCD Screen.
// 2. Capture and buffer the image to FIFO when shutter pressed quickly.
// 3. Store the image to Micro SD/TF card with BMP format.
// 4. Playback the capture photos one by one when shutter buttom hold on for 3 seconds.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM shield V2
// and use Arduino IDE 1.6.8 compiler or above
#include <UTFT_SPI.h>
#include <SD.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
//This demo was made for Omnivision MT9D111A/MT9D111B/MT9M112/MT9V111_CAM/
//                                  MT9M001/MT9T112/MT9D112/OV7670/OV7675/
//                                  OV7725/OV2640/OV3640/OV5640/OV5642 sensor.
#if !(defined ARDUCAM_SHIELD_V2 && (defined MT9D111A_CAM|| defined MT9D111B_CAM || defined MT9M112_CAM  \
                                 || defined MT9V111_CAM || defined MT9M001_CAM || defined MT9T112_CAM \
                                 || defined MT9D112_CAM || defined OV7670_CAM  || defined OV7675_CAM  \
                                 || defined OV7725_CAM  || defined OV2640_CAM  || defined OV5640_CAM  \
                                 || defined OV5642_CAM|| defined OV3640_CAM))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#if defined(__arm__)
  #include <itoa.h>
#endif
#if defined(ESP8266)
 #define SD_CS 0 
 const int SPI_CS = 16;
#else 
 #define SD_CS 9 
 const int SPI_CS =10;
#endif
#define BMPIMAGEOFFSET 66
const int bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};
#if defined (MT9D111A_CAM)
  ArduCAM myCAM(MT9D111_A, SPI_CS);
#elif defined (MT9D111B_CAM)
  ArduCAM myCAM(MT9D111_B, SPI_CS);
#elif defined (MT9M112_CAM)
  ArduCAM myCAM(MT9M112, SPI_CS);
#elif defined (MT9V111_CAM)
  ArduCAM myCAM(MT9V111, SPI_CS);
#elif defined (MT9M001_CAM)
  ArduCAM myCAM(MT9M001, SPI_CS);
#elif defined (MT9T112_CAM)
  ArduCAM myCAM(MT9T112, SPI_CS);
#elif defined (MT9D112_CAM)
  ArduCAM myCAM(MT9D112, SPI_CS);
#elif defined (OV7670_CAM)
  ArduCAM myCAM(OV7670, SPI_CS);
#elif defined (OV7675_CAM)
  ArduCAM myCAM(OV7675, SPI_CS);
[... 218 lines cut ...]
}
//Only support RGB565 bmp format
void dispBitmap(File inFile)
{ 
  char VH = 0, VL = 0;
  int i, j = 0;
  for (i = 0 ; i < BMPIMAGEOFFSET; i++)
  inFile.read();
  for (i = 0; i < 320; i++)
  for (j = 0; j < 240; j++)
  {
    VL = inFile.read();
    //Serial.write(VL);
    VH = inFile.read();
    //Serial.write(VH);
    #if defined(ESP8266)
      yield();
    #endif
    myGLCD.LCD_Write_DATA(VH, VL);
  }
  myGLCD.clrXY();
}
Let us know if you need more help.
Regards,
ArduCAM support team.

WilliamJohnsson, 2017-03-25:
Hey,

I think I have to rephrase my question: IS there any easy (sort of) way to modify the ArduCAM_Mini_4CAM_Capture2SD example to take the pictures once triggered by an external signal ?

Thanks for a great job by the way.

Regards
William
23 mars 2017 kl. 15:22 skrev William Johnsson <WilliamJohnsson@hotmail.com<mailto:WilliamJohnsson@hotmail.com>>:

Thank you! That would probably work but I need to use two cameras at the same time? This exemple only defines 1 camera right?

Best regards

supprot, 2017-03-25:
@WilliamJohnsson 
Hi,
You can do like below code mode:
#define KEY 2  //#define your key
keyState= digitalRead(KEY);
if(!keyState)  {
  isCaptureFlag = true;   //to start capture signal 
  while(!digitalRead(KEY));  
}

Let us know if you need more help.
Regards,
ArduCAM support team.