ArduCAM/Arduino #161: OV2640 How to change JPEG compression quantizer?

wdouglas1999, 2017-04-01:
I purchased the esp8266 mini V2 shield with OV2640 camera kit. The examples all work fine,but now I would like to change the JPEG quantizer value to be able to change the size of the output file. I looked at the data sheet. Then I wrote different values to register address 0x44 but the image size and quality does not change.
Does anyone have any advice on how to achieve this? Any examples maybe?

lxxl, 2017-04-03:
Hey,
what about set_JPEG_size ?
myCAM.OV2640_set_JPEG_size(OV2640_1622x1200);

wdouglas1999, 2017-04-03:
Hi, thx for the reply. Changing the JPEG resolution works fine. Also, I managed to solve the quantizer problem. The issue was that I was using 16 bit address instead of 8 bit. Now it is working.

closed 2017-04-03 by wdouglas1999

ms1802, 2023-10-19:
Few years late, but if anyone is searching then here you go...
yes, Its the Quantization scale you need to change (as you said)
for ov2640 is reg at 0x44
for ov5640 is reg at 0x4407