ArduCAM/Arduino #163: OV2640 will not work at all...SPI Error (sometimes)

coolb2200, 2017-04-03:
I have tried using both an Arduino Mega and an Arduino UNO to interface the Arducam (OV2640) and both have proven to not work. The farthest I got was with the Mega. I would open the Host application with everything configured and pinned properly and it would say "COM is open!" then "OV2640 detected" then "SPI Error......". 

Currently all I can get is "COM is open!" and that's it. I have tried contacting support but they have since stopped responding to my emails for some reason; Tried a completely different OV2640 unit; Tried interfacing with the Arduino UNO R3. All avenues of troubleshooting seem to not be yielding any results. I NEED HELP!!!

My pin configuration with the Mega2560 is as follows:

CS -> Communication pin10
MOSI -> ICSP pin4
MISO -> ICSP pin1
SCK -> ICSP pin4
GND -> GND
VCC -> 5V
SDA -> A5
SCL -> A4

A little background on what im doing is I need to take a picture using the camera. I then need to take the RGB values of each individual picture and create some conditional logic with it. Im running out of time and Im truly baffled as to why this has been so hard to setup as everybody else seems to be having such an easy time using the ArduCam.

coolb2200, 2017-04-03:
After running this code (it fails) I am able to run the Host application and get to this: 

COM is open!
ArduCAM Start!

SPI Interface Error!
SPI Interface Error!
SPI Interface Error!
....................
....................

----------------------------------
Code I ran:

// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM Mini camera, and can run on any Arduino platform.
// This demo was made for ArduCAM_Mini_5MP_Plus.
// It needs to be used in combination with PC software.
// It can take photo continuously as video streaming.
//
// The demo sketch will do the following tasks:
// 1. Set the camera to JPEG output mode.
// 2. Read data from Serial port and deal with it
// 3. If receive 0x00-0x08,the resolution will be changed.
// 4. If receive 0x10,camera will capture a JPEG photo and buffer the image to FIFO.Then write datas to Serial port.
// 5. If receive 0x20,camera will capture JPEG photo and write datas continuously.Stop when receive 0x21.
// 6. If receive 0x30,camera will capture a BMP  photo and buffer the image to FIFO.Then write datas to Serial port.
// 7. If receive 0x11 ,set camera to JPEG output mode.
// 8. If receive 0x31 ,set camera to BMP  output mode.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM_Mini_5MP_Plus
// and use Arduino IDE 1.6.8 compiler or above
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
//This demo can only work on OV2640_MINI_2MP or OV5642_MINI_5MP or OV5642_MINI_5MP_BIT_ROTATION_FIXED platform.
#if !(defined OV5642_MINI_5MP || defined OV5642_MINI_5MP_BIT_ROTATION_FIXED || defined OV2640_MINI_2MP)
  #error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};
// set pin 7 as the slave select for the digital pot:
const int CS = 7;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;
#if defined (OV2640_MINI_2MP)
  ArduCAM myCAM( OV2640, CS );
#else
  ArduCAM myCAM( OV5642, CS );
#endif
uint8_t read_fifo_burst(ArduCAM myCAM);
void setup() {
// put your setup code here, to run once:
uint8_t vid, pid;
uint8_t temp;
#if defined(__SAM3X8E__)
  Wire1.begin();
  Serial.begin(115200);
#else
  Wire.begin();
[... 353 lines cut ...]
  while ( length-- )
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      Serial.println(F("ACK IMG"));
      Serial.write(temp_last);
      Serial.write(temp);
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
    break;
    delayMicroseconds(15);
  }
  myCAM.CS_HIGH();
  is_header = false;
  return 1;
}

----------------------------------

supprot, 2017-04-04:
@coolb2200 
Hi, 
the SCk is pin 3 and in the code you have set the CS to 7 instead of 10.
Pease check it again.
Let us know if you need more help.
Regards,
ArduCAM support team.

coolb2200, 2017-04-04:
I guess my problem is that I can't even get the Host app to do anything with the camera much less, get the code to even work. I changed the CS pin to both 7 and 10, swapping the pin configuration for both cases. 

I can't even get the host app to work. 

Here is a picture of my hardware setup:

[image]
[image]
[image]

coolb2200, 2017-04-07:
Hello?

supprot, 2017-04-08:
@coolb2200 
Hi,

Don't worry and we will try our best to help you.
Can your camera work on other platform?
Please attach us some pictures about how do you use the host V2 software.
 
Let us know if you need more help.
Regards,
ArduCAM support team.

coolb2200, 2017-04-12:
Well I got it to somehow "work" on the Arduino UNO R3. 

How would I go about getting individual RGB values from each pixel?

coolb2200, 2017-04-12:
Also, why can't I just run the example code and have it work without using the Host app? Isn't your code supposed to work after uploading it? I shouldn't have to keep using the Host app. 

I have been trying to get your code to work but every time I try to use it, it spits back a bunch of errors.

supprot, 2017-04-13:
@coolb2200 
Hi,
Sorry for our late reply.

supprot, 2017-04-13:
@coolb2200 
Hi,
The hardware connnection you have attached is right.
Do you have our adapter board?We suggest you use 
our adapter board which will ensure the connection is 
stable.
About getting RGB values form each pixel,please refer to this code : 
//Read 320x240x2 byte from FIFO
//Save as RGB565 bmp format
for (i = 0; i < 240; i++)
for (j = 0; j < 320; j++)
{
  VH = myCAM.read_fifo();
  VL = myCAM.read_fifo();
  buf[k++] = VL;
  buf[k++] = VH;
  #if defined(ESP8266)
    yield();
  #endif
  //Write image data to bufer if not full
  if (k >= 256)
  {
    //Write 256 bytes image data to file from buffer
    outFile.write(buf, 256);
    k = 0;
  }
}

Let us know if you need more help.
Regards,
ArduCAM support team.

coolb2200, 2017-04-18:
Thank you! Should that code just work? I've yet to test it.

Are you talking about one of these? https://smile.amazon.com/Arducam-Multi-Cameras-Adapter-Board/dp/B01735GYWC?sa-no-redirect=1

Is there anyway I could have code JUST for the OV2640 and nothing else? I notice that a lot of your example code accommodates for both the OV2640 and the OV5640. Do you have live streaming code for JUST the OV2640? It's a lot to go through and the API doesn't really help me all that much.

supprot, 2017-04-19:
@coolb2200 

Hi,

Yes,if you want to use OV2640,you just should enable OV2640_MINI_2MP and disable others
in the memorysaver.h file.Then you can use our examples in MINI folder for OV2640.
IF you want to only have OV2640,you can delete other sensor by yourself.
This link is our demo .
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/mini

Let us know if you need more help.
Regards,
ArduCAM suport team.

coolb2200, 2017-04-21:
Thank you!

Anyway I could just run this code without having to use the host app every time? Is that possible?

supprot, 2017-04-21:
@coolb2200 
Hi,

Yes,
after you get the RGB datas,you can write it to your LCD screen or other device.
After  start capture and capture down, you can read the RGB data from the FIFO.

Let us know if you need more help.
Regards,
ArduCAM support team.

coolb2200, 2017-04-21:
I can't read the RGB values with the given code. I keep getting an "invalid type" error when adding the code in. What is wrong with the code? 

// 1. Set the camera to JPEG output mode.
// 2. Read data from Serial port and deal with it
// 3. If receive 0x00-0x08,the resolution will be changed.
// 4. If receive 0x10,camera will capture a JPEG photo and buffer the image to FIFO.Then write datas to Serial port.
// 5. If receive 0x20,camera will capture JPEG photo and write datas continuously.Stop when receive 0x21.
// 6. If receive 0x30,camera will capture a BMP  photo and buffer the image to FIFO.Then write datas to Serial port.
// 7. If receive 0x11 ,set camera to JPEG output mode.
// 8. If receive 0x31 ,set camera to BMP  output mode.
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
#include <ov2640_regs.h>
//This demo can only work on OV2640_MINI_2MP or OV5642_MINI_5MP or OV5642_MINI_5MP_BIT_ROTATION_FIXED platform.
#if !(defined OV2640_MINI_2MP)
  #error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};
// set pin 10 as the slave select for the digital pot:
const int CS = 10;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;
#if defined (OV2640_MINI_2MP)
  ArduCAM myCAM( OV2640, CS );
#endif
uint8_t read_fifo_burst(ArduCAM myCAM);
void setup() {
// put your setup code here, to run once:
uint8_t vid, pid;
uint8_t temp;
#if defined(__SAM3X8E__)
  Wire1.begin();
  Serial.begin(115200);
#else
  Wire.begin();
  Serial.begin(921600);
#endif
Serial.println(F("ACK CMD ArduCAM Start!"));
// set the CS as an output:
pinMode(CS, OUTPUT);
// initialize SPI:
SPI.begin();
while(1){
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if (temp != 0x55){
    Serial.println(F("ACK CMD SPI interface Error!"));
    delay(1000);continue;
  }else{
    Serial.println(F("ACK CMD SPI interface OK."));break;
  }
}

#if defined (OV2640_MINI_2MP)
  while(1){
    //Check if the camera module type is OV2640
    myCAM.wrSensorReg8_8(0xff, 0x01);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
[... 336 lines cut ...]
  temp =  SPI.transfer(0x00);
  length --;
  while ( length-- )
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    if (is_header == true)
    {
      Serial.write(temp);
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      Serial.println(F("ACK IMG"));
      Serial.write(temp_last);
      Serial.write(temp);
    }
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
    break;
    delayMicroseconds(15);
  }
  myCAM.CS_HIGH();
  is_header = false;
  return 1;
}

supprot, 2017-04-21:
@coolb2200 
Hi,
Please attach us some picture about your phenomenon.
we send 0x30 and read the RGB data by the serial port
and we have test it.

let us know if you need more help.
Regards,
ArduCAM suppor team.

coolb2200, 2017-04-21:
Pictures? I just sent you the code Im using and it won't work due to some "invalid type" errors. You said if I just put in the RGB data code "after start capture" it should work........Could you send me the full code that you're using since it works?

supprot, 2017-04-21:
@coolb2200 
Hi,
Please try this code :

#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
#define BMPIMAGEOFFSET 66
const char bmp_header[BMPIMAGEOFFSET] PROGMEM =
{
  0x42, 0x4D, 0x36, 0x58, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x42, 0x00, 0x00, 0x00, 0x28, 0x00,
  0x00, 0x00, 0x40, 0x01, 0x00, 0x00, 0xF0, 0x00, 0x00, 0x00, 0x01, 0x00, 0x10, 0x00, 0x03, 0x00,
  0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0xC4, 0x0E, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xF8, 0x00, 0x00, 0xE0, 0x07, 0x00, 0x00, 0x1F, 0x00,
  0x00, 0x00
};
// set pin 7 as the slave select for the digital pot:
const int CS = 7;
bool is_header = false;
int mode = 0;
uint8_t start_capture = 0;
ArduCAM myCAM( OV2640, CS );
void setup() {
// put your setup code here, to run once:
uint8_t vid, pid;
uint8_t temp;

  Wire.begin();
  Serial.begin(921600);
Serial.println(F("ACK CMD ArduCAM Start!"));
// set the CS as an output:
pinMode(CS, OUTPUT);
// initialize SPI:
SPI.begin();
while(1){
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if (temp != 0x55){
    Serial.println(F("ACK CMD SPI interface Error!"));
    delay(1000);continue;
  }else{
    Serial.println(F("ACK CMD SPI interface OK."));break;
  }
}
  while(1){
    //Check if the camera module type is OV2640
    myCAM.wrSensorReg8_8(0xff, 0x01);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
    if ((vid != 0x26 ) && (( pid != 0x41 ) || ( pid != 0x42 ))){
      Serial.println(F("ACK CMD Can't find OV2640 module!"));
      delay(1000);continue;
    }
    else{
      Serial.println(F("ACK CMD OV2640 detected."));break;
    } 
  }

    myCAM.set_format(BMP);
    myCAM.InitCAM();
    myCAM.wrSensorReg16_8(0x3818, 0x81);
    myCAM.wrSensorReg16_8(0x3621, 0xA7);
}
void loop() {
// put your main code here, to run repeatedly:
uint8_t temp = 0xff, temp_last = 0;
bool is_header = false;
if (Serial.available())
{
  temp = Serial.read();
  switch (temp)
  {
    case 0x30:
    mode = 3;
    temp = 0xff;
    start_capture = 3;
    Serial.println(F("CAM start single shoot."));
    break;
  }
}

if (mode == 3)
{
  if (start_capture == 3)
  {
    //Flush the FIFO
    myCAM.flush_fifo();
    myCAM.clear_fifo_flag();
    //Start capture
    myCAM.start_capture();
    start_capture = 0;
  }
  if (myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK))
  {
    Serial.println(F("ACK CMD CAM Capture Done."));
   Serial.println(F("ACK IMG"));
    uint8_t temp, temp_last;
    uint32_t length = 0;
    length = myCAM.read_fifo_length();
    if (length >= MAX_FIFO_SIZE ) 
    {
      Serial.println(F("ACK CMD Over size."));
      myCAM.clear_fifo_flag();
      return;
    }
    if (length == 0 ) //0 kb
    {
      Serial.println(F("ACK CMD Size is 0."));
      myCAM.clear_fifo_flag();
      return;
    }
    myCAM.CS_LOW();
    myCAM.set_fifo_burst();//Set fifo burst mode
    
    Serial.write(0xFF);
    Serial.write(0xAA);
    for (temp = 0; temp < BMPIMAGEOFFSET; temp++)
    {
      Serial.write(pgm_read_byte(&bmp_header[temp]));
    }
    SPI.transfer(0x00);
    char VH, VL;
    int i = 0, j = 0;
    for (i = 0; i < 240; i++)
    {
      for (j = 0; j < 320; j++)
      {
        VH = SPI.transfer(0x00);;
        VL = SPI.transfer(0x00);;
        Serial.write(VL);
        delayMicroseconds(12);
        Serial.write(VH);
        delayMicroseconds(12);
      }
    }
    Serial.write(0xBB);
    Serial.write(0xCC);
    myCAM.CS_HIGH();
    //Clear the capture done flag
    myCAM.clear_fifo_flag();
  }
}
}

Let us know if you need more help.
ArduCAM support team.

coolb2200, 2017-04-21:
I ran your code....it doesn't read or print the RGB values of each pixel and it doesn't automatically launch, let alone work. That's all I need the camera to do. I don't know what it is about this camera but it doesn't want to work. Either that or the API needs updating. 
[image]
[image]

supprot, 2017-04-22:
@coolb2200 
Hi,
What's your baud rate? 
It seems your baud rate is not right.
Regards,
ARduCAM support team.

DigitalSocrates, 2017-11-21:
Hi, recently purchased OV2640 and followed guide on how to connect to mega 2560 board - but can not make it work at all. SPI test passes but vid and pid are both set to 0 so the code throws an error as module is not recognized

UCTRONICS, 2017-11-21:
@DigitalSocrates
Hi,
How do you connect the camera?
Please attach us some pictures for your hardware connection.
You‘d better check your I2C bus with an oscilloscope.
Arducam support team.

DigitalSocrates, 2017-11-21:
thank you for you response. i opened a different issue for this
see https://github.com/ArduCAM/Arduino/issues/278 

will include more info there

Gaby-8886, 2022-06-11:
i need help on how to connect OV2640 18 pin camera to Arduino Uno and also the code.
i try using the code for ArduCam mini OV2640 the message i got is a continues SPI interface Error